tbnobody/OpenDTU #1965: only last Version 2024.01.26 support HMS 4t 2000 Inverter

krelac3, 2024-05-05:
### What happened?

the current and down to 2024.2.x Firmware doesn't support the HMS 4t 2000 valid. the live view Shows not the complete Inverter data and always the message 'warten auf daten' apeaers. Is this Inverter out of support?

### To Reproduce Bug

yes, always there

### Expected Behavior

Support of HMS, live view data shown

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

Any after 2024.01.26

### Anything else?

https://github.com/helgeerbe/OpenDTU-OnBattery/issues/912

krelac3, 2024-05-05:
[image]

krelac3, 2024-05-05:
Brand new HMS-2000-4T. Downgraded the ESP firmware and seems only 2024.01.26 shows the data from inverter, any newer firmware shows "wait for data..." forever.

tbnobody, 2024-05-05:
Just clear your browser cache after the upgrade or reload the page with CTRL+F5

krelac3, 2024-05-05:
yep, all tried and no go. directly connected network, nothing in between. fw version 2024.01.26 works, data is there and integration with VenusOS nf rpi works fine. 

No go for any newer firmware..

Logs attached (my inverter serial replaced with XXX_MY_HOYMILES_XXX)

[opendtu_capture_2024.01.26.txt](https://github.com/tbnobody/OpenDTU/files/15213730/opendtu_capture_2024.01.26.txt)
[opendtu_capture_2024.03.23.txt](https://github.com/tbnobody/OpenDTU/files/15213731/opendtu_capture_2024.03.23.txt)

tbnobody, 2024-05-05:
```
Fetch inverter: XXX_MY_HOYMILES_XXX
TX RealTimeRunData 865.00 MHz --> 15 93 30 38 14 80 18 19 76 80 0B 00 66 37 B2 AF 00 00 00 00 00 00 00 00 1C D3 65 
Interrupt received
RX 865.00 MHz --> 95 93 30 38 14 80 18 19 76 01 00 01 01 39 01 37 00 8F 00 91 01 C2 01 C4 00 00 FB | -56 dBm
Interrupt received
RX 865.00 MHz --> 95 93 30 38 14 80 18 19 76 02 2E CD 00 00 2D C8 06 AE 06 95 01 3C 01 3D 00 8E 5D | -56 dBm
Interrupt received
RX 865.00 MHz --> 95 93 30 38 14 80 18 19 76 03 00 8E 01 C1 01 C3 00 00 2E 94 00 00 2D 25 06 A0 76 | -56 dBm
Interrupt received
RX 865.00 MHz --> 95 93 30 38 14 80 18 19 76 04 06 6B 08 AC 13 86 06 B1 00 00 00 4D 03 E8 01 07 A2 | -56 dBm
Interrupt received
RX 865.00 MHz --> 95 93 30 38 14 80 18 19 76 85 00 01 46 5D 72 | -56 dBm
RX Period End
Success
```

All data received. But it seems that your browser does not support web sockets or that there is a firewall issue. But it seems you are not using the original image from this Image. 
These log outputs are not from this firmware
```
[DPL::announceStatus] waiting for valid date and time to be available
``` 

So please report this issue in the correct repository.

ms49434, 2024-05-05:
Just a heads up, same inverter model / manufacturing date is running on my site without problems with OpenDTU 24.4.24:
[image]
[image]

schlimmchen, 2024-05-05:
> So please report this issue in the correct repository.

I suspected this might be an upstream issue, that might be the reason why this issue was also reported here. I had hoped that there is a similar issue already known here.

There is one other user (in the downstream issue) who also said that he has no problems with this inverter. However, there are also two users who have the same problem.

We'll try to debug this in the downstream issue and will let you know.

tbnobody, 2024-05-05:
I have a HMS-2000 in front of me which works perfectly.

krelac3, 2024-05-05:
I stand corrected. 

All works fine and as expected on either openDTU or OpenDTU-OnBattery latest firmware. Seems my local running tinyproxy on loopback made an issue although there were no issues when running older firmware versions. Not sure what changed in last 6 months or so within firmware. In any case, bypassing DTU's ip in proxy sort it out and data from inverter shows up. 

Thank you for the assistance.

closed 2024-05-05 by krelac3

tbnobody, 2024-05-05:
The only change was, that websockets are absolutly necessary now. In previous versions, the whole set of data was transmittet once via the initial GET request and then just updated via web sockets. Now the websocket transmits all the data.