tbnobody/OpenDTU #2003: Limit Value of inverter displayed wrong

schwidl, 2024-05-23:
### What happened?

After restarting of openDTU, the Limit of one inverter is displayed 0%, but the inverter is producing something.
The permanent limit of the inverter is definetly set to 100%.
After around 5 minutes, the displayed limit changes to 100%.

### To Reproduce Bug

Just restart the openDTU and watch the limits of the inverters. They are not displaying the real limit. 
[image]

### Expected Behavior

I expect the limit value to display correctly from the beginning and not needing several minutes to display correctly.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.5.6

tbnobody, 2024-05-23:
Thats normal behavior. The limit can only read every 4min without flooding the event log with error messages. Therefor there is a waiting time of 4 minutes before reading the value.

closed 2024-05-23 by tbnobody

schwidl, 2024-05-23:
Alright, then it´s a feature. Thanks!