tbnobody/OpenDTU #2008: Display freezes completely

Dopfen, 2024-05-25:
### What happened?

Hi, 
I'm using OpenDTU on a display unit with OLED display. 
In irregular periods, the display freezes completely. It happens surely once within 4 days always in the morning, some minutes after the micronverter had started and some data was displayed. 

It first occurred when using firmware 23.12.19, offering the great new diagram drawing. It happens never before. 
Then, I almost installed every sucessing firmware. Even with the nevest 24.5.6 the issue occurrs.
 
Some info:
Microinverter is Hoymiles HM-400.

Display unit:
ESP32-D0WD-V3 with OLED display. 

Firmware:
Version: 24.5.6
Host name: OpenDTU-28ED64
SDK versiom: 4.4.6
Configuration version: 0.1.28

See also screenshots from memory and heap. 

Notes:
For OLED displays, freezing pixels tend to burn into the display, so it is bad for the hardware. 

When the display freezed, it is still possible to enter the OpenDTU dashboard.

Would be great if you could help me or at least investigate the issue.

Best regards,
Daniel

### To Reproduce Bug

Not predictable. 

### Expected Behavior

No freezing display. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.5.6

### Relevant log/trace output

```Shell
No log file yet.
If helpful, I could upload a log file (please show me how)
```

### Anything else?

See pics from the device and screenshots from OpenDTU dashboard for detailed info on memory and heap. 

[image]
[image]
[image]

tbnobody, 2024-05-25:
That lookls like that either the display is defekt or the power supply is bad. (Power Supply itself, wiring, capacitor etc.)
When updating more pixels (like redrawing a whole diagram) the display consumes more power.

tbnobody, 2024-05-25:
I am going to convert this to a discussion as it is for sure no software issue.