tbnobody/OpenDTU #2014: Arduino-ESP 3.0 & ESP32-{H2,C6,C2} support

hd1ex, 2024-05-27:
### Is your feature request related to a problem? Please describe.

Today, the shiny new [Arduino Release v3.0.0 based on ESP-IDF v5.1.4](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0) arrived.

This is the new major version of the Arduino abstraction of the Espressif framework. In the long run, it seems like the vast majority of further development will happen there, so this should be motivation enough to switch to it (maybe compare it with changing from python 2 to 3).

My main motivation is the support of the ESP32-C6 because I have designed a DTU-Dev-Board around it.

### Describe the solution you'd like

My question is if someone is already working on supporting this?

A month ago, I created a [really hacky version](https://github.com/hd1ex/OpenDTU/tree/esp32-c6) with working MCU, Webserver and OLED by commenting a lot of code out and mocking some missing classes.

For realizing this (potentially major) rebase of the OpenDTU code, I see two possible paths:
1. Either, we create a long pull request for people to test on and finally merge it after positive feedback.
2. Or we create an `arduino-esp32-v3` dev branch for this if work will be a lot.

Maybe we can also take a look at the (new) API changes of Espressif's Arduino abstraction and get rid of a dependency or another.

### Describe alternatives you've considered

We could also switch to bare [esp-idf](https://github.com/espressif/esp-idf), but I this would take a lot more (unnecessary) effort.

hd1ex, 2024-05-27:
Have forgotten to say this: If someone of this community will be at the [Gulaschprogrammiernacht 22 in Karlsruhe](https://gulas.ch) this weekend, we can also take a look at this together or talk about it. I will also bring my dev boards :)

tbnobody, 2024-05-27:
Give it some time. Its not event integrated into the platformIO build system. 

Not and GPN. Only at VVoid Camp.

hd1ex, 2024-05-31:
> Give it some time. Its not even integrated into the platformIO build system.

Yeah, you are right. This is issue currently is more meant to track progress and collect if someone is working on this.
Sadly, it [seems like there is some political stuff](https://github.com/espressif/arduino-esp32/issues/8796#issuecomment-2141622241) going on which blocks the progress.

> @hd1ex as fun as it is, would you rather classify this as an issue, or a possible enhancement discussion within this specific project ?

At the moment, I would classify this as an enhancement. Especially for supporting the new ESP32 chips.
In the long run, it probably will become more of an issue because the development may will follow the newer version (but I did not have a look further into this).

stefan123t, 2024-08-30:
> ESP32-C6 support would be good.

i recall the C6 is the RISC based architecture. I do not know if this can be easily supported instead of the classic ESP32-S3 architecture we normally use otherwise ?

> Many have WiFi problems in connection with a FritzBox and Mesh. I have these problems too. It would be interesting to know if the same problem occurs with WiFi6.
> 
> So far I have only found one solution: Set WiFi channel to 1. Then you can achieve what Webif. All other channels do not work.

The second part of your question is off topic here and has been analysed there https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2289412118

Both
* 802.1k/r for mesh roaming, aka (Fast) Basic Service Set Transitioning
* 802.1v for mesh steering, or Wireless Network Management
need to be implemented in OpenDTU to support Mesh WLAN on ESP32 using either of the ESP-IDF roaming examples code:
https://github.com/espressif/esp-idf/tree/master/examples/wifi/roaming

stefan123t, 2024-08-30:
@Andre0be please watch or subscribe thebother issue. This will be discussed or announced there.