tbnobody/OpenDTU #2016: Anzeige von Ertrag falsches Tausender Trennzeichen "," an Stelle von "."

Balteus, 2024-05-28:
### What happened?

"Inverter Gesamtertrag Heute" : Korrekter Punkt als Tausender Trennzeichen.
"Inverter Gesamtertrag Insgesamt": Hier wird ein Komma verwendet als Tausender Trennzeichen
"Gesamtertrag" bei den Modulen: Wird ein Komma verwendet.
"Tagesertrag" bei den Modulen: Wird korrekter Weise ein Punkt verwendet.

### To Reproduce Bug

Generell falsch verwendetes Zeichen für die Anzeige.

### Expected Behavior

Es sollte immer ein Punkt als Tausender Trennzeichen verwendet werden.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v4.4.6-dirty - 2024.05.07

tbnobody, 2024-05-28:
> Es sollte immer ein Punkt als Tausender Trennzeichen verwendet werden.

Das ist so nicht korrekt! Punkt und Komma sind abhängig von der eingestellten Sprache. 

Bitte stelle mal einen Screenshot der Werte inkl. eingestellter Sprache bereit.

Balteus, 2024-05-28:
Sprache ist deutsch, auch wenn ich keine Möglichkeit sehe, diese zu ändern (außer bei Hardware/Display, was aber nur für das kleine Display am Gerät von Bedeutung sein sollte).
[image]

ms49434, 2024-05-28:
Stimmt doch alles, das Tausender-Trennzeichen ist ein Punkt und der Dezimal-Trenner ist ein Komma auf dem Screenshot. Welcher Wert wird denn falsch angezeigt?

Balteus, 2024-05-28:
Danke! Mein Bug Report ist dann leider ein falscher Alarm, Sorry.