tbnobody/OpenDTU #2018: Anzeige der CPU-Temperatur unter System möglich

tm-107, 2024-05-28:
### Describe the solution you'd like

Da der ESP32 meines Wissens nach einen internen Temperatursensor hat, wäre es vielleicht schön diesen Wert unter System darzustellen. Würde mich interessieren wie warm das Teil im meist geschlossenen Gehäuse wird.
Meine Shellys (Plus 1PM mit Tasmota) bringen es in der Verteilung ja schon auf über 60°C.

Schaefers, 2024-06-03:
This is definitely possible, I just did a quick test using `temperatureRead()`: 

[image]

@tbnobody if it helps, I could create a MR, but I guess it's just copy paste and some formatting.

tbnobody, 2024-06-03:
It's already implemented in my local branch. Will be included in the next release.

tbnobody, 2024-06-10:
Implemented in 6ce474053e280182f1f41327e9dd42d055171ee0

closed 2024-06-10 by tbnobody