tbnobody/OpenDTU #2026: Nach Neustart geht der Wechselrichter Typ verloren und API Abfrage ist nicht mehr möglich

dimi21, 2024-06-02:
### What happened?

Wenn ich die openDTU neu flashe (V24.5.6) und einrichte (HMS-1600), bekomme ich mit folgendem Python Code brauchbare Antworten (Seriennummer uvm): 

r = requests.get(url = f'http://{dtu_ip}/api/livedata/status/inverters' ).json()
(oder im Browser: http://192.168.178.64/api/livedata/status)
[image]
[image]

Wenn ich  die DTU danach neu starte oder vom Strom trenne und wieder anstecke, kommen keine Werte mehr über die Selbe Abfrage, obwohl die DTU online und per IP Adresse erreichbar ist. 
Wenn man dann auf Einstellungen -> Wechselrichter geht hat sich der Typ von HMS-1600 auf unknown geändert.
[image]
[image]

Lösen kann man das Problem, wenn man dann bei Aktion -> bearbeiten -> ohne Änderung Speichern -> und alles ist wieder OK

### To Reproduce Bug

ein einfacher Neustart den Browser oder eine Unterbrechung der Stromversorgung führt zum Fehler.

### Expected Behavior

Dass nach einem Neustart der DTU, der Wechselrichter Typ nicht manuell in der DTU eingestellt werden muss und die API Kommunikation möglich ist.  

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.5.6

### Anything else?

ein Video vom beschriebenen Verhalten kann ich hier zur Verfügung stellen:

https://github.com/tbnobody/OpenDTU/assets/35991362/bb9c3ab9-5e2b-442e-84bf-73a98fcd461b

tbnobody, 2024-06-02:
Habe das gerade eben getestet.... deine Seriennummer eingetragen, neu gestartet. wird wieder als `HMS-1600/1800/2000-4T` erkannt.
Kannst du ein Backup deiner Config machen, deine Wifi Credentials entfernen und mir zukommen lassen?

dimi21, 2024-06-02:
Vielen Dank für die rasche Rückmeldung, im Anhang ist ein Backup der Config die ich im "Fehlerzustand" erstellt habe. 

[config (2).json](https://github.com/user-attachments/files/15526843/config.2.json)

tbnobody, 2024-06-02:
Hmmm was ist die Ausgabe von Info --> System und hier speziell "Funkmodulinformationen". 
Wenn kein CMT Modul konfiguriert ist entsteht u.a. genau dieses verhalten nach einem Neustart.

dimi21, 2024-06-03:
Vielen Dank für diesen Hinweis!
Konsole vor dem lösen:

[image]

Ich habe das Profil geändert und es funktioniert.
Konsole nach dem lösen:

[image]

closed 2024-06-03 by dimi21