tbnobody/OpenDTU #2027: Unknown device

sissiwup, 2024-06-02:
### What happened?

Hello,

my hoymiles hm-600 is listed as unknown device 
Hardwarenumber 132449357

### To Reproduce Bug

Take my hoymiles .)

### Expected Behavior

Device is known

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	2024.05.07

Solarteur, 2024-06-05:
When setting up an inverter, you are not asked for the hardware number, but for the serial number. 
For the HM-600, it starts with 1141xxxx xxxx and can be found on the inverter.

tbnobody, 2024-06-06:
Please add a screenshot where you see the "Unknown" device message. If it's in the inverter overview then you entered the wrong number as @Solarteur already mentioned.

sissiwup, 2024-06-13:
Hi, it is not the inverter serial number. It is the hardwarenumber. The error message says, make a ticket with the hardware number.

tbnobody, 2024-11-07:
No screenshot since june... going to close this issue.