tbnobody/OpenDTU #2028: Grid Profile is detected as XX - EN 50549-1:2019 but Smiles Cloud shows as 2.0.0 (DE_VDE4105_2018)

pioneer-01, 2024-06-03:
### What happened?

The Grid Profile interpreted by OpenDTU as XX - EN 50549-1:2019 but Smiles Cloud shows as 2.0.0 (DE_VDE4105_2018)

I use a Hoymiles HM-600-2T.

The Grid profile shown in OpenDTU is

[image]

Hoymiles cloud shows something different
[image]

### To Reproduce Bug

Not sure how :-)

### Expected Behavior

The Grid Profile is shown correctly, which I guess is the DE_VDE4105_2018

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.5.6

pioneer-01, 2024-06-03:
I checked here as well https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser
and found the profile code in this table as well
[image]

tbnobody, 2024-06-03:
> Hoymiles cloud shows something different

Thats interessting because `20 01`
[image]

in your screenshot should be the version number of the profile. This does not match the version number in your Holymiles account. The screenshot of the wiki pages shows the right information.

tbnobody, 2024-06-03:
Have you tried to restart OpenDTU? The gridprofile is only read once. If you maybe changed in during the runtime it is not updated again.

ms49434, 2024-06-03:
> Yes, I restarted OpenDTU by even unplugging the USB power adapter. After restart the profile shows still as XX - EN 50549-1:2019

I've seen this problem in the past with one of my inverters where the information provided by the Hoymiles Cloud was different to the information read from the inverter (via OpenDTU). I managed to solve the issue by reapplying the correct grid profile from a desktop browser (It didn't work with the Hoymiles Installer App).

ms49434, 2024-06-03:
> To reapply the profile, you used the Hoymiles DTULite stick instead of OpenDTU for a while and applied the profile via S-Miles website?

The profile was set/updated with a Hoymiles DTU Pro S, an installer account and login from a desktop browser at https://global.hoymiles.com/platform/login

pioneer-01, 2024-06-03:
Thanks, I will use my Hoymiles DTU Stick now to give it a try to re-apply the profile. Then later I can check in OpenDTU again, which is of course currently not running in parallel. 

I have to add, that I am located in Munich, Germany.

the Hoymiles cloud offers only one profile for my location:
[image]

pioneer-01, 2024-06-03:
Ok, after reapplying the grid profile everything is shown correctly in OpenDTU. 
[image]
[image]

It seems the issue was never on OpenDTU side but Hoymiles cloud screwed something up.

tbnobody, 2024-06-03:
In this case I would close the issue. 

But interesting. It seems that the original DTU does send the grid profile but does not check whether it was successfull.

closed 2024-06-03 by tbnobody