tbnobody/OpenDTU #2034: Login-Felder bei "Anmelden" sind verdeckt

Fokker-1, 2024-06-04:
### What happened?

Hallo, 

ich habe eine fusion DTU mit aktueller FW, funktionirt prima, außer einem Problem am Samsung Smartphone:

Wenn ich auf "Anmelden" tippe, dann sind im Folgescreen die Eingabefelder für "User" und "Passwort" verdeckt.
Erst durch mehrmaliges  Hin- und Herschieben des Anmelde-Buttons erscheinen sie dann plötzlich, dabei kann ich keine Systematik erkennen.

Gruß und danke
Werner

[image]

[image]

### To Reproduce Bug

Auf "Anmelden" tippen sollen die Eingabefelder für "User" und "Passwort" automatisch sichtbar werden (das war früher auch mal so).

### Expected Behavior

Nach Antippen von "Anmelden" müssen die Anmelde-Felder erscheinen.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.5.6

### Relevant log/trace output

```Shell
2 Fotos
```

### Anything else?

nein

Fokker-1, 2024-06-06:
Hallo,
vielen Dank für den Hinweis, da muss man erstmal drauf kommen!
Prima, so funktioniert es jetzt auch bei mir.

Trotzdem sollte das Problem in einem Update gelegentlich behoben weren.

Gruß
Werner

tbnobody, 2024-11-07:
Please doublecheck if the issue still persists in the latest version.

Fokker-1, 2024-11-08:
Hallo,

danke für das Update.
Es ist besser geworden, aber leider fehlt noch eine Teil der vollständigen Darstellung.

Ausgangspunkt:

[image]

Nach Klick auf "Anmelden" unten rechts (grüner Text) kommt:

[image]

Nun Klick auf "Anmelden" (blauer Text) unten rechts, dann kommt:

[image]

Hier ist jetzt zwar das rote Feld "Passwort" vollständig sichtbar, aber das Feld "Benutzername" bleibt weiterhin verdeckt.

Erst wenn man nun erneut oben auf die 3 horizontalen Striche klickt, erscheint es:

[image]

Soweit meine Infos.

Grüße
Werner

tbnobody, 2024-11-08:
> Erst wenn man nun erneut oben auf die 3 horizontalen Striche klickt, erscheint es:

Hm ja... Das ist ja absicht so... man klappt auf handys mit dem "sandwich" menü die einträge auf und anschließend wieder zu.

stefan123t, 2024-11-09:
Also wird beim Klick auf dem grünen Button im Hamburger Menü zwar die Login view der Single Page Application geladen, aber das Hamburger Menü bleibt offen ?

Fokker-1, 2024-11-09:
Hallo,

sorry, aber ich verstehe es (noch) nicht.

Ziel sollte es m. E. idealerweise doch sein, dass bereits beim Anklicken des grünen Buttons "Anmelden" das komplette (!) rote Menü "Systemanmeldung" geöffnet wird. Warum noch Zwischenstufen?

Interessant aber, dass es am WIN-PC genau so funktioniert, beim hier beschriebenen Android-Handy aber anders.

Klicke ich am WIN-PC auf des grüne Menü "Anmelden", wird sofort und ohne weitere Zwischenstufen genau dieses rote Menü "Systemanmeldung" komplett geöffnet.

Gruß
Werner