tbnobody/OpenDTU #2038: Button "Gesamtertrag" bei String 2 im Startscreen nicht sichbar am Handy

Fokker-1, 2024-06-06:
### What happened?

Hallo,

bei meinem Samsung S20 und auch beim A40 ist beim Startscreen die letzte Zeile mit dem Button "Gesamtertrag" nicht sichbar.

Versucht man, den Screen weiter nach oben zu schieben, verschwindet weder die Zeile "OpenDTU" noch die erste Zeile ganz oben mit der IP-Adresse. Dieses Problem hatte ich in früher nicht.

Die DTU hat die aktuelle FW, es ist eine fusion DTU.

Vielen Dank an Euch noch mal an dieser Stelle für die tolle Arbeit!

Gruß
Werner

[image]

[image]

### To Reproduce Bug

Einfach nur das Programm starten im Startscreen nach unten scrollen.

### Expected Behavior

Die Zeile "Gesamtertrag" soll wieder sichtbar werden.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v.24.5.6

### Relevant log/trace output

```Shell
nein
```

### Anything else?

Vielen Dank an Euch!

tbnobody, 2024-06-06:
> bei meinem Samsung S20 und auch beim A40 ist beim Startscreen die letzte Zeile mit dem Button "Gesamtertrag" nicht sichbar.

Es gibt keinen Button "Gesamtertrag". Was genau meinst du?

> Einfach nur das Programm starten im Startscreen nach unten scrollen.

Hier funktioniert es sowohl im Browser als auch in den Dev-Tools von Chrome und Firefox

> Versucht man, den Screen weiter nach oben zu schieben, verschwindet weder die Zeile "OpenDTU" noch die erste Zeile ganz oben mit der IP-Adresse. Dieses Problem hatte ich in früher nicht.

Was bedeutet früher? Welche Version? Und das die IP Adresse verschwindet liegt nicht in der macht der Webseite. Das muss der Browser selbst machen.

Generell kann es sein, dass der Browser vom S20 so alt ist (das Handy selbst kam 2020 raus und Android ist nicht dafür bekannt, dass es häufig aktuelle Betriebssysteme bekommt), dass es mit neueren Web-Techniken nicht zurecht kommt. Wobei sich an der Webseite im Grundlegenden schon ewig nichts mehr geändert hat.

Fokker-1, 2024-06-06:
Danke für das Feedback.
Du schreibst:

"Es gibt keinen Button "Gesamtertrag". Was genau meinst du?
    Einfach nur das Programm starten im Startscreen nach unten scrollen."

Natürlich gibt es diese Button für beide Strings, verschieben nützt nichts am Handy.
Hier mal die beiden Button bei Strin 1 + 2 über den WIN-PC dargestellt:

[image]

Du schreibst:

"Hier funktioniert es sowohl im Browser als auch in den Dev-Tools von Chrome und Firefox

    Versucht man, den Screen weiter nach oben zu schieben, verschwindet weder die Zeile "OpenDTU" noch die erste Zeile ganz oben mit der IP-Adresse. Dieses Problem hatte ich in früher nicht."

Verschieben bringt nichts, habe ich ja alles schon versucht. Wann genau das "früher" war, habe ich natürlich nicht notiert. Der Unterschied mag sein, dass ich "früher" einen HM-800 hatte und damit auch eine andere DTU, jatzt habe ich ja einen HMS-1000 mit einer fusion DTU. Ich verwende FF beim Handy und unter WIN.

Du schreibst:

"Generell kann es sein, dass der Browser vom S20 so alt ist (das Handy selbst kam 2020 raus und Android ist nicht dafür bekannt, dass es häufig aktuelle Betriebssysteme bekommt), dass es mit neueren Web-Techniken nicht zurecht kommt. Wobei sich an der Webseite im Grundlegenden schon ewig nichts mehr geändert hat."

Bei allen anderen Anwendungen, die ich daraufhin getestet hatte, verschwindet die oberste Zeile mit der IP, wenn man hoch scrollt, hier aber nicht.

Gruß
Werner