tbnobody/OpenDTU #2046: Diverse Probleme mit OpenDTU

grasmax, 2024-06-10:
### What happened?

-	Keine stabile WLAN-Verbindung, auch mit oder gerade wegen externer Antenne.
-	Das Gerät ist im Netzwerk angemeldet, man kommt aber nicht ran., d.h. nur „nicht erreichbar“ im Browser.
-	Man sieht von außen nicht, dass das Gerät lebt.
-	Im Chrome-Browser in Android 13 /OneUI 5.1 wird das Limit nicht angezeigt.
-	Die Öffnung im Gehäuse ist zu klein für manche USB-Stecker.
-	Vermutlich Wackelkontakt an der USB-Buchse. 
-	Sendeleistung ist nicht per Script änderbar.
-	Habe keine realistische Angabe zur Leistungsaufnahme gefunden. Die 2A-Angabe scheint mir viel zu hoch zu sein. 

### To Reproduce Bug

Test der WLAN-Verbindung mit der empfohlenen externen Antenne.
Test  mit Chrome-Browser in Android 13 /OneUI 5.1 
Handelsübliche USB-Stecker testen.

### Expected Behavior

Stabile WLAN-Verbindung, gerne auch mit externer Antenne.
Betriebs-LED soll auch bei geschlossenen Gerät sichtbar sein.
Anzeige des eingestellten Limits auch im  Chrome-Browser in Android 13 /OneUI 5.1.
Größere Öffnung im Gehäuse für den USB-Stecker
Anderer Steckverbinder oder Lötkontakte für die Stromversorgung.
Sendeleistung soll per Script änderbar sein.
Realistische Angabe zur Leistungsaufnahme. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.1.14

tbnobody, 2024-06-10:
Wenn du eine Änderung an der Hardware möchtest solltest du diese einfach machen. OpenDTU ist ein Software Projekt und liefert keinerlei Hardware. Dies liegt beim Benutzer hier etwas zu bauen oder anderweitig zu besorgen.

closed 2024-06-10 by tbnobody

grasmax, 2024-06-10:
btw, sdk version is v4.4.6-dirty....how to get a stable one?

grasmax, 2024-06-10:
"OpenDTU ist ein Software Projekt" - ok, alles klar. Das Software-Projekt ist super.
Gehört der GUI zum "Software-Projekt"? Gibt es im "Software-Projekt" die Möglichkeit, die Sendeleistung zu ändern und ist das nur nicht in den GUI eingeflossen?

tbnobody, 2024-06-10:
> btw, sdk version is v4.4.6-dirty....how to get a stable one?

Wende dich hier bitte an Espressif, den Hersteller des Microcontrollers. Dieser nennt nun mal seine SDK so wie er sie nennt.

> Gibt es im "Software-Projekt" die Möglichkeit, die Sendeleistung zu ändern und ist das nur nicht in den GUI eingeflossen?

Ja gibt es. Steht u.a. in der Dokumentation.

grasmax, 2024-06-10:
Vielen Dank für die schnellen Antworten.

grasmax, 2024-06-10:
Sorry, hatte die Frage falsch gestellt. Nochmal richtig: Im "Software-Projekt" gibt es eine Einstellung für die Sendeleistung, die auch im GUI verfügbar ist.
Im API https://www.opendtu.solar/firmware/web_api/ gibt es sie offenbar nicht.
Wäre es möglich, das nachzuholen. Wenn ja; wie? 
Viele Grüße, verbunden mit der Bitte um eine kurze Anleitung zum Thema "API-Erweiterung".

ms49434, 2024-06-10:
Kann man denn die  Sendeleistung nicht schon über die Web-API konfigurieren?
Siehe nrfpalevel / cmtpalevel unter http[s]://[IP OpenDTU]/api/dtu/config

grasmax, 2024-06-10:
@ms49434: Stimmt, funktioniert! Danke schön.
Ich frage mich, wo das dokumentiert ist, wenn nicht hier: https://www.opendtu.solar/firmware/web_api/

grasmax, 2024-06-10:
Beispiel, in Python, zum Setzen der CMT-Sendeleistung auf 17dBm und Abfrageintervall auf 8 Sekunden:

```
      ret = requests.get(
                  url = f'http://{dtu_ip}/api/dtu/config',
                  #data = f'data={{"serial":"{serial}", "limit_type":0, "limit_value":{setpoint}}}',
                  auth = HTTPBasicAuth(dtu_nutzer, dtu_passwort),
                  headers = {'Content-Type': 'application/x-www-form-urlencoded'}
              )
      jData = ret.json()

      ret = requests.post(
                  url = f'http://{dtu_ip}/api/dtu/config',
                  data = f'data={{"serial":"{serial}", "pollinterval":8, "nrf_enabled": "True", "nrf_palevel":0, "cmt_enabled": "True", 
                  "cmt_palevel":17,"cmt_frequency": 865000}}',
                  auth = HTTPBasicAuth(dtu_nutzer, dtu_passwort),
                  headers = {'Content-Type': 'application/x-www-form-urlencoded'}
              )

```

grasmax, 2024-06-10:
Oh je, die Sendeleistung wurde zwar per Script geändert, wie oben zu sehen.
Aber danach funktioniert der Zugriff auf den Wechselrichter nicht mehr.
Nochmaliges Setzen der Sendeleistung über den GUI  hilft nicht.
Strom ein/aus hilft auch nicht.
Einen ganzen Tag mit WLAN und Sendeleistung verdaddelt und am Ende funktioniert gar nichts mehr. rrrrrrrrrrrrrrr

ms49434, 2024-06-10:
Diese Einstellungen sind ja auch vorsichtig anzuwenden. Wenn ein Funkmodul zu laut schreit, kann es sein daß der Empfänger taub wird. Da hilft es nur, die Sendeleistung wieder zurückzunehmen und abzuwarten. Es kann bis zu 15 Minuten dauern, bis sich WR und DTU wiederfinden.

grasmax, 2024-06-11:
@ms49434: Danke für die sicher gut gemeinten Ratschläge. Da hat sich aber auch über Nacht und durch Neustart nichts "wiedergefunden".
@alle: Das Setzen von Einstellungen per Script unterbricht die Funkverbindung zum Wechselrichter dauerhaft.
Ich rate dringend, VOR solchen Versuchen, die Konfigurationsdatei zu retten. 
Zum Glück konnte ich das Problem lösen durch Einlesen der vorher gesicherten config.json.
**Und AUWEIA: So was geht gar nicht! In der Konfig-Datei steht eurer WLAN-Passwort im Klartext drin!!!**
**Mich würde interessieren, wer dafür verantwortlich ist.**

Omega13x, 2024-06-11:
Wo liegt das Problem?
Du sollst ja mit der Konfig. nicht hausieren gehen. In meinem Ausweis steht mein Name, Geburtsdatum und Wohnort in Klartext drin. Da dran zukommen ist deutlich einfacher.

grasmax, 2024-06-11:
Klartext-Passworte zu verteidigen geht gar nicht. Sag mir lieber, wer dafür verantwortlich ist. Ich helfe gern bei der Behebung.