tbnobody/OpenDTU #2054: falsche Berechnung der gesammten AC Leistung Phase 1 bei niedriger String Spannung (Zendure AB2000)

TvT73, 2024-06-11:
### What happened?

Seitdem ich den Zendure Solarflow 1200 mit einem AB 2000 an meinem Hoymiles HMS1800-4T betreibe beobachte ich folgendes Fehlerbild:
- Wenn der Akku Leistung abgibt mit mehr als 60W und einer Spannung von >40V stimmt die Berechnug der Summer der beiden Ports, in meinem Fall 3 + 4
- Abends im Schwachlast Betrieb senkt der Zendure die Batteriespannung auf ca. 18-<25V in der Summe wegen der Parallelschaltung von WR 3+4 werden nach Zendure weiterhin z.B. 60W eingespeist. 

Die OpenDTU in egal welcher Version ermittelt dann aber immer nur die Leistung AC von einem String!

In der Folge davon wird sowohl der Wirkungsgrad falsch berechnet als auch möglicherweise die Tagesertragssumme? Da bin ich mir nicht sicher. Der WIrkungsgrad ist mit <50% dann definitiv falsch. 

Ich sehe immer ein Springen der AC Leistung auf Phase 1 wenn die Spannung am Hub auf >40V geht.

### To Reproduce Bug

Immer reproduzierbar

### Expected Behavior

Summen aller Strings müssen adiert werden, auch bei einer Stringspannung von <20V

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

OpenDTU und Open DTU on Battery v4.4.6-dirty

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-06-11:
> Die OpenDTU in egal welcher Version ermittelt dann aber immer nur die Leistung AC von einem String!

> In der Folge davon wird sowohl der Wirkungsgrad falsch berechnet als auch möglicherweise die Tagesertragssumme?

OpenDTU errechnet hier garnichts. Die Werte kommen direkt so aus dem Inverter. Wenn dann rechnet der Inverter falsch. Dann kann hier aber auch nichts geändert werden.

TvT73, 2024-06-18:
[image]
[image]

Nun, da wäre ich mir nicht so ganz sicher. Wir können den Wert Summe Inverter nicht ändern. Das sehe ich auch so. Allerdings ist dei Summe Tagesertrag doch auch nur eine berechnete Summe. Könnte man nicht diese selber berechnen statt vom Inverter zu nehmen?

Defakto ist der Messfehler extrem hoch. Als Beispiel von heute werden hier rund 10mal so hohe Ertragswerte gemessen. 
Ich habe sowohl vom Hub gemessen knapp 560wh erzeugt, als auch der Shelly mir die Werte vom Zendure bestätigt.

So kann man auf jeden Fall die Messwerte vollkommen vergessen. 

Ja es mag daran liegen, das der Hoymiles es falsch macht, aber dann verstehe ich nicht, warum man das nicht häufiger findet. Wenn ich das gewusst hätte, wäre ich vermutlich auf einen anderen WR gegangen. Auch wenn ich einen Workoround mit einem einzelnen Shelly machen werde, schön ist das nicht!

Zumal der Zendure grotten langsam seine Ausgabeleistung erhöht. Das istz auch im Zendure Forum bekannt. Daran ändert auch die Open DTU on Battery nichts, die gibt die Leistungsanforderung innerhalb rund 2s weiter.

Allerdings wundere ich mich auch darüber, dass der Messintervall des Shelly 3 em Pro in der OpenDTU nicht auch im 1s takt geht.

TvT73, 2024-06-28:
Um den Fehler einmal besser darzustellen, hier ein paar Screenshots dazu.
Villeicht liegt der Fehler ja doch nicht im WR. 
[image]
[image]

closed 2025-04-18 by github-actions[bot]