tbnobody/OpenDTU #2056: Kritisch: Setzen von Einstellungen über den Web-API unterbricht die Verbindung zum Wechselrichter

grasmax, 2024-06-11:
### What happened?

Dieser Post-Befehl s.u. unterbricht die Verbindung zum Wechselrichter dauerhaft. 
Die Funktion kann nur wiederhergestellt werden, in dem die vorher gerettete config.json wieder eingelesen und die Sendeleistung über den GUI gesetzt wird!

```
      ret = requests.get(
                  url = f'http://{dtu_ip}/api/dtu/config',
                  #data = f'data={{"serial":"{serial}", "limit_type":0, "limit_value":{setpoint}}}',
                  auth = HTTPBasicAuth(dtu_nutzer, dtu_passwort),
                  headers = {'Content-Type': 'application/x-www-form-urlencoded'}
              )
      jData = ret.json()

      ret = requests.post(
                  url = f'http://{dtu_ip}/api/dtu/config',
                  data = f'data={{"serial":"{serial}", "pollinterval":8, "nrf_enabled": "True", "nrf_palevel":0, "cmt_enabled": "True", 
                  "cmt_palevel":17,"cmt_frequency": 865000}}',
                  auth = HTTPBasicAuth(dtu_nutzer, dtu_passwort),
                  headers = {'Content-Type': 'application/x-www-form-urlencoded'}
              )

```

### To Reproduce Bug

config.json sichern
Script ausführen

### Expected Behavior

Script darf nicht die Funkstrecke zum Wechselrichter unterbrechen.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.1.14

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-06-11:
Das komplette Frontend verwendet nur und ausschließlich die API zum setzen der Einstellungen. Das funktioniert hier super. Wenn du die WebAPI verwenden möchtest (was ausdrücklich auf eigene Gefahr geschieht) musst du mit deinem Script exakt die Verhaltensweise der Webseite nachbilden.

Robert65779, 2024-06-12:
Hallo,

ich habe seit dem kurzem das gleiche Problem. Ca. 1x am Tag unterbricht die Verbindung zwischen OpenDTU und dem Wechselrichter. Auch scheint die Verbindung nicht ganz unterbrochen zu sein. Ich kann via OpenDTU Wechselrichter den Inverter noch ein- und ausschalten, sowie neu starten. Limits setzen und Daten empfangen geht nicht mehr.
Heilung bringt nur ein abschalten/trennen des Wechselrichters von der Batterie/Panels.
OpenDTU Fusion & HMS-2000
Normale Nutzung via Home Assistent

Auszug Konsole:
20:35:54.328 > RX Period End
20:35:54.328 > All missing
20:35:54.328 > Nothing received, resend count exeeded
20:35:54.435 > TX PowerControl 865.00 MHz --> 51 94 81 28 39 80 11 58 56 81 02 00 D0 00 99 
20:35:56.474 > RX Period End
20:35:56.474 > All missing
20:35:56.474 > Nothing received, resend whole request
20:35:56.474 > TX PowerControl 865.00 MHz --> 51 94 81 28 39 80 11 58 56 81 02 00 D0 00 99 
20:35:58.511 > RX Period End
20:35:58.511 > All missing
20:35:58.511 > Nothing received, resend whole request
20:35:58.511 > TX PowerControl 865.00 MHz --> 51 94 81 28 39 80 11 58 56 81 02 00 D0 00 99 
20:36:00.547 > RX Period End
20:36:00.547 > All missing
20:36:00.547 > Nothing received, resend whole request
20:36:00.547 > TX PowerControl 865.00 MHz --> 51 94 81 28 39 80 11 58 56 81 02 00 D0 00 99 
20:36:02.585 > RX Period End
20:36:02.585 > All missing
20:36:02.585 > Nothing received, resend whole request
20:36:02.585 > TX PowerControl 865.00 MHz --> 51 94 81 28 39 80 11 58 56 81 02 00 D0 00 99 
20:36:04.589 > RX Period End
20:36:04.589 > All missing
20:36:04.589 > Nothing received, resend count exeeded
20:36:04.589 > Fetch inverter: 116494812839
20:36:04.589 > Request SystemConfigPara
20:36:04.589 > Resend ActivePowerControl
20:36:04.589 > Resend PowerCommand
20:36:04.694 > TX ChannelChangeCommand 868.00 MHz --> 56 94 81 28 39 80 11 58 56 02 15 21 14 14 FB 
20:36:04.748 > RX Period End
20:36:04.748 > All missing
20:36:04.748 > Nothing received, resend whole request
20:36:04.748 > TX ChannelChangeCommand 868.00 MHz --> 56 94 81 28 39 80 11 58 56 02 15 21 14 14 FB 
20:36:04.800 > RX Period End
20:36:04.800 > All missing
20:36:04.800 > Nothing received, resend whole request
20:36:04.800 > TX ChannelChangeCommand 868.00 MHz --> 56 94 81 28 39 80 11 58 56 02 15 21 14 14 FB 
20:36:04.849 > RX Period End
20:36:04.849 > All missing
20:36:04.849 > Nothing received, resend whole request
20:36:04.849 > TX ChannelChangeCommand 868.00 MHz --> 56 94 81 28 39 80 11 58 56 02 15 21 14 14 FB 
20:36:04.932 > RX Period End
20:36:04.932 > All missing
20:36:04.932 > Nothing received, resend whole request
20:36:04.932 > TX ChannelChangeCommand 868.00 MHz --> 56 94 81 28 39 80 11 58 56 02 15 21 14 14 FB 
20:36:05.035 > RX Period End
20:36:05.035 > All missing
20:36:05.035 > Nothing received, resend count exeeded
20:36:05.082 > TX RealTimeRunData 865.00 MHz --> 15 94 81 28 39 80 11 58 56 80 0B 00 66 69 EA 94 00 00 00 00 00 00 00 00 6F 43 58 
20:36:05.492 > RX Period End
20:36:05.492 > All missing
20:36:05.492 > Nothing received, resend whole request
20:36:05.492 > TX RealTimeRunData 865.00 MHz --> 15 94 81 28 39 80 11 58 56 80 0B 00 66 69 EA 94 00 00 00 00 00 00 00 00 6F 43 58

stefan123t, 2024-07-10:
@grasmax Du sendest m.E. nur einen Teil der DTU Config, d.h. die PIN Settings sind nach Deinem POST Befehl nicht mehr aktiv.
Bitte mal in der GUI die Einstellungen anpassen und den POST mit den Developer Tools tracen, wie Thomas geschrieben hat.

---

@Robert65779 bei Dir kommt er irgendwo durcheinander. Es wäre gut das Console Log nicht nur im Fehlerfall sondern auch davor zu haben. Kannst Du die OpenDTU an die USB Serial Console anschließen und einen längeren Zeitraum loggen. Interessant ist vor allem der Übergang von erreichbar zu nicht mehr erreichbar.

```
20:36:04.589 > Fetch inverter: 116494812839
20:36:04.589 > Request SystemConfigPara
20:36:04.589 > Resend ActivePowerControl
20:36:04.589 > Resend PowerCommand
20:36:04.694 > TX ChannelChangeCommand 868.00 MHz --> 56 94 81 28 39 80 11 58 56 02 15 21 14 14 FB
```

Dieser Abschnitt ist seltsam, er versucht ein 0x56 MainCmd aber er beginnt mit dem zweiten Frame 0x02. 
@tbnobody Kannst Du Dir das erklären, wie das passieren könnte ?

Allgemein empfehle ich Dir ein eigenes Issue aufzumachen, da dies m.E. nichts mit dem von grasmax geschilderten Problem zu tun hat.

tbnobody, 2024-10-03:
```
data = f'data={{"serial":"{serial}", "pollinterval":8, "nrf_enabled": "True", "nrf_palevel":0, "cmt_enabled": "True", 
                  "cmt_palevel":17,"cmt_frequency": 865000}}',
```

Was für eine Serial setzt du hier ein?

grasmax, 2024-10-07:
Habe den kompletten Aufruf ganz oben in die Beschreibung gestellt....

tbnobody, 2024-10-07:
> Habe den kompletten Aufruf ganz oben in die Beschreibung gestellt....

Jedoch meine Frage nicht gelesen. Was ist '{serial}'?`

Des weiteren macht es keinen Sinn Limits an `http://{dtu_ip}/api/dtu/config` zu senden. Das ist der komplett falsche API Endpunkt.

grasmax, 2024-10-07:
Ja, sorry. serial ist die Seriennummer des Hoymiles-Wechselrichters. Ist das falsch?

Und danke für den Hinweis zur API.

Die Abfrage der Ertragsdaten funktioniert mit diesem Aufruf fehlerfrei:
r = requests.get(url = f'http://{dtu_ip}/api/livedata/status/inverters' ).json()

Habe ich für das Setzen des Limits einen falschen Aufruf aus einem frei verfügbaren Beispiel abgeschrieben?
url = f'http://{dtu_ip}/api/dtu/config',
Hast du funktionierenden Beispielcode, mit dem die Einstellungen gesetzt werden können?

tbnobody, 2024-10-07:
> Ja, sorry. serial ist die Seriennummer des Hoymiles-Wechselrichters. Ist das falsch?

Ja... du änderst in dem Moment die Seriennummer der DTU auf die gleiche wie die deines Inverters. Damit gibt es 2 gleiche Seriennummer in deiner Kommunikation. Das ist das gleiche als hättest du in einem Netzwerk 2x die gleiche IP bzw. 2x die gleiche MAC Adresse.

grasmax, 2024-10-07:
Vielen Dank an [tbnobody ](https://github.com/tbnobody) für das hartnäckige Dranbleiben und die richtige Lösung!

**Zusammenfassung**
Ziel war es, über ein Python-Script die OpenDTU-Sendeleistung zu setzen.
Das funktioniert auch, wenn man alle Parameter für den POST vorher mit GET /api/dtu/config abruft.
Und diese Parameter bei POST alle richtig auflistet.

Leider hatte ich übersehen, dass meine OpenDTU eine eigene Seriennummer hat. Und GENAU DIESE muss man beim Post angeben. Mein Fehler war, dass ich hier die Seriennummer des Hoymiles-Wechselrichters angegeben hatte,

```
    # u.a. Setzen der Sendeleistung auf 15 (dBm) und des Abfrageintervalls auf 30 Sekunden
    ret = requests.get(
                url = f'http://{dtu_ip}/api/dtu/config',
                auth = HTTPBasicAuth(dtu_nutzer, dtu_passwort),
                headers = {'Content-Type': 'application/x-www-form-urlencoded'}
            )
    jData = ret.json()
    serialOpenDtu = jData["serial"]

    ret = requests.post(
                url = f'http://{dtu_ip}/api/dtu/config',
                data = f'data={{"serial":"{serialOpenDtu}", "pollinterval":30, "nrf_enabled": "True", "nrf_palevel":0, "cmt_enabled": "True", "cmt_palevel":15,"cmt_frequency": 865000}}',
                auth = HTTPBasicAuth(dtu_nutzer, dtu_passwort),
                headers = {'Content-Type': 'application/x-www-form-urlencoded'}
            )

```

Das Ticket kann geschlossen werden.

closed 2024-10-07 by tbnobody

stefan123t, 2024-10-07:
@tbnobody sollen wir in der API Doku bzw in der Swagger Beschreibung #378 evtl zwischen InvSerial und DtuSerial unterscheiden ?

grasmax, 2024-10-08:
Gute Idee. Inverter vielleicht ausschreiben.