tbnobody/OpenDTU #2086: OpenDTU reports huge solar production spike at midnight if there has been production during the last hour of the day

GileanEU, 2024-06-22:
### What happened?

This time of the year, in Finland at least, my solar panels produce small amount of electricity also between 23:00-23:59. When that happens, OpenDTU reports huge solar production spike at 0:00 which causes a spike to my energy panel and I need to manually fix it. 
[image]

### To Reproduce Bug

Make OpenDTU to report solar energy between 23:00-23:59

### Expected Behavior

Some sort of solar production spike is going to be expected at 0:00 or 0:05.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v24.2.16

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-06-22:
This is not a bug but most likely a expected behavior. I would guess that you've enabled the [Zero daily yield at midnight](https://www.opendtu.solar/firmware/configuration/inverter_settings/#zero-daily-yield-at-midnight)
In your case you have to disable it.

Background: This function just sets OpenDTU internal yield to zero at midnight. When the next reading from the inverter occours the yield value begins at zero (as it was normally dark at night and the inverter was turned off -> reset). In your case the yield value of the last day occours after the zero value and counts up (which causes your huge peak).

GileanEU, 2024-06-23:
I had enabled "zero daily yield at midnight" so I disabled it yesterday from all my four micro inverters, but it didn't help. According to Home Assistant, the solar production between 23:00 and 0:00 was 0,01kWh, and here's this huge spike again (seems like the whole yesterday's yield):
[image]

tbnobody, 2024-06-23:
If just the 0,1kwh was sent from opendtu to the mart broker, then the problem is maybe home assistant or it's config. Many people using this inverter on a battery and this problem does not occlusion over night. You should log what value is sent really from opendtu.

GileanEU, 2024-06-23:
I am using MQTT sensor, and apparently yield_day value, but that's because I used DTU Pro modem earlier and yield_day value as well (there was never this behaviour with that integration) So because I didn't want to lose my energy panel history, I had to use  the same sensor name with OpenDTU as well.

But even with OpenDTU, I never get these "midnight spikes" unless there's some production recorded between 23:00-23:59. Naturally, for the most part of the year, production at that time of day is zero (and no spikes at midnight), especially if it's rainy/cloudy etc.

mbo18, 2024-06-29:
I used to have the same behavior. 
How many inveters do you have? If more than one, do you sum their total yield using home assistant sensor group?
If yes that’s why. At midnight, inveters total yield is updated to 0 but not at the same time.
The workaround is to use total yield published by OpenDTU

GileanEU, 2024-06-30:
@mbo18 
I have 4 inverters. A couple of days ago I  noticed that I also had another big spike at 3AM. And last night this happened as well:

[image]

And when I look at the sensor history, it shows like this:
[image]

I have this YAML code:

```
mqtt:
  sensor:
    - name: "Sähkön hetkellinen tuotto"
      state_topic: "solar/ac/power"
      unit_of_measurement: "W"
      device_class: "energy"
      state_class: "measurement"
    - name: "solarH_549330_today_eq"
      state_topic: "solar/ac/yieldday"
      unit_of_measurement: "kWh"
      value_template: "{{ (value | float / 1000) | round(3) }}"
      device_class: "energy"
      state_class: "total_increasing"
    - name: "Solar Yield Total"
      state_topic: "solar/ac/yieldtotal"
      unit_of_measurement: "kWh"
      device_class: "energy"
      state_class: "total"
```
So, I am using yieldday now. I have named the sensor with the same name I had earlier with my DTU Pro. So you suggest I am going to use yieldtotal instead? I would not want to lose my history though, so should I rename "Solar Yield Total" with "solarH_549330_today_eq", is it that easy? And naturally that another sensor with some other name. Naturally, that would cause some odd spikes when I do that but I guess it's fixable later...

mbo18, 2024-06-30:
I’m currently using the same config as you and for the energy dashboard I’m using the same sensor as "solarH_549330_today_eq"… but I don’t have the same behavior

GileanEU, 2024-07-01:
Good to hear that it works for you, but odd that I have this behavior. However, have you had situations where you have had solar production between 23:00-0:00 ? Because that's when this happens in my case (not sure though why sometimes I get this spike at 03:00 as well). I switched to OpenDTU last winter so this is the first summer with this configuration for me and it's  basically midsummer so... Didn't have this behaviour with DTU Pro / Hoymiles integration though during previous last two summers.

But now it was cloudy/rainy here so last night there was no production between 23-00 and therefore no spikes on my energy dashboard either.

closed 2025-04-18 by github-actions[bot]