tbnobody/OpenDTU #2103: Random Total Yield Reduce

math322, 2024-06-28:
### What happened?

The total yield value randomly reduced by big steps.

### To Reproduce Bug

-
[image]

### Expected Behavior

increasing by time

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v24.6.10

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

math322, 2024-08-16:
It seems this is a bug in the inverter firmware. If the 16 bit daily yield value overflows the total yield is reduced by this value.

stefan123t, 2024-08-17:
Is this `main.power.etq.battery.yield` a value from OpenDTU or OpenDTU-OnBattery or even something else ?

math322, 2024-08-18:
This value comes directly from opendtu/mqtt, i checked the raw data from the inverter and everything seems to be correct. To the background, the inverter is connected to a battery and is running 24/7.

stefan123t, 2024-09-30:
@math322 I still do not understand where this topic / value "battery" is generated in OpenDTU or do you have named your Inverter / String "battery" ?

Here is an example tree of the MQTT topics used by OpenDTU: https://github.com/tbnobody/OpenDTU/discussions/317#discussioncomment-4345649

I do not see how `main.power.et1.battery.yield` fits anywhere in this topic structure or "tree".

Could you please post a picture of your topic structure under/above `main.power.et1.battery.yield` using MQTT Explorer or something similar ?

mibcat, 2024-10-20:
Can confirm the reset behavior on my system:
using a Hoymiles HM-400 inverter which is permanently connected via a DC/DC converter to a battery.

Last night I got a drop of 65535 Wh which is a clear indication of a 16bit counter roll-over
[image]

and it was the first time the daily counter exceeds the 16bit threshold since I've started my project
[image]

Inverter settings:
[image]
[image]
[image]

stefan123t, 2024-10-20:
@mibcat i do understand why and that this happens on an PV inverter connected to a battery. Actually the PV inverter firmware is not expecting to run for more than a daylight‘s time.

You are using the PV inverter out of the manufacturer’s intended use case and the rollover of the daily yield occurs.

The daily yield will start at zero again, right ?
I can not really spot that on my mobile.

What happens to the yield total will it also be reduced by the 65636 Wh ?

mibcat, 2024-10-20:
> The daily yield will start at zero again, right ?

obviously not - OpenDTU shows a correct accumulated "daily" value (and it is not compensated by Home Assistant)
[image]

> What happens to the yield total will it also be reduced by the 65636 Wh ?

yes: from the HomeAssistant graph I get the numbers 216,65kWh - 151.11kWh = 65,54kWh

Did I get it right, that the rollover issue of the total yield comes from the inverter ? So the setting in OpenDTU would not help ?
[image]

stefan123t, 2024-10-20:
@mibcat thanks for your reply. I understand that the value for daily yield shown in the OpenDTU UI / Live view does still show the right value ?

As far as I recall this is calculated / corrected by the last daily yield in case the difference between the current value and previous value is negative / reset to zero during daytime. It is obviously working fine and will be reset at day start / midnight, according to the settings you have shown / marked.

Do I further understand it right that the value sent for daily yield to your MQTT broker / Home Assistant is not corrected as in the Ui ?

You kay your second answer confirms my hunch that also the total yield is wrong / offset by the error in the daily yield somehow.

Could you extract some RealTimeRunDebug command from the Serial / Web Console sonwe get the raw values sent / replied to OpenDTU at this point in time ? This may allow us to determine with which values the OpenDTU has to calculate and where things may be going wrong. Thanks!

mibcat, 2024-10-20:
> I understand that the value for daily yield shown in the OpenDTU UI / Live view does still show the right value ?

Yes, the daily value is correct - I have the same values on my MQTT broker, OpenDTU UI and Home Assistant UI.

> As far as I recall this is calculated / corrected by the last daily yield in case the difference between the current value and previous value is negative / reset to zero during daytime. It is [...]

I'm a little bit confused now when trying to interpret the setting. The daily yield is definitely not reset at midnight as you could see in my second screen shot of my first post. Here the time scale are months and the resets in the daily yield was caused by resets of OpenDTU due to firmware updates or others especially at the beginning of my project in June.

> Your second answer confirms my hunch that also the total yield is wrong / offset by the error in the daily yield somehow.

Yes, indeed the total yield value is now 65k lower ...

> Could you extract some RealTimeRunDebug command from the Serial / Web Console so we get the raw values sent / replied to OpenDTU at this point in time ? This may allow us to determine with which values the OpenDTU has to calculate and where things may be going wrong. Thanks!

Sure I can. But this will take some time I guess. Because I've designed my battery system only for low power the inverter's maximum power is limited to 200W (and the available battery capacity is only 2kWh). 

So the next rollover will take place in a month (65kWh/2kWh/d = 33d) ... :smile:

And btw many thanks for taking time on that issue ...

mibcat, 2024-10-20:
Ok - thanks for the clarification.

stefan123t, 2024-10-20:
@mibcat or in your case once every ~30 days

@tbnobody @schlimmchen should we anticipate this feature of the Hoymiles firmware or could we protect the total yield against such rollover issues on the OpenDTU software side ?

But then if the rollover of the daily yield takes place on the inverter side the raw total yield is impacted, hence we can not prevent it from happening again.

This somehow defeats the purpose of the total yield on the Hoymiles inverters if it does not increase if just running long enough for the daily yield rollover to reset the total yield. So actually a inverter running 24/7 has lower total yield in the long run !?

I believe I read some other reports of these odd resets / sawtooth graphs in the issues. Maybe we can close them as not a bug on OpenDTU side and add a disclaimer / note in the manual ?

closed 2025-04-18 by github-actions[bot]