tbnobody/OpenDTU #2108: HMS-800-2T-LV not connecting to OpenDTU Fusion

QTT-Marco, 2024-06-29:
### What happened?

I bought an openDTU Fusion from Amazon DE, shipped to Mexico. I've updated the Fusion with the latest version, set-up the WiFi and added both inverters but I am not getting any data. Changed the CMT2300A region to North America,verified power supply (currently using genuine apple charger), tried different cables, moved the Fusion closer to the panels (even did a test up on the roof) but nothing has worked. Both inverters have not connected a single time and I have confirmed that they are generating electricity (led on the back of inverters and with meter readings).

[image]
[image]
[image]
[image]
[image]

### To Reproduce Bug

Plug in OpenDTU Fusion.

### Expected Behavior

Being able to get inverter data.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.29

### Relevant log/trace output

```Shell
RX Period End
14:56:24.127 > All missing
14:56:24.127 > Nothing received, resend whole request
14:56:24.127 > TX PowerControl Channel: 3 --> 51 93 10 81 82 80 19 18 60 81 00 00 B0 01 00 
14:56:26.131 > RX Period End
14:56:26.131 > All missing
14:56:26.131 > Nothing received, resend count exeeded
14:56:26.131 > Fetch inverter: 114393107954
14:56:26.131 > Request SystemConfigPara
14:56:26.182 > TX RealTimeRunData Channel: 23 --> 15 93 10 79 54 80 19 18 60 80 0B 00 66 80 74 FA 00 00 00 00 00 00 00 00 DA B9 DA 
14:56:26.713 > RX Period End
14:56:26.713 > All missing
14:56:26.713 > Nothing received, resend whole request
14:56:26.713 > TX RealTimeRunData Channel: 40 --> 15 93 10 79 54 80 19 18 60 80 0B 00 66 80 74 FA 00 00 00 00 00 00 00 00 DA B9 DA 
14:56:27.255 > RX Period End
14:56:27.255 > All missing
14:56:27.255 > Nothing received, resend whole request
14:56:27.255 > TX RealTimeRunData Channel: 61 --> 15 93 10 79 54 80 19 18 60 80 0B 00 66 80 74 FA 00 00 00 00 00 00 00 00 DA B9 DA 
14:56:27.793 > RX Period End
14:56:27.793 > All missing
14:56:27.793 > Nothing received, resend whole request
14:56:27.793 > TX RealTimeRunData Channel: 75 --> 15 93 10 79 54 80 19 18 60 80 0B 00 66 80 74 FA 00 00 00 00 00 00 00 00 DA B9 DA 
14:56:28.333 > RX Period End
14:56:28.333 > All missing
14:56:28.333 > Nothing received, resend whole request
14:56:28.333 > TX RealTimeRunData Channel: 3 --> 15 93 10 79 54 80 19 18 60 80 0B 00 66 80 74 FA 00 00 00 00 00 00 00 00 DA B9 DA 
14:56:28.835 > RX Period End
14:56:28.836 > All missing
14:56:28.836 > Nothing received, resend count exeeded
14:56:28.914 > TX AlarmData Channel: 23 --> 15 93 10 79 54 80 19 18 60 80 11 00 66 80 74 FA 00 00 00 00 00 00 00 00 00 A2 01 
14:56:29.669 > RX Period End
14:56:29.669 > All missing
14:56:29.669 > Nothing received, resend whole request
14:56:29.669 > TX AlarmData Channel: 40 --> 15 93 10 79 54 80 19 18 60 80 11 00 66 80 74 FA 00 00 00 00 00 00 00 00 00 A2 01 
14:56:30.459 > RX Period End
14:56:30.459 > All missing
14:56:30.459 > Nothing received, resend whole request
14:56:30.459 > TX AlarmData Channel: 61 --> 15 93 10 79 54 80 19 18 60 80 11 00 66 80 74 FA 00 00 00 00 00 00 00 00 00 A2 01 
14:56:31.250 > RX Period End
14:56:31.250 > All missing
14:56:31.250 > Nothing received, resend whole request
14:56:31.250 > TX AlarmData Channel: 75 --> 15 93 10 79 54 80 19 18 60 80 11 00 66 80 74 FA 00 00 00 00 00 00 00 00 00 A2 01 
14:56:32.041 > RX Period End
14:56:32.041 > All missing
14:56:32.041 > Nothing received, resend whole request
14:56:32.041 > TX AlarmData Channel: 3 --> 15 93 10 79 54 80 19 18 60 80 11 00 66 80 74 FA 00 00 00 00 00 00 00 00 00 A2 01 
14:56:32.801 > RX Period End
14:56:32.801 > All missing
14:56:32.801 > Nothing received, resend count exeeded
14:56:32.861 > TX SystemConfigPara Channel: 23 --> 15 93 10 79 54 80 19 18 60 80 05 00 66 80 74 FA 00 00 00 00 00 00 00 00 14 B6 15 
14:56:33.078 > RX Period End
14:56:33.078 > All missing
14:56:33.078 > Nothing received, resend whole request
14:56:33.078 > TX SystemConfigPara Channel: 40 --> 15 93 10 79 54 80 19 18 60 80 05 00 66 80 74 FA 00 00 00 00 00 00 00 00 14 B6 15 
14:56:33.318 > RX Period End
14:56:33.318 > All missing
14:56:33.318 > Nothing received, resend whole request
14:56:33.318 > TX SystemConfigPara Channel: 61 --> 15 93 10 79 54 80 19 18 60 80 05 00 66 80 74 FA 00 00 00 00 00 00 00 00 14 B6 15 
14:56:33.558 > RX Period End
14:56:33.558 > All missing
14:56:33.558 > Nothing received, resend whole request
14:56:33.558 > TX SystemConfigPara Channel: 75 --> 15 93 10 79 54 80 19 18 60 80 05 00 66 80 74 FA 00 00 00 00 00 00 00 00 14 B6 15 
14:56:33.798 > RX Period End
14:56:33.798 > All missing
14:56:33.798 > Nothing received, resend whole request
14:56:33.798 > TX SystemConfigPara Channel: 3 --> 15 93 10 79 54 80 19 18 60 80 05 00 66 80 74 FA 00 00 00 00 00 00 00 00 14 B6 15 
14:56:34.001 > RX Period End
14:56:34.001 > All missing
14:56:34.001 > Nothing received, resend count exeeded
14:56:34.001 > Fetch inverter: 114393108182
14:56:34.001 > Request SystemConfigPara
14:56:34.001 > Resend PowerCommand
14:56:34.054 > TX RealTimeRunData Channel: 23 --> 15 93 10 81 82 80 19 18 60 80 0B 00 66 80 75 02 00 00 00 00 00 00 00 00 DB 8B 3E 
14:56:34.586 > RX Period End
14:56:34.586 > All missing
14:56:34.586 > Nothing received, resend whole request
14:56:34.586 > TX RealTimeRunData Channel: 40 --> 15 93 10 81 82 80 19 18 60 80 0B 00 66 80 75 02 00 00 00 00 00 00 00 00 DB 8B 3E 
14:56:35.126 > RX Period End
14:56:35.126 > All missing
14:56:35.126 > Nothing received, resend whole request
14:56:35.126 > TX RealTimeRunData Channel: 61 --> 15 93 10 81 82 80 19 18 60 80 0B 00 66 80 75 02 00 00 00 00 00 00 00 00 DB 8B 3E 
14:56:35.667 > RX Period End
14:56:35.667 > All missing
14:56:35.667 > Nothing received, resend whole request
14:56:35.667 > TX RealTimeRunData Channel: 75 --> 15 93 10 81 82 80 19 18 60 80 0B 00 66 80 75 02 00 00 00 00 00 00 00 00 DB 8B 3E 
14:56:36.214 > RX Period End
14:56:36.214 > All missing
14:56:36.214 > Nothing received, resend whole request
14:56:36.214 > TX RealTimeRunData Channel: 3 --> 15 93 10 81 82 80 19 18 60 80 0B 00 66 80 75 02 00 00 00 00 00 00 00 00 DB 8B 3E 
14:56:36.708 > RX Period End
14:56:36.708 > All missing
14:56:36.708 > Nothing received, resend count exeeded
14:56:36.800 > TX AlarmData Channel: 23 --> 15 93 10 81 82 80 19 18 60 80 11 00 66 80 75 02 00 00 00 00 00 00 00 00 01 90 E5 
14:56:37.541 > RX Period End
14:56:37.541 > All missing
14:56:37.541 > Nothing received, resend whole request
14:56:37.541 > TX AlarmData Channel: 40 --> 15 93 10 81 82 80 19 18 60 80 11 00 66 80 75 02 00 00 00 00 00 00 00 00 01 90 E5
```

### Anything else?

I've also requested help on Discord but after troubleshooting, I've had no success.

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

schneeer, 2024-06-30:
Hi,
have you actually flashed the latest firmware onto the DTU?
That would be: [24.6.29](https://github.com/tbnobody/OpenDTU/releases/tag/v24.6.29)

The title of your error message is:
"HMS-800-2T-LV not connecting to OpenDTU Fusion"
However, the DTU does not recognize an HMS-800-2T based on your serial numbers, but rather: HM ... 2T

It could be that you are not using the latest firmware, or the serial numbers of your inverters are not yet supported.
According to the [class overview](https://github.com/tbnobody/OpenDTU/tree/master/lib/Hoymiles/src/inverters#class-overview), the serial numbers should start with 1144....
Your serial numbers start with 1143...

Your inverters are suitable for a mains voltage of 120-127 V. In Europe, 230 V is common.
Inverters like yours may not yet be supported in the openDTU firmware.

QTT-Marco, 2024-06-30:
> Hi, have you actually flashed the latest firmware onto the DTU? That would be: [24.6.29](https://github.com/tbnobody/OpenDTU/releases/tag/v24.6.29)
> 
> The title of your error message is: "HMS-800-2T-LV not connecting to OpenDTU Fusion" However, the DTU does not recognize an HMS-800-2T based on your serial numbers, but rather: HM ... 2T
> 
> It could be that you are not using the latest firmware, or the serial numbers of your inverters are not yet supported. According to the [class overview](https://github.com/tbnobody/OpenDTU/tree/master/lib/Hoymiles/src/inverters#class-overview), the serial numbers should start with 1144.... Your serial numbers start with 1143...
> 
> Your inverters are suitable for a mains voltage of 120-127 V. In Europe, 230 V is common. Inverters like yours may not yet be supported in the openDTU firmware.
> 
> You have the complete serial numbers here. This is a potential risk. Better to make the last digits unrecognizable.

Thank you, I've replaced the photos with the full inverter number.
Yes, I can confirm it is the latest firmware
[image]

I believe the issue is that my inverters are for 120-127 V based on your comment of the serial first 4 digits and that they are recognized as HM-2T on the list instead of HMS-2T. 
Is there any way to confirm if this is the issue?
Is there any plans on adding support to LV models? Any way that I can assist with this?
I would also like to request if this information regarding HMS-2T-LV models is added to the project so that people don't accidentally buy or build an OpenDTU Fusion thinking it will work with their LV models, if possible.

tbnobody, 2024-06-30:
[generic_esp32s3_usb.zip](https://github.com/user-attachments/files/16045434/generic_esp32s3_usb.zip)
Could you please try this firmware version? 
Using this version your inverter should be detected as `HMS-600/700/800/900/1000-2T`.
Then lets see whether data is received and if it is interpreted correctly.

QTT-Marco, 2024-06-30:
> [generic_esp32s3_usb.zip](https://github.com/user-attachments/files/16045434/generic_esp32s3_usb.zip) Could you please try this firmware version? Using this version your inverter should be detected [...]

Thank you, it did work. I do need to actually compare the data obtained with my panel information, meter readings etc. after a few days, but I can confirm that information is being received now and it does seem okay.
[image]

tbnobody, 2024-06-30:
Thanks for the reply. I will add this patch to the new version as soon as you confirm that everything is working correctly (did you also check the eventlog etc?)
But you are right, the values in the screen shot look very realistic.

QTT-Marco, 2024-06-30:
Here is a picture of the eventlog. I'll be replying back here in a few days to let you know if I found any inconsistencies in the information or to confirm that everything seems fine. Thank you for your help and please let me know if there's any other kind of tests or information that you need from my end.
[image]

QTT-Marco, 2024-07-01:
I can confirm everything's working fine, the only thing missing is this info:
[image]

tbnobody, 2024-08-01:
Implemented in d09be3a3841ec7cf49c6d8f8cbaa06f43e56efb3

closed 2024-08-01 by tbnobody