tbnobody/OpenDTU #2110: No secure connection to MQTT broker in Azure Event Grid

wohlfrom, 2024-06-30:
### What happened?

When configuring secure TLS MQTT connection to [Azure Event Grid](https://learn.microsoft.com/en-us/azure/event-grid/mqtt-transport-layer-security-flow) the ESP crashed during connection and enters a reboot loop until MQTT is disabled in OpenDTU again.

### To Reproduce Bug

Create a MQTT broker in Azure Event Grid following [this tutorial](https://learn.microsoft.com/en-us/azure/event-grid/mqtt-publish-and-subscribe-portal)
Configure OpenDTU with the server `*.eventgrid.azure.net`
After saving the configuration and first attempt is made to connect to the MQTT broker the crash occurs.

### Expected Behavior

Succesfull connection or failure message that no successful connection was made.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.10

### Relevant log/trace output

```Shell
15:38:01.346 > [  6449][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 236760
15:38:01.355 > [  6457][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
15:38:01.387 > [  6493][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
15:38:01.399 > [  6502][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
15:38:01.405 > [  6510][V][ssl_client.cpp:178] start_ssl_client(): Loading CA cert
15:38:01.418 > [  6521][V][ssl_client.cpp:234] start_ssl_client(): Loading CRT cert
15:38:01.419 > [  6530][V][ssl_client.cpp:243] start_ssl_client(): Loading private key
15:38:01.436 > [  6541][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...

15:38:01.436 > [  6550][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...

15:38:01.806 > Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 

15:38:01.814 > Debug exception reason: Stack canary watchpoint triggered (mqttclient)
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

fawick, 2024-07-27:
This is not limited to Azure in any way. I ran into the same problem when using a private MQTT server encrypted with TLS.

stefan123t, 2024-07-28:
@fawick can you try the patch @wohlfrom added in #2109 ?
Actually it increases the mqtt buffer / stack size from 5120 to 6400 bytes which fixes the issue for him and @scorpi11

fawick, 2024-07-29:
I'm afraid I cannot, at least not on a short time-frame. For the time being, I am using the released binaries and have no build environment for OpenDTU. 

I've been hoping, once it is known this problem indeed affects more users and is not limited to a single MQTT broker, the OpenDTU maintainers might consider replicating the problem and decide from there whether increasing the default stack size would be acceptable.

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.

I'd like to propose to retitle the issue to `Crash upon establishing TLS-secured MQTT connection`

PR #2109 (2024-08-01): Support MQTT CA certificate validation for Azure Event Grid
The use case is to publish MQTT events to the [Azure Event Grid broker](https://learn.microsoft.com/en-us/azure/event-grid/mqtt-overview). You will get a MQTT server url in the form of `*.germanywestcentral-1.ts.eventgrid.azure.net:8883`. Only secure TLS connections [are supported](https://learn.microsoft.com/en-us/azure/event-grid/mqtt-transport-layer-security-flow).

During the handshake the ESP produces are core dump and enters a reboot loop since the stack limit is reached for the MQTT task that is used for communicating to the br [... cut]
closed 2024-08-01 by tbnobody