tbnobody/OpenDTU #2113: Firmware Update v24.6.29 stucks at 100% upload

HebeROF, 2024-06-30:
### What happened?

I am on:

OpenDTU-B6B170
v4.4.7-dirty
0.1.28
v24.6.10
generic

a new firmware v24.6.29 is available. As always, I downloaded the opendtu-generic.bin (https://github.com/tbnobody/OpenDTU/releases/download/v24.6.29/opendtu-generic.bin)

But the upload stucks at 100% and is not processed (no restart happens).

A manual restart works, but shows that the old version is still installed. 

Also redownloaded the new firmware several times. 

### To Reproduce Bug

Upload the new Version directly via /firmware/upgrade

### Expected Behavior

Once the upload is complete, it is processed and the system restarts with the new firmware. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.10

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-06-30:
Sometimes the ESP does such things... You have two possibilites:
1. Power cycle the ESP before the upgrade
2. Reflash the factory firmware using USB (make sure to backup all config files before)

HebeROF, 2024-06-30:
Thanks for the quick reply. 

Alternative 1 actually worked. De-energized for 20 seconds and then it ran through. I'll keep that in mind! I've actually never had this before.

closed 2024-06-30 by HebeROF