tbnobody/OpenDTU #2126: smart micro solar lose conection to openDTU 3 or 4 times a day

Tom-Solar, 2024-07-07:
### What happened?

smart micro say there is a conection to dtu but the parameter do not arrive in dtu.

### To Reproduce Bug

i just start both

### Expected Behavior

smart micro send values to dtu but dtu stays at the value to the time the connection get lost.
1 x hoymiles hm 800 and 1 x hoymiles hm1500. sometimes hm1500 is in funktion and hm800 akt with one value the hole time. i have to restart both then it will go on normal til it wil loose conection again. ( i‘m sorry about my english)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.10

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

taneltumanski, 2024-07-09:
Same here, connection to the inverter drops several times per day, sometimes for a long time.
I have a feeling it is related to the inverter temperature but no data to verify.

Home assistant logs:
[image]

stefan123t, 2024-07-09:
@Tom-Solar I am sorry for you english too :)

It is normal for the inverter to be unreachable during night times. As the whole inverter is powered only by the DC / PV panels commected to it.

Regarding the smaller outages during daytime this could be connection problems. You are using HM models which can be reached via NRF24L01+ Modules using the so called Enhanced Shockburst protocol by Nordic Semiconductors. This is usually only for short distance communication like Bluetooth mice/headphones. Ie the link relies heavily on your hardware setup, a stable power/voltage source and also what you send to the inverter. It has been seen that the inverter may seize responding if you send it repeatedly or too frequently (the same) Active Power Limit commands or query it too frequently.
However OpenDTU eg on the OpenDTU fusion board does work very well during normal (heat) conditions.

taneltumanski, 2024-07-09:
@stefan123t Current setup:
HMS-2000-4T
Blinkyparts OpenDTU with CMT module, powered from PC USB port.
Distance to inverter ~6m with sheet metal roof between controller and inverter. Inverter near edge of the roof not center.
No power limit messages sent to inverter.
Poll interval set to 5s in OpenDTU.

In mornings and evenings the connection is stable, most outages happen during the afternoon at the hottest parts of the day.
I will continue to monitor the outages but I doubt it is an OpenDTU issue.
Would increasing the "CMT2300A Transmitting power" maybe solve the problem?

stefan123t, 2024-07-09:
@taneltumanski can you show a picture or two of your setup. I recall that the BlinkyParts CMT Breakout board has some tiny weeny pins to solder.
Is your Power Source stable enough ?
Did you consider adding some capacitors for buffering the voltage due to sending data ?
Which schematic did you follow ?
What Antenna is used and is the connector fitting the plug ?
Some are missing the male pin in between plug and antenna which can cause difficulties.
Given the fact that you are under a tin roof, did you consider placing the OpenDTU somewhere else where the sheet metal does not obstruct your reception ?
We still have no RSSI or packet loss data from the inverter afaik.

lehmartin, 2024-08-29:
Maybe same issue as [here](https://github.com/tbnobody/OpenDTU/issues/2137)? As I mentioned in the corresponding thread, changing the frequency "solves" the issue. Happens several times day on my side.

stefan123t, 2024-08-30:
@lehmartin no as explained elsewhere the HM models NRF communication chip is distinct from CMT communication chip in HMS/HMT models. Frequency hopping is used in CMT and your situation, whereas @Tom-Solar has three HM model inverters and needs to respond to this Issue in order to answer our questions and progress this issue.

closed 2025-04-18 by github-actions[bot]