tbnobody/OpenDTU #2133: Faulty Sketch size reported in Info Web page

Web-svg, 2024-07-09:
### What happened?

Hello,
Using Olimex POE card (16MB) and CMT2300A running with firmware V24.6.29 I'm facing what it could be considered as a kind of typo issue:
Info page part related to memories space usage report inconsistent value for sketch (see picture) in all 3 languages.

### To Reproduce Bug

Go to Webapp and Info/system
[image]

### Expected Behavior

Just showing appropriate result of calculation

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V24.6.29

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-07-09:
What exactly is inconsistent?

Web-svg, 2024-07-09:
Hello,
Percentage vs memory size i assumeDoes I'm wrong?
Kind regards

tbnobody, 2024-07-09:
1630 / 1792 = 91%
Seems right for me

Web-svg, 2024-07-09:
I understood my misunderstanding suddendly....sorry for that.
Could you delete it?

closed 2024-07-09 by tbnobody