tbnobody/OpenDTU #2137: Lost connection to hms-1600-4t / frequency hopping

Paule-Panter, 2024-07-11:
### What happened?

Sporadically the dtu lost the connection between dtu and inverter. 
Then the inverter is showing red in dtu.
Next day (after dc cut off) the connection will reestablished. 

A other way i found, is to look at a spectrum analyser. There are some periodicaly burst at a nearby frequency visible. If i tune the DTU to this frequency (e.g. 865.25MHz) the dtu gets immediately actual data.
After a while (some hours) the inverter switches (i dont know why) the frequency to e.g. 865.75MHz. Then the dtu cant receive any data from the inverter. 
If I retuned the dtu to the frequency, where some burst visibility from the inverter, then it get immediately reconnected and gets actuall data.
At the Console i could see, that the dtu sometimes try to switch to 868.00MHz.
But at the spectrum analyser i see the burst at other frequencies.eg 865.25 / 865.75Mhz

Maybe the HMS1600-4t look for a undisturbed channel and switch to that channel.
Is there a well known procedure for a channel switch ? (initiated from both side)
Or is a automaticly frequency scanning available, that could be retriggered after a connection lost to the inverter?

### To Reproduce Bug

Switch on a hms1600-4t.
Wait until the connection to the inverter is lost. 
Look at a external spectrumAnalyser for a peridically pattern.
Tune to this frequency.
Get fresh inverter data.(successful reconnect)

### Expected Behavior

No lost connection /
Automatic reconnect.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

git Hash	v24.6.29

### Relevant log/trace output

```Shell
Fetch inverter: 1164945xxxxx
21:23:19.282 > TX RealTimeRunData 865.75 MHz --> 15 94 50 34 70 80 18 28 29 80 0B 00 66 90 31 27 00 00 00 00 00 00 00 00 8B 3A D6 
21:23:19.489 > Interrupt received
21:23:19.686 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 01 00 01 01 10 01 09 00 05 00 06 00 10 00 10 00 00 96 | -12 dBm
21:23:19.942 > Interrupt received
21:23:19.956 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 02 3C C9 00 00 3C 01 08 4A 08 4A 00 FE 01 01 00 06 BE | -12 dBm
21:23:19.965 > Interrupt received
21:23:19.975 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 03 00 06 00 10 00 10 00 00 3C 74 00 00 3B 76 08 4B CF | -12 dBm
21:23:19.987 > Interrupt received
21:23:19.992 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 04 08 38 09 56 13 89 00 3D 00 00 00 02 03 E7 01 4F E8 | -12 dBm
21:23:20.001 > Interrupt received
21:23:20.008 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 85 00 01 3E 76 40 | -12 dBm
21:23:20.017 > RX Period End
21:23:20.017 > Success
21:23:29.023 > Fetch inverter: 1164945xxxxx
21:23:29.036 > TX RealTimeRunData 865.75 MHz --> 15 94 50 34 70 80 18 28 29 80 0B 00 66 90 31 31 00 00 00 00 00 00 00 00 EB DC 46 
21:23:29.078 > Interrupt received
21:23:29.087 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 01 00 01 01 10 01 09 00 05 00 06 00 10 00 10 00 00 96 | -12 dBm
21:23:29.130 > Interrupt received
21:23:29.138 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 02 3C C9 00 00 3C 01 08 4A 08 4A 00 FC 01 01 00 06 BC | -13 dBm
21:23:29.180 > Interrupt received
21:23:29.190 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 03 00 06 00 0F 00 0F 00 00 3C 74 00 00 3B 76 08 4B CF | -12 dBm
21:23:29.229 > Interrupt received
21:23:29.238 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 04 08 38 09 54 13 8A 00 3D 00 00 00 02 03 E8 01 4F E6 | -12 dBm
21:23:29.269 > Interrupt received
21:23:29.278 > RX 865.75 MHz --> 95 94 50 34 70 80 18 28 29 85 00 01 3C 87 B3 | -12 dBm
21:23:29.622 > RX Period End
21:23:29.622 > Success
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

btuerk89, 2024-07-12:
Hi, 

i can confirm this bug or maybe expected behavior which is not considered in OpenDTU. 

What happend at my side: 
Connection gets sometimes lost since a few days ago. (Not sure if it since the beginning I installed my inverter begin of July.)
Restart of inverter (over night or manual disconnetion of modules) solved the problem.

Today connection loss happened again and checked for issues here.
Test with changing the frequency to a neigbour band solved the problem immediately. (in my case from 864.00 to 873.75 MHz)

BR

stefan123t, 2024-07-12:
Do you both have fairly new HMS models ?

Can you post the first seven digits of your serial number (first 4 are the model, then follow three digits one for year and two for calendar week). The remaining five are the weekly unit production number which is unique.

Could this be a new firmware on the imverter side ?

I read a report that the german Netzbehörde has requested some changes to the RF protocol. Maybe this is to avoid sticking to one channel for too long.

btuerk89, 2024-07-12:
Seriennummer	1164948xxxxx
Produktionsjahr	2023
Produktionswoche	48
Modell	HMS-2000-4T
Ermittelte max. Leistung	2.000 W
Bootloader-Version	0.1.1
Firmware-Version	1.0.27
Firmware-Erstellungsdatum	2023-06-05 10:24:00
Hardware-Teilenummer	270692628
Hardware-Version	01.10

Paule-Panter, 2024-07-12:
Seriennummer	1164945xxxxx
Produktionsjahr	2023
Produktionswoche	45
Modell	HMS-1600-4T
Ermittelte max. Leistung	1.600 W
Bootloader-Version	0.1.1
Firmware-Version	1.0.27
Firmware-Erstellungsdatum	2023-06-05 10:24:00
Hardware-Teilenummer	270680326
Hardware-Version	01.10

broth-itk, 2024-07-13:
To share my findings:

Today I had the same problem with my HMS-1800-4T, Firmware 1.0.27

It was shown offline/red but did produce power and was blinking green.

Eventually I got it back online when disconnecting OpenDTU from power, powering up the Hoymiles DTU and waiting a bit.
First it showed "Alarm" state in S-Miles cloud but then went green shortly after.

Now with Hoymiles DTU disconnected and OpenDTU back online, I wait for the inverters to answer ;-)

Paule-Panter, 2024-07-13:
Today i get the failure again.
I fetch some logs and screenshots from the spectrum after the frequency changed already.
See attached logfile with some additional comments, what i have done at some timestamp.

For me it looks like the openDTU sends some
>TX ChannelChangeCommand 868.00 MHz (hopefully with the correct payload to the selected frequency)
commands, but there is no reaction from the inverter. (Maybe because the inverter is tuned to a other frequency?)
The inverter ignores the commands and continues to sending at his selfselected frequency (865.5MHz in this case)

I then set the frequency seen at the spectrum in the openDTU menu and the current values were immediately available again

[20240713_openDTU_console.log](https://github.com/user-attachments/files/16202129/20240713_openDTU_console.log)

OpenDTU expected @ 865.75MHz / inverter sends @ 865.5MHz
[image]

OpenDTU send channelSwitching Commands @ 868MHz, 
but inverter ignores them?
[image]

set frequency in OpenDTU @ 865.5 MHz, gets immediately connected to inverter and fetch actuall data.
[image]

Paule-Panter, 2024-07-13:
I have just seen again that the inverter has changed frequency.
I set openDTU to the frequency recognised in the spectrum (this time 866 MHz) and the connection is up and running again.
[image]

Seppl01, 2024-07-15:
Same issue here:
2x HMS-1600-4T
1164a00xxxxxx

bytecorner-jan, 2024-07-19:
I do have the same issue, but my setup has one HMS-2000-4T and one HMS800-2T. 
The HMS800 does always reconnect in the morning. HMS2000 is the only one that is controlled (i.e. commands send to) 
I noticed that the opendtu is sending commands for every received mqtt message also when the inverter is not sending anything. 

The last days I disabled in the evening mqtt command sending, and did not have any issues reconnecting in the morning. 

I would suggest only sending commands when the inverter is "online".

stefan123t, 2024-07-19:
> I would suggest only sending commands when the inverter is "online".

This is a known issue the command pipeline is not very long, but if you queue ActivePowerLimit commands this may actually prevent your OpenDTU to make normal contact / send queries to your inverter, but instead will try to set the limit first.

Also setting the limit too frequently can prevent the inverter from responding to any other requests for some 15 or even more seconds.
So do not set a new limit too frequently or when it ain‘t necessary. 

The Dynamic Power Limiter in OpenDTU-OnBattery has a hysteresis set to +/-30W to make it nly reasonable updates.

stefan123t, 2024-07-31:
@Paule-Panter @btuerk89 @spcqike thanks for your details
Especially the Firmware Version and Firmware Build date may give us some additional hints.

@Seppl01 @bytecorner-jan @broth-itk can you please provide the first seven digits of your Serial Number plus the requested details from the Inverter Info ?

@Paule-Panter @spcqike IMHO when you change / save the settings the OpenDTU code runs a reinitialisation of several components. Hence this would explain why it is able to sync to the new frequency again.

broth-itk, 2024-07-31:
@stefan123t 

Here is the information from my three inverters.
With the one in bold I observed the connectivity issues.

Today I had the same problem again, out of the blue not available.
Frequency was set to 865MHz. Reboot of DTU did not help.
After changing to 868MHz and wit ~15 minutes all inverters, inclusing the "missing one", were found again.

Serial	1164807xxxxx
Production Year	2022
Production Week	7
Model	HMS-1800-4T
Detected max. Power	1,800 W
Bootloader Version	0.0.1
Firmware Version	1.0.27
Firmware Build Date	2023-06-05 10:24:00
Hardware Part Number	269635841
Hardware Version	01.00

**Serial	1164813xxxxx
Production Year	2022
Production Week	13
Model	HMS-2000-4T
Detected max. Power	2,000 W
Bootloader Version	0.0.1
Firmware Version	1.0.27
Firmware Build Date	2023-06-05 10:24:00
Hardware Part Number	269644033
Hardware Version	01.00**

Serial	1124831xxxxx
Production Year	2022
Production Week	31
Model	HMS-350-1T
Detected max. Power	350 W
Bootloader Version	0.1.2
Firmware Version	1.0.14
Firmware Build Date	2021-12-09 12:46:00
Hardware Part Number	270541056
Hardware Version	02.00

stefan123t, 2024-07-31:
Thanks Bernhard @broth-itk for looking this up!

I am a bit tricked because you mentioned the 15 minutes wait time before the inverter(s) responded again.

Please refer to the [SystemConfigParam](https://github.com/lumapu/ahoy/wiki/Protocol#systemconfigpara--0x05) description of the protocol. We use it to verify the ActivePowerLimit has been set correctly. It is quite timing sensitive and may render an inverter unresponsive for ~15 minutes!

Do you have a zero export or similar ActivePowerLimit control for your inverters and if for which ?

broth-itk, 2024-08-01:
@stefan123t 

The 15 minutes are an estimation and are unrelated to zero export or ActivePowerLimit.
Maybe due to the documentation mentioning the time until the inverters are using the new frequency.

What about other inverters in the neighborhood? There are possibly some.
Can they interfere with our communication in a way that one unit gets lost?

Are thoses interferences detectable?

I think about my own products when we were implementing communication statistics per unit, like how many messages were sent, how many repetitions and how many failures (no reponse) happened.
These stats help actually a lot to evaluate the EF environment.

stefan123t, 2024-08-01:
@broth-itk The RX/TX Packet Loss counters of your Inverter can be read using the [GetLossRate](https://github.com/lumapu/ahoy/wiki/Protocol#getlossrate--0x15) command, that is probably what you are referring to.
The documentation in the Protocol Wiki is based on analysing the leaked Hoymiles DTU source code on gitee.com, traces created by Ahoy and OpenDTU users and observations during reverse engineering the protocol. The 15 minutes dead / no answer time were actually witnessed by both lumapu and tbnobody during development of the ActivePowerLimit command and subsequent SystemConfigParam query.

broth-itk, 2024-08-01:
@stefan123t 

I was thinking of counters on the OpenDTU side.
This would help judging the link quality between DTU and the inverters.
RSSI values would help as well if NRF/CMT support that.

stefan123t, 2024-08-01:
RSSI values are only supported by the CMT link. The NRF link (chip and library) only has pseudo RSSI, i.e. < or >= 64dBi as a Boolean.

Yes you could create counters on the OpenDTU/Ahoy side as well and even compare them with the counters on the inverter side after executing the GetLossRate command mentioned above.
That way you could calculate a differential value between consecutive queries or comparisons, eg based on a certain interval / number of packets exchanged.

fir3bird, 2024-08-04:
Unfortunately it seems like I also face this issue. I cannot get my OpenDTU connected to my Hoymiles HMS-1600-4T.
Manually tried several difference frequencies and never had success. The status information at least shows a successfully connected CMT2300 module. 

Since I do not have access to a spectrum analyzer I cannot tell if the frequencies I have tested correspond to my inverter. 
Maybe a "auto" frequency scanner might be a solution? Like scanning for radio stations in a car. Sorry, might be a stupid idea - I am not that introduced into RF applications. :-)

stefan123t, 2024-08-04:
@fir3bird the DTU can only connect to HMS-1600 not HMS-1600**W** can you double check the HMS Model you have and report the first seven digits of your HMS, ie Model and built year / calendar week ?

fir3bird, 2024-08-04:
@stefan123t I can confirm, that it is the model without *W*.
First seven digits of the serial number are the same like the ones from Seppl01: 1164a00xxxxxx

RS2480, 2024-08-06:
@fir3bird How did you coonect the cmt2300 and what did u write in the mapping file? I am also facing a issue .
HMS350 holymiles with opendtu, esp32,cmt2300 . Hello everyone, I need a help i have used HMS350 to use with opendtu with esp-wroom-32 and cmt 2300. I followed the steps on the opendtu documentation but I am not getting the live data of the inverter on the opendtu platform.

my pin mapping file looks like this

[
{

"name": "Generic NodeMCU 38 pin",
"Cmt": {
"clk": 12,
"cs": 27,
"fcs": 26,
"sdio": 14,
"gpio2": -1,
"gpio3": -1
},

"eth": {
"enabled": false,
"phy_addr": -1,
"power": -1,
"mdc": -1,
"mdio": -1,
"type": -1,
"clk_mode": -1
}
}
]

stefan123t, 2024-08-06:
@RS2480 you have not connected the GPIO2/3. I believe either is required, i just don’t know which 🤷 But your first time setup question is probably not related to this issue posted by @Paule-Panter please consider to use the Discord chat instead for your problem.

@tbnobody regarding the original issue of @Paule-Panter it might be good to start a frequency sweep across all possible CMT2300A frequencies in order to send the ChannelChangeCommand, eg in case the inverter hasn’t answered for a while.

To me this looks like somewhere in Fall 2023 the HMS firmware was upgraded for all new HMS inverters to:
Firmware Version 1.0.27
Firmware Build Date 2023-06-05 10:24:00

i think Bernhard @broth-itk may have upgraded his 2022 HMS to that firmware version, too ?

fir3bird, 2024-08-06:
@RS2480 
To answer your question what my pin mapping looks like:
```json
[
    {
        "name": "Generic NodeMCU 38pin",
        "cmt": {
            "sdio": 14,
            "clk": 12,
            "cs": 27,
            "fcs": 26,
            "gpio3": 34,
            "gpio2": 35
        }
    },
    {
        "name": "Generic NodeMCU 38pin + I2C Display",
        "cmt": {
            "sdio": 14,
            "clk": 12,
            "cs": 27,
            "fcs": 26,
            "gpio3": 34,
            "gpio2": 35
        },
        "display":{
            "data": 21,
            "clk": 22,
            "type": 2
        }
    }

]
```

But yours might look different. It depends how you soldered your CMT2300A to your ESP32.

For my problem: I am still out of luck. I already checked the solder joints twice if there is something broken. But I would have expected to see an error in the system info anyway if that would have been the case.

I checked different frequencies now, but always without success.

broth-itk, 2024-08-06:
@stefan123t Both HMS-1800 and HMS-2000 inverters running 1.0.27 in my case.

stefan123t, 2024-08-06:
Bernhard @broth-itk that would somehow confirm the hunch that your Model HMS-350-1T
Firmware Version 1.0.14
Firmware Build Date 2021-12-09 12:46:00
is not affected by the issue, right ?

Or is this also impacted if it is the sole / first inverter instead of the last ?

broth-itk, 2024-08-06:
@stefan123t 

I have not seen the HMS-350T-1T with 1.0.14 going down/offline as the others did.

Beside that observation, I started to notice the issue with the last few versions which were released.
Maybe coincidence.

fir3bird, 2024-08-16:
I received my DTU lite stick from hoymiles today and the stick connects immediately to my HMS-1600-4T. But I still cant get my OpenDTU connected.
The logs do not even show `interrupt received` like it does for Paule-Panter.

My logs
```
12:30:50.928 > RX Period End
12:30:50.928 > All missing
12:30:50.928 > Nothing received, resend count exeeded
12:30:50.977 > TX SystemConfigPara 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 05 00 66 BF 2A 54 00 00 00 00 00 00 00 00 D3 AF 0D 
12:30:51.233 > RX Period End
12:30:51.233 > All missing
12:30:51.233 > Nothing received, resend whole request
12:30:51.233 > TX SystemConfigPara 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 05 00 66 BF 2A 54 00 00 00 00 00 00 00 00 D3 AF 0D 
12:30:51.490 > RX Period End
12:30:51.490 > All missing
12:30:51.490 > Nothing received, resend whole request
12:30:51.490 > TX SystemConfigPara 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 05 00 66 BF 2A 54 00 00 00 00 00 00 00 00 D3 AF 0D 
12:30:51.745 > RX Period End
12:30:51.745 > All missing
12:30:51.745 > Nothing received, resend whole request
12:30:51.745 > TX SystemConfigPara 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 05 00 66 BF 2A 54 00 00 00 00 00 00 00 00 D3 AF 0D 
12:30:52.004 > RX Period End
12:30:52.004 > All missing
12:30:52.004 > Nothing received, resend whole request
12:30:52.004 > TX SystemConfigPara 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 05 00 66 BF 2A 54 00 00 00 00 00 00 00 00 D3 AF 0D 
12:30:52.207 > RX Period End
12:30:52.207 > All missing
12:30:52.207 > Nothing received, resend count exeeded
12:30:52.207 > Fetch inverter: 1164A00948F5
12:30:52.207 > Request SystemConfigPara
12:30:52.256 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 09 48 F5 80 16 40 84 02 15 21 20 14 12 
12:30:52.408 > RX Period End
12:30:52.408 > All missing
12:30:52.408 > Nothing received, resend whole request
12:30:52.408 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 09 48 F5 80 16 40 84 02 15 21 20 14 12 
12:30:52.459 > RX Period End
12:30:52.459 > All missing
12:30:52.459 > Nothing received, resend whole request
12:30:52.459 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 09 48 F5 80 16 40 84 02 15 21 20 14 12 
12:30:52.612 > RX Period End
12:30:52.612 > All missing
12:30:52.612 > Nothing received, resend whole request
12:30:52.612 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 09 48 F5 80 16 40 84 02 15 21 20 14 12 
12:30:52.664 > RX Period End
12:30:52.664 > All missing
12:30:52.664 > Nothing received, resend whole request
12:30:52.664 > TX ChannelChangeCommand 868.00 MHz --> 56 A0 09 48 F5 80 16 40 84 02 15 21 20 14 12 
12:30:52.819 > RX Period End
12:30:52.819 > All missing
12:30:52.819 > Nothing received, resend count exeeded
12:30:52.872 > TX RealTimeRunData 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 0B 00 66 BF 2A 5C 00 00 00 00 00 00 00 00 DD C7 6D 
12:30:53.084 > RX Period End
12:30:53.084 > All missing
12:30:53.084 > Nothing received, resend whole request
12:30:53.084 > TX RealTimeRunData 868.00 MHz --> 15 A0 09 48 F5 80 16 40 84 80 0B 00 66 BF 2A 5C 00 00 00 00 00 00 00 00 DD C7 6D 
```

I double checked solder joints and config and the DTU lite stick from hoymiles was disconnected.

Paule-Panter, 2024-08-16:
> ```
> 
> I double checked solder joints and config and the DTU lite stick from hoymiles was disconnected.

Your problem looks totally different than the discussed one in that thread. 
Please open a separate thread.

Anyway:
It seems to me you have a first time connection problem as there is interrupt receiving. 
Maybe you should define a gpio1 pin connected to your cmt-modul in your file. And make a corresponding hw-connection between the both modules.
And double, double check the pin/gpio reference table for your specific esp32-modul.

So,  and please open up a new thread for your problem.

Paule

Paule-Panter, 2024-08-16:
A short interim report from my system:

At the moment it is working for the most part. At least to the that I can live with it.

The DTU loses the connection to the WR on some days (for whatever reason). But after a while it always finds it again on its own without me having to intervene. I also do NOT have to wait until the next day or restart the WR.

The only change that I am aware of is setting the frequency in the DTU to 868.00 MHz.

As the channel change commands always seem to be sent on this channel, I thought the channel was a good fit.

Fortunately, the spectrum on the channel at my location is not being used for anything else.

Since then the DTU has at least found the WR again after a while.

Maybe i have to look deeper to that, if i want to create a "nulleinspeisung". Up to that i have to relax at the beach ;-)
[image]
[image]

swatermann, 2024-08-26:
Hi there,

I also encountered this problem, my inverter is brand-new, the serie number is 1164A011xxxx.

I tried different frequencies and tried to connect after restarting the module every time. The console window is no longer updated after about 5-10 minutes. Check it in Putty and the error is CMT: Buffer full.

Someone say to wait 15 minutes or even hours. I waited for a whole day, but no attempt was made to connect successfully. In putty always reported the same error.

Can anyone help me?

Best Regards

[image]

lehmartin, 2024-08-26:
Hello there, same problem here: HMS-2000-4T (so using the CMT module), connection after (re)starting works for a few hours (both direction, can successfully set a custom power limit), sometimes even a day or two, then connection drops even though the inverter still works (LED flashes green once each second which indicates regular operation). 

Serial number is 1164914xxxxx. Will add the current operating frequency later, I'm not at home right now.

lehmartin, 2024-08-29:
I can confirm that every time my inverter loses connection to OpenDTU, changing the frequency works. The new frequency was always in the range of 865.00 +/-0.50 MHz.

spcqike, 2024-08-29:
in my experience, you dont need to change the frequency, its enough to just save the DTU settings page, again. 

> Is there any way to set a different frequency automatically, e.g. via MQTT?

not that i know. but you can send a http post request and "click save". of course you can also set another frequency there. just have a look at the web_api doc

https://www.opendtu.solar/firmware/web_api/

lehmartin, 2024-08-30:
> in my experience, you dont need to change the frequency, its enough to just save the DTU settings page, again.
@spcqike tried your suggestion, doesn't work, I have to find the right frequency to work. Using the same as before and just hitting the save button has no effect. As soon as I find the correct frequency, data is being received again.

stefan123t, 2024-08-30:
> CMT: Buffer full.

@Paule-Panter can you confirm the `CMT: Buffer full.` error in the USB Serial Console ? It may not be visible in the Web Console as it may arise after WiFi case nnection is lost.

@swatermann can you attach a log from the restarting the ESP until it shows the above error code ?
We atill lack the first occurrence of this error hence a detailled and long log is necessary to debug. Thanks!

swatermann, 2024-08-30:
@stefan123t 

Poll Inverval: 10s
CMT2300A Frequency: 865.5MHz

LOG:

Using 'COM6' as serial port.
Showing logs:
[17:37:39]Fetch inverter: 1164A011xxxx
[17:37:49]Fetch inverter: 11449480xxxx
[17:37:59]Fetch inverter: 1164A011xxxx
[17:38:09]Fetch inverter: 11449480xxxx
[17:38:09]TX RealTimeRunData 865.50 MHz --> 15 94 80 19 30 80 19 82 56 80 0B 00 66 D1 E7 61 00 00 00 00 00 00 00 00 63 95 29 
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Frame kaputt
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Frame kaputt
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Frame kaputt
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Frame kaputt
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[17:38:09]Interrupt received
[... 1319 lines cut ...]
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
[17:38:12]CMT: Buffer full
...

stefan123t, 2024-08-30:
@swatermann thanks this is almost instantly in your case. I also only see Interupt received but no RX data from the CMT module being parsed.
I would suggest to log a separate issue as the others have their issuebonly after several hours.

You can use the [/] slash commands Details and Code Block to collapse/expand and format your log while editing. Or simply attach it as a log file using drag and drop.

Paule-Panter, 2024-09-03:
> > CMT: Buffer full.
> 
> @Paule-Panter can you confirm the `CMT: Buffer full.` error in the USB Serial Console ? It may not be visible in the Web Console as it may arise after WiFi connection is lost.
> 
I have just started the usb serial log. 
Have to wait now to a new connection loss.

As reported three weeks ago, it is currently running reasonably stable. 
I'll check this over the next few days and upload the corresponding log file extract in the event of an error

stefan123t, 2024-10-21:
All the issues #1476 #1550 #1627 #1921 #1961 #2056 and especially this #2137 do exhibit problems with the ChannelChangeCommand and ultimately loose the connection to the inverter.

@Paule-Panter thanks for your Pictures / Screenshots of the RF Signal here https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2226888434

According to the logfiles from @broth-itk we can now see the ChannelChangeCommand recorded on his original Hoymiles DTU in the wild. Actually it is repeated quite frequently, i.e. five time ChannelChangeCommand 0x56 interleaved with one time RealTimeRunData_Debug 0x15 0x0b.
The ChannelChangeCommand usually tries it four times on Channel 0x18 but also sometimes starts the first try on 0x21.
According to the timestamps the Training phase lasts from 09:11:31.234812 till 09:24:49.324898, i.e. about 13:18.090086 minutes.
During this phase we can also see the command 0x07 but mostly 0x53 interleaved every five Commands sent from the DTU and as a reply only 0xd6 to/from Serial `80164633`.

@broth-itk can you confirm my understanding of your Serial IDs ?

* DTU ? `80164633`
* HMS 1: `80729785`
* HMS 2: `81313070`

<details><summary>ChannelChangeCommand DTU Training</summary>
<p>

```hex
A: 2024-02-04 09:11:31.226933: b'7e 07 00000000 00000000 00 07 7f'
A: 2024-02-04 09:11:31.979214: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:32.283397: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:32.587554: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:32.875677: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:33.179846: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:34.076269: b'7e 56 81313070 81313070 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:34.380435: b'7e 56 81313070 81313070 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:34.684593: b'7e 56 81313070 81313070 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:34.988754: b'7e 56 81313070 81313070 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:35.276918: b'7e 56 81313070 81313070 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:36.189402: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:36.477504: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:36.781624: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:37.085757: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:37.389914: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:38.286318: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:11:38.334319: b'7e 07 80164633 80164633 00 07 7f'
A: 2024-02-04 09:11:38.638496: b'7e 56 80164633 80164633 01 1521 2114 56 7f'
A: 2024-02-04 09:11:38.686518: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:38.846598: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:38.990674: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:39.086686: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:11:39.150703: b'7e 53 83199047 83199047 53 7f'
A: 2024-02-04 09:11:39.438861: b'7e 56 80164633 80164633 01 1521 2114 56 7f'
A: 2024-02-04 09:11:39.502853: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:39.646900: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:39.790968: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:39.903102: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:11:39.951099: b'7e 53 83199047 83199047 53 7f'
A: 2024-02-04 09:11:40.255225: b'7e 56 80164633 80164633 01 1521 2114 56 7f'
A: 2024-02-04 09:11:40.303240: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:40.463286: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:40.607390: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:11:40.703417: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:11:40.767437: b'7e 53 83199047 83199047 53 7f'
...
A: 2024-02-04 09:24:41.048413: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:24:41.096415: b'7e 15 81313070 81313070 80 0b0065bf57f90000000000000000 5f97 22 7f'
A: 2024-02-04 09:24:41.560623: b'7e 56 80164633 80164633 01 1521 2114 56 7f'
A: 2024-02-04 09:24:41.608649: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:24:41.752703: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:24:41.912807: b'7e 56 80729785 80729785 02 1521 1814 6c 7f'
A: 2024-02-04 09:24:42.008836: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:24:42.056885: b'7e 15 80729785 80729785 80 0b0065bf57fa0000000000000000 af83 c5 7f'
A: 2024-02-04 09:24:42.505114: b'7e 56 80164633 80164633 01 1521 2114 56 7f'
A: 2024-02-04 09:24:42.569117: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:24:42.713192: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:24:42.857237: b'7e 56 83199047 83199047 02 1521 1814 6c 7f'
A: 2024-02-04 09:24:42.969279: b'7e 56 80164633 80164633 01 1521 1814 6f 7f'
A: 2024-02-04 09:24:43.017311: b'7e 15 83199047 83199047 80 0b0065bf57fb0000000000000000 3f8e 59 7f'
[... 220 lines cut ...]
B: 2024-02-04 09:25:08.248488: b'7e 95 80729785 80164633 02 c28c00045e30000900090003015c 0000 ee 7f'
B: 2024-02-04 09:25:08.296512: b'7e 95 80729785 80164633 03 001b00000060000515570004e2cc 0000 83 7f'
B: 2024-02-04 09:25:08.344542: b'7e 95 80729785 80164633 04 000708f6138a0135ffff000d0002 0045 8c 7f'
B: 2024-02-04 09:25:08.392652: b'7e 95 80729785 80164633 85 0004 385c 73 7f'
[... cut ...]
B: 2024-02-04 09:25:11.786711: b'7e 95 80729785 80164633 01 000100e500e60031003200720072 0003 95 7f'
B: 2024-02-04 09:25:11.850780: b'7e 95 80729785 80164633 02 c28c00045e30000900090003015e 0000 ec 7f'
B: 2024-02-04 09:25:11.898807: b'7e 95 80729785 80164633 03 001b0000005f000515570004e2cc 0000 bc 7f'
B: 2024-02-04 09:25:11.946834: b'7e 95 80729785 80164633 04 000708f8138b0133ffff000d0002 0045 85 7f'
B: 2024-02-04 09:25:11.978813: b'7e 95 80729785 80164633 85 0004 deea 23 7f'
B: 2024-02-04 09:25:12.235036: b'7e 95 80729785 80164633 81 0001 70c0 a6 7f'
...
``` 

</p>
</details>

broth-itk, 2024-10-22:
@stefan123t 

Thanks for this very interesting analysis!
These are my IDs and SW version:

- DTU: 80164633 (v00.02.23)
- HMS 1: 80729785 (1800-4T, v1.00.27)
- HMS 2: 81313070 (2000-4T, v1.00.27)
- HMS 3: 83199047 (350-1T, v1.00.14)

SG-fabian, 2024-10-25:
Everyday after a couple of hours the connection to my HMS-1600 is lost and only regained around six hours later. So I have a connection for a couple of hours at the start and the end of each day but not in between.

Let me know if I can help to solve this issue.

lehmartin, 2024-10-25:
@SG-fabian: At least for me (problem stated [here](https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2320164321)), changing the frequency helped, so I wrote an automation in Home Assistant which is a workaround for now. You could try changing the frequency and find whether that's the problem for you, too.

stefan123t, 2024-10-30:
@broth-itk I have merged the above two sequences from DTU outgoing / A: and Inverter incoming / B:.

```bash
awk '{out = ""; for (i = 4; i <= NF; i++) {out = out " " $i}; print $2, $3, $1, out}' channel_change.txt | sort > channel_change_merged.txt
```

In order to have it easier readable I have replaced your above DTU and Inverter IDs as follows:

* DTU: `80164633` (v00.02.23) -> `<DTU_ID>`
* HMS 1: `80729785` (1800-4T, v1.00.27) -> `<HMS_01>`
* HMS 2: `81313070` (2000-4T, v1.00.27) -> `<HMS_02>`
* HMS 3: `83199047` (350-1T, v1.00.14) -> `<HMS_03>`

Especially interesting is the following Init / Training Sequence, which we have seen with one of the earliest DTU WLite as well. 

[**Welche Kommandos (MainCmd) und Sub Kommandos (SubCmd) gibt es ?**](https://github.com/lumapu/ahoy/wiki/Protocol#welche-kommandos-maincmd-und-sub-kommandos-subcmd-gibt-es-)

`0x07` REQ_RF_RVERSISON mit SubCmd = `0x00` alle in UsartNrf_Send_NetCmdToNrfCmd().
`0x87` ANSWER_REQ_RF_RVERSISON in UsartNrf_Process_Inverter_Version_DtuRf(),

```
2024-02-04 09:11:31.226933: A:  b'7e 07 00000000 00000000 00 07 7f'
2024-02-04 09:11:31.234812: B:  b'7e 87 <DTU_ID> 00000000 02 10fd <DTU_ID> 050200020000 0005 68 7f'
...
2024-02-04 09:11:38.334319: A:  b'7e 07 <DTU_ID> <DTU_ID> 00 07 7f'
2024-02-04 09:11:38.343179: B:  b'7e 87 <DTU_ID> <DTU_ID> 02 10fd <DTU_ID> 050200020000 0005 8b 7f'
```

<details><summary>Details</summary>
<p>

```
Inverter Reply ChannelChangeCommand 
2024-02-04 09:11:31.226933: A:  b'7e 07 00000000 00000000 00 07 7f'
2024-02-04 09:11:31.234812: B:  b'7e 87 <DTU_ID> 00000000 02 10fd <DTU_ID> 050200020000 0005 68 7f'
2024-02-04 09:11:31.979214: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:32.283397: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:32.587554: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:32.875677: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:33.179846: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:34.076269: A:  b'7e 56 <HMS_02> <HMS_02> 02 1521 1814 6c 7f'
2024-02-04 09:11:34.380435: A:  b'7e 56 <HMS_02> <HMS_02> 02 1521 1814 6c 7f'
2024-02-04 09:11:34.684593: A:  b'7e 56 <HMS_02> <HMS_02> 02 1521 1814 6c 7f'
2024-02-04 09:11:34.988754: A:  b'7e 56 <HMS_02> <HMS_02> 02 1521 1814 6c 7f'
2024-02-04 09:11:35.276918: A:  b'7e 56 <HMS_02> <HMS_02> 02 1521 1814 6c 7f'
2024-02-04 09:11:36.189402: A:  b'7e 56 <HMS_01> <HMS_01> 02 1521 1814 6c 7f'
2024-02-04 09:11:36.477504: A:  b'7e 56 <HMS_01> <HMS_01> 02 1521 1814 6c 7f'
2024-02-04 09:11:36.781624: A:  b'7e 56 <HMS_01> <HMS_01> 02 1521 1814 6c 7f'
2024-02-04 09:11:37.085757: A:  b'7e 56 <HMS_01> <HMS_01> 02 1521 1814 6c 7f'
2024-02-04 09:11:37.389914: A:  b'7e 56 <HMS_01> <HMS_01> 02 1521 1814 6c 7f'
2024-02-04 09:11:38.279167: B:  b'7e d6 <DTU_ID> <DTU_ID> 01 00 1521 e3 7f'
2024-02-04 09:11:38.286318: A:  b'7e 56 <DTU_ID> <DTU_ID> 01 1521 1814 6f 7f'
2024-02-04 09:11:38.334319: A:  b'7e 07 <DTU_ID> <DTU_ID> 00 07 7f'
2024-02-04 09:11:38.343179: B:  b'7e 87 <DTU_ID> <DTU_ID> 02 10fd <DTU_ID> 050200020000 0005 8b 7f'
2024-02-04 09:11:38.638496: A:  b'7e 56 <DTU_ID> <DTU_ID> 01 1521 2114 56 7f'
2024-02-04 09:11:38.647416: B:  b'7e d6 <DTU_ID> <DTU_ID> 01 00 1521 e3 7f'
2024-02-04 09:11:38.686518: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:38.846598: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:38.990674: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:39.086686: A:  b'7e 56 <DTU_ID> <DTU_ID> 01 1521 1814 6f 7f'
2024-02-04 09:11:39.095643: B:  b'7e d6 <DTU_ID> <DTU_ID> 01 00 1521 e3 7f'
2024-02-04 09:11:39.150703: A:  b'7e 53 <HMS_03> <HMS_03> 53 7f'
2024-02-04 09:11:39.438861: A:  b'7e 56 <DTU_ID> <DTU_ID> 01 1521 2114 56 7f'
2024-02-04 09:11:39.447882: B:  b'7e d6 <DTU_ID> <DTU_ID> 01 00 1521 e3 7f'
2024-02-04 09:11:39.502853: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:39.646900: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:39.790968: A:  b'7e 56 <HMS_03> <HMS_03> 02 1521 1814 6c 7f'
2024-02-04 09:11:39.896166: B:  b'7e d6 <DTU_ID> <DTU_ID> 01 00 1521 e3 7f'
2024-02-04 09:11:39.903102: A:  b'7e 56 <DTU_ID> <DTU_ID> 01 1521 1814 6f 7f'
2024-02-04 09:11:39.951099: A:  b'7e 53 <HMS_03> <HMS_03> 53 7f'
[... 222 lines cut ...]
2024-02-04 09:25:09.301895: A:  b'7e 15 <HMS_03> <HMS_03> 80 0b0065bf58150000000200000000 6b42 22 7f'
2024-02-04 09:25:09.401256: B:  b'7e 95 <HMS_03> <DTU_ID> 01 00010160001d0066000065920007 08f3 2a 7f'
2024-02-04 09:25:09.449240: B:  b'7e 95 <HMS_03> <DTU_ID> 82 138b00610000000403e800570002 3609 c5 7f'
2024-02-04 09:25:09.750097: A:  b'7e 15 <HMS_03> <HMS_03> 80 110065bf58160000000200000000 414d 1e 7f'
2024-02-04 09:25:09.833504: B:  b'7e 95 <HMS_03> <DTU_ID> 81 0001 70c0 0b 7f'
2024-02-04 09:25:10.502505: A:  b'7e 15 <HMS_02> <HMS_02> 80 0b0065bf58170000000200000000 0b5b 59 7f'
2024-02-04 09:25:10.585982: B:  b'7e 95 <HMS_02> <DTU_ID> 01 000101650156001e001f006d006a 0000 b3 7f'
2024-02-04 09:25:10.633996: B:  b'7e 95 <HMS_02> <DTU_ID> 02 a8d50000a34800080008015a0164 001f 33 7f'
2024-02-04 09:25:10.698031: B:  b'7e 95 <HMS_02> <DTU_ID> 03 0020006c0073000113be0000b42e 0008 84 7f'
2024-02-04 09:25:10.746042: B:  b'7e 95 <HMS_02> <DTU_ID> 04 000908fd138b01a20351001201ba 004b f5 7f'
2024-02-04 09:25:10.778059: B:  b'7e 95 <HMS_02> <DTU_ID> 85 0002 2b67 4d 7f'
[... cut ...]
2024-02-04 09:25:11.850780: B:  b'7e 95 <HMS_01> <DTU_ID> 02 c28c00045e30000900090003015e 0000 ec 7f'
2024-02-04 09:25:11.898807: B:  b'7e 95 <HMS_01> <DTU_ID> 03 001b0000005f000515570004e2cc 0000 bc 7f'
2024-02-04 09:25:11.946834: B:  b'7e 95 <HMS_01> <DTU_ID> 04 000708f8138b0133ffff000d0002 0045 85 7f'
2024-02-04 09:25:11.978813: B:  b'7e 95 <HMS_01> <DTU_ID> 85 0004 deea 23 7f'
2024-02-04 09:25:12.151253: A:  b'7e 15 <HMS_01> <HMS_01> 80 110065bf58180000000400000000 2189 b2 7f'
2024-02-04 09:25:12.235036: B:  b'7e 95 <HMS_01> <DTU_ID> 81 0001 70c0 a6 7f'
```

</p>
</details>

SG-fabian, 2024-11-08:
Here's another observation: Like I already wrote my OpenDTU unit usually loses the connection before noon und regains it in the evening. But on cloudy days with a power output under 100 watts the connection is much more stable. So maybe the changes in frequency are somehow be related to the generated power? If I had to guess I'd say that the converter circuits create some kind of interference.

Maybe this helps to isolate the issue.

spcqike, 2024-11-08:
i have observed similar behavior with my HMS-2000, but not at 100W but at ~ 700-800W.

after a restart of my DTU and a "save cycle" of my DTU settings (to apply the set frequency, at 12:47) everything works as usual again

[image]

DavidDevZone, 2025-01-12:
Hello, are there any news? I have the same issue with the newest version.

fishpepper, 2025-01-31:
I observe the same Problem. I have A HMS-800-2T (FW 1.0.4, KW 15/2023) and a HMS-2000-4T (FW v1.00.27, KW 0/2024).
The HMS-2000 looses conenction around 10/11 or at ~700W. The HMS800 is not affected.

sauvant, 2025-02-27:
Same here, solution/workaround would be highly appreciated 👍

stefan123t, 2025-02-27:
@tbnobody @schlimmchen would you consider adding the ChannelChangeCommand 0x56 being repeated several times,
until the connection has been established again ?
https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2427788809

Have you seen the training / init sequence using REQ_RF_RVERSISON 0x07 and ANSWER_REQ_RF_RVERSISON 0x87 
that @broth-itk recorded on his Hoymiles DTU Pro S ? 
This was already in the very first trace on the mikrocontroller Forum 
https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2448651049

*please open the details* 🧐 for the actual **traces / logs**

schlimmchen, 2025-03-06:
I am not inclined to touch the Hoymiles lib...

fishpepper, 2025-04-05:
I was originally posting in https://github.com/tbnobody/OpenDTU/pull/2573, but i think it might be better to discuss it here.

I have a HMS-800-2T (FW 1.0.4, KW 15/2023) and a HMS-2000-4T (FW v1.00.27, KW 0/2024).
Right now only the HMS2000 is switched on. At around 900W i loose the connection (see the pr for graphs).
For me, the frequency manager branch did not help so i bought a hoymiles dtu, cracked it open, and attached a logger to the serial line. See the frequency hopping pr for some details.

Today was sunny, the hoymiles dtu reported fine. I captured the rx/tx lines inside the hoymiles DTU and wrote a quick and dirty parser.
Some statistics (from 10:00 - 16:15 o'clock):
```
30464 x MID 0x15 rtRunData request
89061 x MID 0x95 rtRunData response

15168 x MID 0x53 ?? request (contains a timestamp)
14919 x MID 0xD3 ?? response (contains a timestamp)

155 x MID 0x56 change channel request
 62 x MID 0xD6 change channel response
```

The change channel requestst happened at 15:23,15:29-15:32, and 15:47o'clock.

One example follows.  I tried to decode it and wrote some comments to it.
a38105e0 = DTU?
a009a45c = WR? 

```
DTU set 0x21?
2025-04-05 15:47:40.396699 TXO b'7e56a38105e0a38105e001 15 21 >21< 14 567f' 
2025-04-05 15:47:40.399005 RXO b'7ed6a38105e0a38105e001001521e37f'

WR set 0x0f
2025-04-05 15:47:40.450377 TXO b'7e56a009a45ca009a45c02 15 21 >0f< 14 7b7f'
2025-04-05 15:47:40.606623 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:40.752717 TXO b'7e56a009a45ca009a45c0215210f147b7f'
no response?

DTU set 0x0f
2025-04-05 15:47:40.855505 TXO b'7e56a38105e0a38105e00115210f14787f'
2025-04-05 15:47:40.856388 RXO b'7ed6a38105e0a38105e001001521e37f'
response

fetch data mid 0x15
2025-04-05 15:47:40.909632 TXO b'7e15a009a45ca009a45c800b0067f1509a0000000200000000597be27f'
2025-04-05 15:47:41.366493 TXO b'7e53a009a45ca009a45c80ff0067f1509a00000002000000006d8e917f'

DTU set 0x21
2025-04-05 15:47:41.664134 TXO b'7e56a38105e0a38105e00115212114567f'
2025-04-05 15:47:41.667693 RXO b'7ed6a38105e0a38105e001001521e37f'

WR set 0x21
2025-04-05 15:47:41.717480 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:41.868700 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:42.019393 TXO b'7e56a009a45ca009a45c0215210f147b7f'
no response

DTU set 0x0f
2025-04-05 15:47:42.120606 TXO b'7e56a38105e0a38105e00115210f14787f'
2025-04-05 15:47:42.128714 RXO b'7ed6a38105e0a38105e001001521e37f'
response

DTU set 0x21
2025-04-05 15:47:42.174544 TXO b'7e56a38105e0a38105e00115212114567f'
2025-04-05 15:47:42.180674 RXO b'7ed6a38105e0a38105e001001521e37f'
response

WR set 0x0f
2025-04-05 15:47:42.228463 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:42.379501 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:42.530608 TXO b'7e56a009a45ca009a45c0215210f147b7f'
no response

DTU set 0x0f
2025-04-05 15:47:42.631502 TXO b'7e56a38105e0a38105e00115210f14787f'
2025-04-05 15:47:42.634520 RXO b'7ed6a38105e0a38105e001001521e37f'
response

fetch data mid 0x15
2025-04-05 15:47:42.686264 TXO b'7e15a009a45ca009a45c800b0067f1509b0000000200000000c9767d5e7f'
2025-04-05 15:47:43.146207 TXO b'7e53a009a45ca009a45c80ff0067f1509c0000000200000000cda51c7f'

DTU set 0x21
2025-04-05 15:47:43.440137 TXO b'7e56a38105e0a38105e00115212114567f'
2025-04-05 15:47:43.442282 RXO b'7ed6a38105e0a38105e001001521e37f'

WR set 0x0f
2025-04-05 15:47:43.493553 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:43.645584 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:43.795641 TXO b'7e56a009a45ca009a45c0215210f147b7f'
no response

DTU set 0x0f
2025-04-05 15:47:43.896784 TXO b'7e56a38105e0a38105e00115210f14787f'
2025-04-05 15:47:43.899164 RXO b'7ed6a38105e0a38105e001001521e37f'
response

DTU set 0x21
2025-04-05 15:47:43.951353 TXO b'7e56a38105e0a38105e00115212114567f'
2025-04-05 15:47:43.957182 RXO b'7ed6a38105e0a38105e001001521e37f'
response

WR set 0x0f
2025-04-05 15:47:44.004501 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:44.155657 TXO b'7e56a009a45ca009a45c0215210f147b7f'
2025-04-05 15:47:44.307295 TXO b'7e56a009a45ca009a45c0215210f147b7f'
no response

DTU set 0x0f
2025-04-05 15:47:44.407832 TXO b'7e56a38105e0a38105e00115210f14787f'
2025-04-05 15:47:44.412805 RXO b'7ed6a38105e0a38105e001001521e37f'
response

data mid 0x15
2025-04-05 15:47:44.463221 TXO b'7e15a009a45ca009a45c800b0067f1509d0000000200000000695df37f'
2025-04-05 15:47:44.552325 RXO b'7e95a009a45ca38105e001000101b301b9020001f608b408a70001ee7f'
```

The hoymiles has 3 CPUS:
EPS <--->G32<---->small uc<---> rf module
Keep in mind that this is the log of the serial connection between the G32 and the small uc.

I need to do some more digging. It seems like the hoymiles does not loose the connection that often and tries only minimal frequency switching (just two channels: 0x15210f and 0x152121?).

What is mid 0x53 used for? I see it quite a lot in the hoymiles log (see stats). Maybe the opendtu does not send it and the HMS assumes a dead connection?

fishpepper, 2025-04-07:
I have an interesting observation to share:
When i was done with collecting the logfiles with my hoymiles dtu, i removed the inverter from the dtu via the hoymiles website.

Even though it was not reporting data any more, it did not stop to send requests:
* MID 0x02 - 2 in one day
* MID 0x15 (62k)
* MID 0x53 (45k)
* MID 0x56 (330k)
* MID 0x57 (68)

The logfile spans roughly one day.

For the change channel command (mid 0x56), it requests channels
* 0x0c - 53k times
* 0x0f - 88k times
* 0x12 - 23k times
* 0x15 - 33k times
* 0x18 - 35k times
* 0x1b - 23k times
* 0x1e - 3k times
* 0x21 -  66k times
* 0x24 - 4k times

I started the opendtu in parallel (running the freq manager branch, https://github.com/functionpointer/OpenDTU/commit/802ca4ed1bea56f3fa4b413431d068bc7c1ab6e3) and it connected just fine to the HMS2000 even though the hoymiles DTU was still sending (some?) requests.

I have now two days with data reception above 900W:
[image]

Every now and then there is a missing dataset, i assume because the hoymiles triggers a channel switch.
This observation shows that this is not a hw problem/interference problem but indeed some kind of protocol issue.

So the question is: why does the packets that the hoymiles dtu sends improve the opendtu receiption? Is it just the regular change channel command or the 0x53 packets?

fishpepper, 2025-04-07:
> Is the content of the 0x53 command always the same or does it change it's values?

The 0x53 command looks like 0x15 packet described here (https://github.com/lumapu/ahoy/blob/5feb293c9f53e980e591a0ec9b8851794a011743/doc/hoymiles-format-description.txt#L250).
The 0x15 packets are also in the logfile.
Keep in mind that i record the data at the serial link inside the hoymiles DTU. 0x53 might be only used for internal communication and might not be sent via the rf link.

> If you are looking at the different 0x56 commands is there a logic in which order the channels are switched? 

When only the hoymiles was connected and the inverter was enabled it was using 0x21 and 0x0f only (if i remember correctly).

In my "dual" setup, the hoymiles frequency commands look like this:

[image]

random zoom in:

[image]

You can find a log file here:
[dtu_hoymiles_and_opendtu.log.gz](https://github.com/user-attachments/files/19629715/dtu_hoymiles_and_opendtu.log.gz)
TXO = serial data sent to the rf module uc
RXO = serial data received from the rf module uc

At around 12:30 i switched off the reporting on the hoymiles website and started the opendtu again.

>  I noticed that the issue only occurred for me during very cold temperatures.

ok good point! i will switch off the hoymiles dtu and see if the connection drops off again tomorrow!
But i think I did not see a temperature correlation yet.

spcqike, 2025-04-07:
do you still see connection problems at a certain output power? like between 600-800W?

tbnobody, 2025-04-07:
@fishpepper I just had a look at the `0x53` (request) and `0xd3` (response) messages.
I've analyzed the time between 07.04.2025 10:00 and 07:04:2025 10:30.
The request and response look indeed like the ones for command `0x15`

Requests (only a few of them):
[image]

Whats interessting, in this time frame I see 993 requests but only 28 response messages. I would have expected that every request would trigger a response.

Responses (Please forget about the column description from R - AD, the last 3 columns should be the CRC codes):
[image]

Column U seems to count upwards.. Then flows over to column T and then something happens with column V
[image]

tbnobody, 2025-04-07:
Another interessting fact is your channel change command:
[image]

The orange marked column (O) is always `0x14` as in OpenDTU.

OpenDTU applies the following values to the violet marked columns
```cpp
void ChannelChangeCommand::setCountryMode(const CountryModeId_t mode)
{
    switch (mode) {
    case CountryModeId_t::MODE_US:
        _payload[9] = 0x03;
        _payload[10] = 0x17;
        _payload[11] = 0x3c;
        break;
    case CountryModeId_t::MODE_BR:
        _payload[9] = 0x03;
        _payload[10] = 0x17;
        _payload[11] = 0x3c;
        break;
    default:
        _payload[9] = 0x02;
        _payload[10] = 0x15;
        _payload[11] = 0x21;
        break;
    }
}
```

In your snapshot column `L` and `M` match up with the values mentioned above.. But column `K` seems to be sometimes `0x01` and `0x02`. Not sure about this.

tbnobody, 2025-04-07:
> [@tbnobody](https://github.com/tbnobody) [@schlimmchen](https://github.com/schlimmchen) would you consider adding the ChannelChangeCommand 0x56 being repeated several times, until the connection has been established again ? [#2137 (comment)](https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2427788809)
> 
> Have you seen the training / init sequence using REQ_RF_RVERSISON 0x07 and ANSWER_REQ_RF_RVERSISON 0x87 that [@broth-itk](https://github.com/broth-itk) recorded on his Hoymiles DTU Pro S ? This was already in the very first trace on the mikrocontroller Forum [#2137 (comment)](https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2448651049)
> 
> _please open the details_ 🧐 for the actual **traces / logs**

@stefan123t and @fishpepper are you maybe able to compile the source by yourself? Then I would prepare a patch which inserts the `0x56` 3 times with parameter `0x02` and one time with `0x01`. I am just not sure whether we have to interrupt the process after a `d6` was received.

fishpepper, 2025-04-07:
> are you maybe able to compile the source by yourself? 

Yes, i can compile the source. I am an embedded dev, mostly time constrained right now and did not yet dig deep into the opendtu code base.

>Whats interessting, in this time frame I see 993 requests but only 28 response messages. I would have expected that every request would trigger a response.

As the openDTU runs in parallel, i am not sure if it is receiving on the same channel as the hoymiles sends the request. In addition, i am not sure if the commands in my traces of the serial link are actually sent via rf.

>Column U seems to count upwards.. Then flows over to column T and then something happens with column V

R also seems to change, to me it looks like there is an extra byte and T and U move to the right.
Is there some encoding that prevents 0x7e in the data? it looks like the ts would increment to ....7e but instead we see 0x7d followed by 5e/5d.

tbnobody, 2025-04-07:
`0x7e` and `0x7d` are only transfered via the UART interface. they never reach the RF part.

fishpepper, 2025-04-08:
> `0x7d` and `0x7f` are only transfered via the UART interface. they never reach the RF part.

They seem to do some replacements on the uart side. 0x7d5e is the replacement for 0x7e and so on. see:
See https://github.com/lumapu/ahoy/wiki/Protocol#was-sind-backwardforwardsubstitution 

So N-Q is indeed the timestamp but with forward replacements of 0x7, 0x7e, and 0x7f

[16 more replies]