tbnobody/OpenDTU #2138: not possible to change the ip address

grasmax, 2024-07-12:
### What happened?

After first start, the device got ip .72
Then in the Fritz.Box the device got the fix ip .39 but after that, it was not possible to connect to opendtu.
Same after off/on power and restart opendtu and restart fritzbox.

### To Reproduce Bug

change the ip in the router network config

### Expected Behavior

if dhcp is activatet, the device should have the ip from the router after restart

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.1.4

### Anything else?

temp solution: remove the check from dhcp and fill in the right ip fields

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-07-12:
Why did the fritzbox change the ip of the device? the esp should only get a new ip if the lease time expired (as every dhcp enabled device). whats your lease time?

stefan123t, 2024-07-12:
Maybe that is the issue ?

@grasmax can you turn off your OpenDTU, then reset the Fritz!Box and any Repeaters. Now the Fritz!Box should not have an active DHCP lease for your OpenDTU anymore.

You may change or update the fixed IPv4 assigned to you OpenDTU before this procedure or after but before you turn on the OpenDTU with the MAC address registered in your Fritz!Box from earlier.

Can you report back on the results ?

grasmax, 2024-07-13:
Zuerst müsste ich es schaffen, die "statische IP-Konfiguration" in "Einstellungen/Netzwerkeinstellungen" wieder loszuwerden.
Wenn ich die 5 Felder lösche, den Haken bei "DHCP aktivieren" setze und auf "Speichern" drücke, erscheint die Fehlermeldung "IP-Adresse ist ungültig!" und die Felder werden unsichtbar wiederhergestellt. Für dieses Verhalten hatte ich auch ein issue angelegt...
Btw, ich hatte natürlich die OpenDTU-Box xmal aus/eingeschaltet und auch die Fritzbox neu gestartet.
Welche "LeaseTime" in meinem Router eingestellt ist, ist unerheblich. Ich werde diese wegen OpenDTU nicht ändern.
Die OpenDTU-Box soll sich einfach die im Router hinterlegte IPv4 holen, wenn ich sie einschalte.

stefan123t, 2024-07-14:
@grasmax und wenn Du einfach die OpenDTU neu flashst ?

Du kannst auch vorher eine config.json als backup erstellen, wobei ich empfehlen würde die Einstellungen danach from-scratch vorzunehmen. Das JSON so zu editieren, wie es ohne statische IP Einträge aussehen muß, könnte man mit dem Vergleich zwischen einem frisch geflashten Backup vs. Deinem statischen IP Backup, das ist aber fast aufwendiger.

tbnobody, 2024-07-14:
Es sollte normal garnicht nötig seine irgendeine ip irgendwo zu löschen. Einfach auf dhcp stellen und gut. Dann werden die manuellen ips nie verwendet.

stefan123t, 2024-07-14:
Ja aber @grasmax scheint das so nicht machen zu können? 

Warum er nicht einfach die alte IP drin lässt und einfach den Schalter wieder auf DHCP stellt weiß ich nicht. Vielleicht will er die Formular-Validierungsfunktion des IP-Adressen Dialogs ausprobiereb, die schlägt vermutlich an wenn er die IP **rauslöscht** und das so speichern will.
Bei mir steht in den Feldern unter Static IP Configuration überall 0.0.0.0 drin, aber auch ohne Einträge lässt sich das speichern: v24.6.29

So lange die DHCP lease time der Fritz!Box noch nicht expired ist, bleibt die bisher feste IP dem ESP32 zugeordnet.
Erst nach einem Neustart der Fritz!Box werden die DHCP Adressen neu vergeben, weil erst dann der DHCP/ARP Cache mit den MAC Adressen und der IP Zuordnung neu initialisiert werden kann.

Man kann m.E. nicht einfach von Static IP zurück auf DHCP stellen, erst nach dem (vorzeitigen) Ende der Lease time (logisches/zeitliches) **und** einem Neustart des ESP32 kann die Fritz!Box überhaupt eine andere neue IP vergeben. 
Daher ja der Vorschlag erst ESP aus, dann Fritz!Box neu starten und erst danach den ESP wieder an.

grasmax, 2024-07-15:
Danke Stefan, das wars: wenn man den Haken bei DHCP rausnimmt und 0.0.0.0 in alle fünf Felder der "Statischen Konfiguration" einträgt, ist das Speichern erfolgreich. Damit hätte sich zumindest #2139 erledigt.

Habe auch NOCH EINMAL OpenDTU und auch die Fritzbox vom Strom getrennt. Gewartet, bis die Fritzbox wieder lief und unter Heimnetz/Netzwerk die OpenDTU mit .39 und dem Hinweis "gültig ab der nächsten Anfrage" angezeigt wurde.
Dann auch OpenDTU wieder eingeschaltet.
Leider ist das Teil immer noch nicht über die .39 erreichbar, sondern immer noch über .72

stefan123t, 2024-07-16:
@grasmax welche DHCP Range hast Du denn für die Fritz!Box zur freien Vergabe reserviert?

Ich habe bei mir alles zwischen 192.168.x.20 und 192.168.x.200 als DHCP frei. Siehe Heimnetz > Netzwerk >Netzwerkeinstellungen > IPv4-Adressen
Deshalb vergebe ich feste IPs im Bereich .2 bis .19 für Drucker, Raspberry, etc oder über .200

Ist die .72 bei Dir überhaupt frei ?
Und ja die Fritz!Box braucht ggf. ein paar Anläufe bis sie die Adresse statisch richtig zuordnen kann. Das ist aber nicht das Problem von OpenDTU sondern eines auf Seite des DHCP Admins. Wenn man die MAC Adresse des ESP32 vorher kennt (zB auf dem Chipdeckel per Lasergravur/QR Code?) kann man die bei einigen DHCP Servern vorher eintragen. Das geht bei der Fritz!Box mW aber nicht.

grasmax, 2024-07-16:
"DHCP Range hast Du denn für die Fritz!Box ": 2-200
"Ist die .72 bei Dir überhaupt frei ?" JA, sonst würde sie DTU nicht bekommen.
Und zur Erinnerung: DTU hatte die 72 bei der Erstinstallation und wird sie nicht mehr los.
Ich möchte, dass die DTU die in der Fritzbox fest eingestellte 39 bekommt.
Und das funktioniert nur, wenn ich in der DTU den Haken bei "DHCP" wegnehme und in die 39 in das Feld der "statischen Konfiguration" eintrage zusammen mit Maske usw.

grasmax, 2024-07-16:
Habe alle mit der Fritzbox verbundenen Geräte bzgl dhcp geprüft: Es gibt da das VictronCerboGX-Steuergerät, das normal über die 38 verbunden ist, aber mit einer zweiten generierten MAC-Adresse und einer IP 169.254... angemeldet ist. Laut Victron sollte das keine Probleme machen (siehe Antwort von [mvader ](https://community.victronenergy.com/questions/133443/cerbo-gx-announces-two-ip-adresses.html)), ist aber im Moment die einzige Spur, weil Geräte bei dchp-Problemen offenbar eine IP aus dem Subnetz 169.254 benutzen.
Leider habe ich gerade meine PV-Insel außer Betrieb gesetzt, weil ich das Steuergerät abgeschaltet habe um zu prüfen, wie sich dann die DTU-IP-Vergabe/Bezug verhält. Ich kann diesen Test also erst im nächsten Wartungsfenster durchführen und lasse es bis dahin bei der in der DTU statisch vergebenen 39.

stefan123t, 2024-08-12:
@grasmax gibt / gab es hierzu etwas neues im letzten Monat ? 

Das Problem dass Du die DHCP Adresse nicht neu / fest vergeben kannst hast m.W. nur Du bisher gemeldet.

closed 2025-04-18 by github-actions[bot]