tbnobody/OpenDTU #2152: HMS-800-2T

Vicio88, 2024-07-17:
### What happened?

Buon giorno. Ho un problema con la mia dtu. Non riconosce il mio inverter HMS-800-2T. Ho fatto anche l’ultimo aggiornamento ma non va 

### To Reproduce Bug

Buon giorno. Ho un problema con la mia dtu. Non riconosce il mio inverter HMS-800-2T. Ho fatto anche l’ultimo aggiornamento ma non va 

### Expected Behavior

v4.4.7-dirtyI

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v4.4.7-dirty

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

broth-itk, 2024-07-17:
Please provide more information like S/N of your inverter and the correct OpenDTU version.
Screenshot would be nice.
v4.4.7-dirtyI is no version from an OpenDTU release

Vicio88, 2024-07-17:
[image]
[image]
[image]

Vicio88, 2024-07-17:
Inverte hoymiles HMS-800-2T

stefan123t, 2024-07-17:
@Vicio88 we need at keast the first seven digits of your inverter serial id. The last five are unique and can be ommitted. Do you have two different serial IDs on the sticker, ie the model number ( first four digits ) may differ for HMS-800**W**-2T Models as they have a built-in DTUBI as well.

Vicio88, 2024-07-17:
[image]

stefan123t, 2024-07-17:
Your first inverter is a HMS-800**W** model. The second most likely too, given the same model prefix.
You can use any of the [other projects for DTUBI and HMS-xxx**W**](https://github.com/lumapu/ahoy/issues/1189#issuecomment-2211210076) HMS-WiFi-Series Models or directly use the S-Miles Installer App / S-Miles Cloud from Hoymiles itself.

Vicio88, 2024-07-17:
Dove lo trovo ?

stefan123t, 2024-07-17:
Ciao Vicio, the most known is this one written in RUST from DennisOSRM https://github.com/DennisOSRM/hms-mqtt-publisher
At the moment it does support only a single HMS-WiFi-Series, though there is a pull request # 94 to support multiple inverters too.

I do prefer the Python implementation by rovo89 as it is more promising, but I do not own an HMS-WiFi-Series inverter so this is only my personal point of view.

broth-itk, 2024-07-23:
Maybe it's  good idea to add all known serial number prefixes to the list and name them "Unsupported" or "WIFI Unsupported"

closed 2024-08-05 by tbnobody

tbnobody, 2024-08-05:
I will add a checkbox to the issue template. I'm not really willing to add a blacklist to the WebGUI, especially as an "Unknown" etc. is also displayed in the inverter overview. It will be also mentioned in the documentation. I mean, in how many other places can people ignore it?

stefan123t, 2024-08-06:
Well if they can ignore it, they will. 🤣
Even if a clear Warning its there they will try to ignore it if possible.

One last thought, though you are reluctant to add support for those built-in DTUBI HMS WiFi series  inverters, they are not Unknown but just Unsupported at the moment.

Maybe we should add a stub for a protobuf implementation via a separate AP network link. 
Once the Setup AP has been closed we could use its connection to connect to at least one of the DTUBI APs.
I wonder if we would have to connect to each AP in case we want to query multiple such DTUBI based inverters ?

broth-itk, 2024-08-06:
IMHO I agree to stefan123t, there should be a "Unsupported" message in the GUI when adding a W inverter.
Since asking questions is always easier as reading documentaton or even the text beside of checkboxes, this change will help keeping motivation up for all involved in OpenDTU development and support (via Issues).

Just my 2 cents