tbnobody/OpenDTU #2154: /api/livedata/status -> too many decimal places via web query

Tobi266, 2024-07-18:
### What happened?

I query the current power via an http query from NodeRed. The return is a json object . But with unnecessarily many decimal places. 

### To Reproduce Bug

I'm using http://IPofDTU/api/livedata/status to get actual data. Result is in picture
[image]

### Expected Behavior

limit the decimals to a relevant number. e.g one decimal -> 111.1

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.29

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-07-18:
You can round it yourself in nodered. The amount of the valid decimal places is shown in the json response. Will not change it in the API as it would require several conversions on the ESP. 
The current decimal places occour due to the float data type. Which has to be converted to String to cut of the decimal places and then back to float (but most likley the decimal places will occour again due to float inaccuracies.

ms49434, 2024-07-18:
You can format the output using the Value Format field in Node-RED.
Set the payload to the field you want to display and the below conversion will display the value as it should:

Power -> Payload -> Text Node -> Value format
{{msg.payload.Power.v.toLocaleString("de-DE", { style: 'decimal', useGrouping: true, minimumFractionDigits: msg.payload.Power.d, maximumFractionDigits: msg.payload.Power.d })}} {{msg.payload.Power.u}}

closed 2024-07-23 by tbnobody