tbnobody/OpenDTU #2157: [Request] Add support for TSUN TSOL_MS2000 inverter

QE-jmm, 2024-07-20:
### Describe the solution you'd like

TSUN TSOL inverters with 1 & 1, 2 & 2, and 4 & 2 DC Inputs & MPP-Trackers are supported; the TSOL-MS2000 with 4 & 4 DC Inputs and 4 MPP-Trackers would be a great addition.

### Additional context

My situation requires 4 DC Inputs and 4 MPP-Trackers; the TSOL-MS2000 inverter came in a bundle with the PV modules.
Since I'd rather not use the manufacturer's cloud, I'd have to sell the TSUN inverter and ,for example, buy an supported Hoymiles HMS-2000-4T.

tbnobody, 2024-08-01:
This will not happen in near future (not sure if even in far future). This specific inverter uses integrated WiFi and no CMT or NRF communication module.

QE-jmm, 2024-08-01:
Thank you for your response.

I now understand the difficulty with the WiFi specifics.

 

I’ll see what other methods of monitoring [not administration] I can deploy.

 

Best regards

Joerg Matiak

stefan123t, 2024-08-03:
The implementationnof HMS WiFi Series like HMS-xxx**W** was already rejected in #1324 as not feasible.
We would have to implement a WiFi connection to the AP of the DTUBI (built in DTU) of the inverter.
This is solved e.g. by the projects described here https://github.com/lumapu/ahoy/issues/1189#issuecomment-2211210076