tbnobody/OpenDTU #2170: "HMS-400BM" recognized as "HMS-500-1T v2"

tt-why, 2024-07-26:
### What happened?

Hi all :wave: 

When adding my "HMS-400BM" inverter using the serial "1125920XXXXX" the openDTU recognize it as "HMS-500-1T v2" hence requiring "CMT2300A" RF module instead of the "NRF24L01+" I currently have.

I've just upgraded the software to the latest version available.

[image]
[image]
[image]

Is there a bug on the parsing of the serial number, or is the inverter datasheet on the back of the invertert misleading ?
According to [the code parsing](https://github.com/tbnobody/OpenDTU/blob/master/lib/Hoymiles/src/inverters/HMS_1CHv2.cpp#L36) the serial what i'm seeing is normal but I would like this to work and I hope there is a small mistake :pray: 

Just for you to have full context, inverter has been provided by Beem in the package "Beem On" [here is the technical doc](https://cdn.shopify.com/s/files/1/0554/6632/8107/files/050224_-_Beem_On_Doc_Tech_compressed.pdf?v=1716212106&gclid=EAIaIQobChMIos_Xz6nEhwMVRD4GAB3PFgdSEAAYASAAEgL7f_D_BwE) (sorry it's only in french)

Thanks you for your help !

### To Reproduce Bug

Add inverter using the serial "1125920XXXXX" and see it's not HMS-400 but HMS-500 :/

### Expected Behavior

In my opinion my inverter serial number should map to "Hoymiles HM-400-1T" using NRF24L01+.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.29

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-07-26:
In general, all HMS inverters require a CMT2300A module as the "s" means sub-gigahertz what is the 868MHz band.

Unfortunatly, neither the back side of the inverter nor the manual shows the communication frequency.

tt-why, 2024-07-26:
Thank you for the quick response !

Do you have a method to find the right frequency, I own a "Nooelec RTL-SDR v5" if it can help.

Else you think that buying a CMT2300A would solve the problem ? Can I have a single ESP32 with both RF receivers working ?

tbnobody, 2024-07-26:
https://media.adeo.com/marketplace/LMFR/90202526/4038281.pdf

Based on this document its 868MHz

tt-why, 2024-07-26:
Ok thank you so much for your support and sorry for the noise !

tt-why, 2024-08-09:
Hi there, thanks to your advice I bought a brand new https://www.opendtu.solar/3rd_party/opendtu_fusion/ and updated it to latest firmware possible (updated it again just right now just in case) and configured it again like the DYI I had before.

[image]

It's working fine for the NRF24 part, but for CMT2300A part it still doesn't work.

One thing to note is that inverter is detected every morning when sun is up:
[image]

[image]

So I guess like I read in another issue threads, that the inverter is waking up on the 868Mhz and then is changing frequency. 
Another important thing is that maybe the openDTU is loosing the track of the new frequency because another device is telling the inverter to change it.
This device is the device from the integrator I bought the inverter+photovoltaic panels from "Beem energy" to be able to have the production in their proprietary app.
What do you think ? What could I try to still get the values from openDTU ? :pray:

I also tried using my "nooelec rtl-sdr v5" and GQRX to find the new frequency, but I don't really know which pattern to look for :/

Thank you very much for your time, it's greatly appreciated :)

stefan123t, 2024-08-09:
@tt-why so having two DTUs querying the same inverter in parallel is never a good idea, for multiple reasons with the Hoymiles Protocol and as you hinted the frequency set by thebother DTU.
So you should always use only one DTU with your inverter(s).

I believe the base frequency is 865 MHz so some +/-5 MHz should be your range in SDR.

Regarding the OpenDTU Fusion board this should work out of the box, though you may need to use the esp32s3 build and the right pin_mapping for it. Can you check your GPIO3 connection is configured ?

Also verify the sending Power / Amplification of the CMT settings may need to be raised from the default 0dBm.

tt-why, 2024-08-10:
Regarding the other DTU, it's the one from the manufacturer because it's plug&play (you just have to place the PV panels somewhere then just plug it on some electrical wall socket and that's it) that's why I cannot remove it :/ 
It's working like that for my others 3 Hoymiles inverter from the same manufacturer (HM-300) it's just using NRF24 instead of CMT2300A.

According to GQRX today I've got something at 868.2Mhz so i've put theses settings (and increased a bit sending power), full disclosure it doesn't work with theses settings on for more than 15 minutes (another question do you have an idea how much time I should wait until changing again the parameters, to have a chance if I fund the correct freqency for it to work ?):
[image]

But really hard to guess if it's the inverter signal or not :/
[image]

According to the opendtu fusion doc [here](https://github.com/markusdd/OpenDTUFusionDocs/blob/main/OTA.md#ota-tutorial) I have to use "opendtu-generic_esp32s3_usb.bin" version, do you think I should try "opendtu-generic_esp32s3.bin" version instead ?

Regarding GPIO3 it seems to be configured correctly isn't it ?
[image]

Another totally different idea is that it's maybe because openDTU recognize my HMS-400BM as another kind of device "HMS-500-1T v2" because of the serial number prefix "1125920xxxxx" ?

Thank you :pray:

tt-why, 2024-08-10:
Regarding the other DTU, it's the one from the manufacturer because it's plug&play (you just have to place the PV panels somewhere then just plug it on some electrical wall socket and that's it) that's why I cannot remove it :/ 
It's working like that for my others 3 Hoymiles inverter from the same manufacturer (HM-300) it's just using NRF24 instead of CMT2300A.

According to GQRX today I've got something at 868.2Mhz so i've put theses settings (and increased a bit sending power), full disclosure it doesn't work with theses settings on for more than 15 minutes (another question do you have an idea how much time I should wait until changing again the parameters, to have a chance if I fund the correct freqency for it to work ?):
[image]

But really hard to guess if it's the inverter signal or not :/
[image]

According to the opendtu fusion doc [here](https://github.com/markusdd/OpenDTUFusionDocs/blob/main/OTA.md#ota-tutorial) I have to use "opendtu-generic_esp32s3_usb.bin" version, do you think I should try "opendtu-generic_esp32s3.bin" version instead ?

Regarding GPIO3 it seems to be configured correctly isn't it (i'm using https://raw.githubusercontent.com/tbnobody/OpenDTU/master/docs/DeviceProfiles/opendtu_fusion.json) ?
[image]

Another totally different idea is that it's maybe because openDTU recognize my HMS-400BM as another kind of device "HMS-500-1T v2" because of the serial number prefix "1125920xxxxx" ?

Thank you :pray:

stefan123t, 2024-08-10:
OpenDTU or Hoymiles in general uses only the last eight digits of your Serial ID when communicating with the Inverter via the radio link, ie both NRF and CMT.

is there a significant difference between the HMS-500-1T v2 and your HMS-400BM ? What does the BM stand for ?

Yes the settings for the CMT look good, as you can verify under Info > System afaik.

For the fusion board you need an esp32s3 binary. I usually use the USB cable and flash the factory image first. If you want to update via OTA you may use the one without USB afaik but USB has some USB debugging too. If you want to see the communication with your inverter you may also check the System > Console which should give you no response as far as you are either not hitting the right base frequency on the inverter ir some other issue in communication.

stefan123t, 2024-08-10:
I only find Certification Results of the HMS-400BM in spanish. Is this eventually a Brazilian Market model ? I believe for the North American and South American markets there are other frequencies which have to be used. Please check for the other issues about North American we had to fix the software to support these models.

stefan123t, 2024-08-15:
@tt-why can you provide a feedback on the last questions ?

tt-why, 2024-08-15:
Hi and sorry for the delay,

> is there a significant difference between the HMS-500-1T v2 and your HMS-400BM ? What does the BM stand for ?
I honestly can't find anywhere what the BM suffix is meaning :/

> Is this eventually a Brazilian Market model ?
I'm currently in France and bought this inverter from a french company called Beem energy selling in France also.

And interesting thing is that since I've increased CMT2300A Transmitting power from 0 to 5dB (like you suggested), I've been able to have the data sporadically:
- 10/08: data once at 08PM
- 11/08: data once at 05PM
- 12/08: no data
- 13/08: data once at 03PM
- 14/08: data once at 02PM

So I've just tried to keep 868.25Mhz and cranked up the transmitting power from 5dB to 20dB hopping it might be better received by the inverter in the following days :pray: 
Do you think I would have better results by changing to default 865MHz in EU ?

Thanks !

tt-why, 2024-08-15:
Another thing I've just tried is to stick the antenna of the SDR really close to the inverter (around 30 centimeters) to find the correct freq, and I'm almost 100% sure this big blast if the inverter so around 868.325MHz.
[image]

So even I choosed 868.25MHz in the opendtu settings no luck so far :/

ruy-silva, 2024-11-20:
Hi @tt-why do you manage to get this work with the HMS-400BM inverter?

stefan123t, 2024-12-16:
We have another user with an HMS-400BM from French importer BEEM in Nantes. See #2458 for details. We will need to have the hardware and firmware version numbers to identify the model with Serial `1400A01…..`

zolicozk, 2025-01-26:
Hi all, I have exactly the same issue.
Inverter provided by BEEM but I do not have CMT hardwarem only nRF24.

[image]

closed 2025-04-18 by github-actions[bot]