tbnobody/OpenDTU #2178: _isAllEnabledProducing and _isAllEnabledReachable are true without connection to inverter

AloisKlingler, 2024-08-01:
### What happened?

Datastore.getIsAllEnabledReachable()
and
Datastore.getIsAllEnabledProducing()

are returning true, even if no inverter is reachable (chip is not connected).

not sure if this happens with connected CMT / NRF24 as well but regarding Datastore.cpp I think Line 50 / line 51 are maybe misplaced.

### To Reproduce Bug

install opendtu on an ESP32 without CMT / NRF24 connected

### Expected Behavior

Datastore.getIsAllEnabledReachable()
and
Datastore.getIsAllEnabledProducing()
should return false if inverters are not reachable.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.29

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-08-01:
Thats right. The Datastore function only looks at inverters which are added to the Hoymiles object. This only happens if a valid RF interface for the specific inverter is present. 
In other words, only valid inverters are analyzed. A inverter which is not supported by your hardware is considered as not valid.

AloisKlingler, 2024-08-02:
@tbnobody
line 50 + 51 just below the "for" would not help?
At least both variables are initialized with false but look just be set too early to true...

tbnobody, 2024-08-02:
Nope... Because `getNumInverters` returns only the amount of valid inverters. Therefor the for loop never gets entered and isProducing and isReachable stays zero.

AloisKlingler, 2024-08-02:
I See, thanks for Explanation 😎