tbnobody/OpenDTU #2180: crashes especially after firmware update at tcp_update_rcv_ann_wnd

AloisKlingler, 2024-08-01:
### What happened?

reproduceable with two ESP32-D0WD-V3:

after firmware update I can see web browser (chrome) trying to redirect to the start page.

it looks like, if the requests come too early, the ES likes to crash, especially with the following trace:

WiFi connected
WiFi got ip: 192.168.0.133
Network connected

assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)

Backtrace: 0x40083a65:0x3ffb2cc0 0x4008ee3d:0x3ffb2ce0 0x40095151:0x3ffb2d00 0x40134bfe:0x3ffb2e30 0x40134cac:0x3ffb2e50 0x401a82ee:0x3ffb2e70 0x40131464:0x3ffb2e90

ELF file SHA256: 191a2d1569b60854

E (8912) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (682) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU

regarding this I found:

https://github.com/mathieucarbou/AsyncTCP/issues/14

just fixed a few days ago.

### To Reproduce Bug

after firmware update on browser redirecting to openDTU start page

### Expected Behavior

should not crash. :-)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.6.29

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-08-01:
Yes. I know. Thats either a issue in the ESP32 TCP stack or another library. Nothing which can be fixed in OpenDTU.

AloisKlingler, 2024-08-01:
> Yes. I know. Thats either a issue in the ESP32 TCP stack or another library. Nothing which can be fixed in OpenDTU.

the fix from https://github.com/mathieucarbou/AsyncTCP/issues/14 in asynctcp does also not help?

tbnobody, 2024-08-01:
Looks like the right solution. Will check soon

tbnobody, 2024-08-01:
Thanks for the hint. Implemented in 03137e15dd80c991da289fedfd89a18aa2b69bbf and seems to fix the issue.

closed 2024-08-01 by tbnobody