tbnobody/OpenDTU #2181: Version v24.8.1 läst sich nicht mit VSC neu bauen.

Juergen2453, 2024-08-02:
### What happened?

Reichlich Fehlermeldungen

### To Reproduce Bug

neu bauen

### Expected Behavior

neu bauen

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

 v24.8.1

### Relevant log/trace output

```Shell
Compiling .pio\build\generic_esp32\src\MqttHandleDtu.cpp.o
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from include/PinMapping.h:4,
                 from include/Configuration.h:4,
                 from src/Configuration.cpp:5:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\Configuration.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from include/PinMapping.h:4,
                 from include/Led_Single.h:4,
                 from src/Led_Single.cpp:5:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\Led_Single.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,    
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from .pio/libdeps/generic_esp32/ESPAsyncWebServer/src/AsyncWebSocket.h:24,
                 from include/MessageOutput.h:4,
                 from src/MessageOutput.cpp:5:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\MessageOutput.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from .pio/libdeps/generic_esp32/U8g2/src/U8g2lib.h:49,
                 from include/Display_Graphic_Diagram.h:5,
                 from include/Display_Graphic.h:4,
                 from src/Display_Graphic.cpp:5:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\Display_Graphic.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from .pio/libdeps/generic_esp32/U8g2/src/U8g2lib.h:49,
                 from include/Display_Graphic_Diagram.h:5,
                 from src/Display_Graphic_Diagram.cpp:5:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\Display_Graphic_Diagram.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,    
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from include/PinMapping.h:4,
                 from include/Configuration.h:4,
                 from src/Datastore.cpp:6:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\Datastore.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from include/PinMapping.h:4,
                 from include/Configuration.h:4,
                 from src/InverterSettings.cpp:6:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\InverterSettings.cpp.o] Error 1
In file included from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:83,
                 from C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,      
                 from include/PinMapping.h:4,
                 from include/Configuration.h:4,
                 from src/MqttHandleDtu.cpp:6:
C:/Users/privat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\generic_esp32\src\MqttHandleDtu.cpp.o] Error 1
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.

tbnobody, 2024-08-02:
Das ist ein Problem deiner Entwicklungsumgebung und nicht von OpenDTU. Die `pins_arduino.h` gehört sich zum Arduino Framework und nicht zu den OpenDTU Sourcen.
Aber schaue ggf. mal hier (letzter Punkt): https://www.opendtu.solar/firmware/faq/#build-errors

Juergen2453, 2024-08-02:
Danke für die Antwort. Bisher hatte ich null Probleme, Vielleicht hat sich ja bei mir etwas eingeschlichen.

Gestern Abend noch ein Update von deinen "[opendtu-generic_esp32.bin]" gemacht,
und Heute morgen ist die DTU nicht aufgewacht (ca. 10Std Laufzeit).
DTU musste aus/ein geschaltet werden. Werde es weiter Beobachten.
(So etwas hatte ich schon einmal nach einem Update, danach lief alles wieder normal)

Juergen2453, 2024-08-02:
Super er baut wieder.
dein Hinweis "Aber schaue ggf. mal hier (letzter Punkt): https://www.opendtu.solar/firmware/faq/#build-errors" hat geholfen.

Wenn nichts dagegen spricht dann CLOSE

closed 2024-08-02 by Juergen2453