tbnobody/OpenDTU #2186: [Request] IPv6 support

imp1sh, 2024-08-03:
### Describe the solution you'd like

I'd wish OpenDTU supported IPv6 so it can integrate into what the standard is today. 

### Additional context

Autoconf and static configuration should work. A DHCPv6 client would be nice to have.

broth-itk, 2024-08-06:
I think the implementation and ressource usage of IPv6 does not really match the usecases in the real world.
I'd say it is going to have a very low priority, likely to be dismissed.

imp1sh, 2024-08-16:
Well of course people are using IPv6. If you think otherwise, look at

- https://www.rfc-editor.org/rfc/rfc6540.html
- https://www.google.com/intl/en/ipv6/statistics.html

broth-itk, 2024-08-16:
@imp1sh 
I don't know if stats of public web services support the need for IPv6 in private networks.

@straaat 
IPv6 only internal network? Guess OpenDTU is not the only think not working correctly.

Again, I'm not against IPv6 in OpenDTU but from a developers/ressource usage perspective, it makes no sense.
Maybe a vote among all OpenDTU users would make sense like "Who had problems OpenDTU not supporting IPv6".

imp1sh, 2024-08-16:
Well this might be new to you but when you run IPv6 you will have public addresses in your home network and NO, this is not a security risk. Well it seems you don't know very much about networks. Using IPv6 is not a rare thing as you seem to believe. It has become the default.

broth-itk, 2024-08-16:
Yes, IPv6 is very new to me, has been released just a "few" years ago. *facepalm*

I won't join this discussion as it's missing the point.
If you feel IPv6 is so important to you, please go ahead and do some research, implementation and propose a pull of your code.

straaat, 2024-08-16:
I do use ipv4 as well but had not setup a DNS name for my MQTT server’s ipv4 address. 

Not a big deal but just wanted to share my experience. If this was a commercial product, I would send it back. I appreciate the work the devs put in and just want to let them know that I do value ipv6 support. I’m very happy with the software as is.

stefan123t, 2024-08-17:
According to this it is almost simplistic as IPv6 support has been added to the WiFi library we are already using. You just have to enable it ;)

**Exploring IPv6 capabilties of microcontrollers like the ESP8266 and ESP32**
https://ipv6.net/blog/exploring-ipv6-capabilities-of-microcontrollers-like-the-esp8266-and-esp32/

```c++
  // Enable IPv6
  WiFi.enableIpV6();

  // Print the ESP32's IPv6 address
  Serial.print("IPv6 Address: ");
  Serial.println(WiFi.localIPv6().toString());
```

broth-itk, 2024-08-17:
If its that easy and if the required code is already present - why not give it a try.

There might be a delay between enabling IPv6 and SLAAC to be completed.
The IPv6 address might not immediately be available.

Apart from enbling IPv6 on Wifi, all configuration parts using addresses or hostnames should be checked to make sure IPv6 addresses are accepted and working.

straaat, 2024-08-17:
I tried. The ESP gets a ipv6 address but MQTT does not work. I think the MQTT library might not support it.

imp1sh, 2024-08-17:
Thank you so much for trying @straaat 
Any chance you could try if the integrated webservice would be v6 compatible? I query the prometheus exporter at /api/prometheus/metrics on port 80.

straaat, 2024-08-17:
I'm not able to reach the webserver. So far only ping responds.

imp1sh, 2024-08-18:
I wasn't able to find where the webserver is spawned in the source code. It might be that by default it will only listen on `0.0.0.0`. It should be `::` instead so that it will bind to both tcp4 and tcp6.

stefan123t, 2024-09-07:
According to this upstream discussion eg Tasmota has implemented IPv6 some 8 months ago based on the [PR](https://github.com/espressif/arduino-esp32/pull/9016)

stumbaumr, 2025-12-11:
I just got fiber at home and run Home Assistant with the Mosquitto AddOn to collect OpenDTU statistics for two remote installations. The provider only gives out IPv6 and CGNAT IPv4 - and charges a fee for real IPv4.

Is that a valid use case to ask for the addition of IPv6?

Having the same [problem with the OVMS](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3) in my smart EQ...