tbnobody/OpenDTU #2191: nRF24 Chip-Status not connected

j0na553n, 2024-08-06:
### What happened?

After Update the Chip ist not connected

### To Reproduce Bug

Update

### Expected Behavior

-

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-08-06:
Provide some screenshots of the info --> system page.

j0na553n, 2024-08-06:
Hi of course here are some screenshots

[image]
[image]
[image]

After downgrade the fault is also there.

tbnobody, 2024-08-06:
Seems your wiring of the NRF module is somehow damaged.

j0na553n, 2024-08-06:
oh well this is possible, i have to check. 
Thanks for you hint. 
Do you know is it possible to user an other version of opendtu?

j0na553n, 2024-08-06:
After a Physical reboot and check of contacts everything works fine.

closed 2024-08-06 by j0na553n