tbnobody/OpenDTU #2197: Hi,

bastianbugla85, 2024-08-08:
Hi,
Ich habe heute meinen Wechselrichter HMT 2250-6T in Betrieb genommen und mit meiner OpenDTU verbunden.
Ich kann den WR in der Weboberfläche sehen, aber bekomme keine Daten angezeigt.

Systeminformationen:
SDK-Version	v4.4.7-dirty
Konfigurationsversion	0.1.28
Firmwareversion / git Hash	[v24.8.5](https://github.com/tbnobody/OpenDTU/releases/tag/v24.8.5)
PIO Umgebung	generic_esp32

Hardwareinformationen:
Chip-Modell | ESP32-D0WD-V3
Chip-Revision | 3

Der Wechselrichter läuft laut Betriebshandbuch korrekt.
[image]

Hat hier jemand eine Idee? Falsche Firmware? 

Vielen Dank vorab schonmal für eine Idee

_Originally posted by @bastianbugla85 in https://github.com/tbnobody/OpenDTU/discussions/2192_

stefan123t, 2024-08-08:
Issue is duplicate of discussion in #2192 
Problem cause between chair and keyboard GPIO3 is neither configured nor clear whether it was soldered / connected to the CMT2300A radio chip. Please fix your setup and close the issue. Thanks

bastianbugla85, 2024-08-08:
Vielen Dank @mroenne2022 ich schaus mir genauer an... Ich bin zugegebener Maßen komplett neu auf dem Gebiet.
Vielen Dank auch dir @stefan123t für den Input, auch wenn ich davon noch nicht viel verstehe.

closed 2024-08-08 by bastianbugla85

bastianbugla85, 2024-08-09:
This was the solution in my case for an approx distance from DTU to inverter of 10m.
[image]
no config change needed. 
Thanks again for your support