tbnobody/OpenDTU #2211: "Poll interval" - Fix documentation or code

broth-itk, 2024-08-15:
### What happened?

The poll interval is a bit confusing IMHO.

With a single inverter the interval specifies the time between poll requests. Data is fetched and published with that exact interval.
So far so good.

With two or more inverters things get tricky. The poll time for one of the inverters is **poll interval * inverter count**
Example: Three inverters, poll interval = 5, data is updated every 15 seconds.

If you still want data of all inverters get updated each 5 seconds, you'd had to set a poll interval of 1,67 seconds.

Maybe I'm wrong but I guess most would expect all inverters to get polled in the same interval.

Like that all inverters are polled at the same time each 5 seconds.

There might be a good reason why this has been done but I can't see this behaviour documented somewhere.

### To Reproduce Bug

Use more than one inverter and observe the interval in which data gets updated.

### Expected Behavior

Poll all inverters one after another without delay each "poll interval" seconds.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-08-16:
The manual states:
> The time (in seconds) between the individual inverter queries. If you choose this value to small the ESP does not have enough time to perform other tasks.

I think thats pretty clear. If you like a different wording here, please provide a PR in the OpenDTU-docs repository

closed 2024-08-16 by tbnobody

broth-itk, 2024-08-16:
Since all other options have a nice tooltip, adding another for "poll interval" might help clarifying even further.
Not everyone is Reading...TFM