tbnobody/OpenDTU #2212: [Request] do not use night time if date is in the year 1970

ke352802081770314, 2024-08-16:
### Is your feature request related to a problem? Please describe.

i am new to openDTU, yesterday was the first time I connected to the inverter and saw the first "electricity harvest". For reasons, i unplugged it during the night. when I powered it up this early morning, there was no Wifi and NTP server available (cable modem failure).
I was puzzled that the display keeps saying 0W at daybreak. date showed 01/01/1970 which I dismissed as irrelevant and was expected due to internet outage. When connecting to the setup AP and manually setting the date/time (with the browser to device sync button - very handy, thank you!), it connected to the inverter and showed data.

### Describe the solution you'd like

seeing that no NTP means the time is almost certainly incorrect, I suggest when the year is 1970, do not obey the night time rule, act as if it was daytime always. 
This gives better satisfaction in adverse circumstances.

tbnobody, 2024-08-16:
Ignoring the day/night mode would not fix your the problem. The current time is sent to the inverter with every request. Therefor requests are only sent when a valid time is available (independed of day/night mode)

closed 2024-08-16 by tbnobody

ke352802081770314, 2024-08-16:
ah, i didn't realize that. in that case, "0W" looks like a valid value but if the DTU isn't even prepared to talk to the inverter, i would prefer that it displays "---" or something else that shows it has no, and doesn't even want to establish, a connection. but that would be a different request.