tbnobody/OpenDTU #2223: 2 HM WR - HMS-2000-4T wird nicht erkannt  der HM-1500-4T aber schon

is58fra, 2024-08-23:
### What happened?

Habe 2 Hoymiles wechselrichter.
1x HM-1500-4T
1x HMS-2000-4T

OpenDTU Version: 24.8.5

Die Daten von dem HM-1500 bekomme ich.
Die vom HMS-2000 nicht.

Die WR sind nicht weiter voneininder entfernt.

[image]

Der HMS-2000-4T läuft led blickt grün wie beim HM-1500-4T

### To Reproduce Bug

Neustart vom WR und OpenDTU

### Expected Behavior

Das er WR von der OpenDTU ausgelesen werden kann

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

is58fra, 2024-08-23:
ESP32 + NRF24

ok verstehe danke für die Info 
Mein Fehler Entschuldigung!

mroenne2022, 2024-08-23:
OK, dann benötigst du wohl eine OpenDTU Fusion !

closed 2024-08-23 by tbnobody