tbnobody/OpenDTU #2236: Support for HERF-300 inverter

xenio, 2024-08-28:
### Is your feature request related to a problem? Please describe.

I just received a micro invert from ESTAR ENERGY, it is a HERF-300.

I added the serial  and the form accepted and converted it, but I get UNKNOWN TYPE.

Is it possible to add support for this model?
Thanks.

[image]

[image]

[image]

[image]

[image]

### Describe the solution you'd like

I don't have a solution :(

### Additional context

The HERF-300 micro inverter it is working with HERF-DCU but I'll love to get off their cloud, and there is no integration with Home Assistant or MQTT.

tbnobody, 2024-08-28:
Are you able to compile the firmware by yourself? Otherwise I will provide you a binary which includes a first detection. The interpretation of the data will still be wrong as I need the console output from your side if it's able to send and receive data.

xenio, 2024-08-28:
@tbnobody yep I can compile the source code (I am on MacOS).

Do I need to change something?

tbnobody, 2024-08-29:
> Do I need to change something in the code?

Yes. Please add at this line:
https://github.com/tbnobody/OpenDTU/blob/0cc55f3b8723723aea234954eaf7ab6cd13b53a4/lib/Hoymiles/src/inverters/HM_1CH.cpp#L37

the following:

```cpp
    uint16_t preSerial = (serial >> 32) & 0xffff;
    if (preSerial == 0x2841) {
        return true;
    }
```

After this the inverter should be recognized as `HM-300/350/400-1T`.
In the console you should see TX and RX lines.

Also the inverter should be visible at the Live View.
But I would expect that the properties just show strange values.
Please provide a screenshot of the Live View and also from the console.
We need the output of those lines:
```
TX RealTimeRunData Channel: 23 --> 15 71 60 35 46 80 12 23 04 80 0B 00 66 D0 96 B2 00 00 00 00 00 00 00 00 44 4B D4
Interrupt received
RX Channel: 75 --> 95 71 60 35 46 71 60 35 46 01 00 01 01 42 00 D5 01 4E 02 AD 04 35 00 11 16 55 80 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 71 60 35 46 71 60 35 46 02 00 11 0A F3 08 89 08 92 01 3D 01 58 00 02 04 40 47 | -80 dBm
Interrupt received
RX Channel: 61 --> 95 71 60 35 46 71 60 35 46 03 00 07 00 11 3C CE 00 00 19 5E 08 81 00 07 09 0D BF | -80 dBm
Interrupt received
RX Channel: 61 --> 95 71 60 35 46 71 60 35 46 84 13 86 0A 9A 00 D3 00 75 03 E5 01 ED 00 02 D2 5E 36 | -80 dBm
Interrupt received
RX Period End
Success
```

xenio, 2024-08-29:
Ok, I added the code and now it is recognised but it is not connecting :(

[image]
[image]

```
18:38:58.823 > Starting OpenDTU
18:38:58.823 > Initialize FS... done
18:38:58.832 > Reading configuration... done
18:38:58.959 > Reading PinMapping... found valid mapping done
18:38:59.048 > Initialize Network... done
18:38:59.175 > Setting Hostname... Configuring WiFi STA using new credentials... done
18:38:59.175 > Initialize NTP... done
18:38:59.175 > Initialize SunPosition... done
18:38:59.175 > Initialize MqTT... done
18:38:59.175 > Initialize WebApi... done
18:38:59.186 > Initialize Display... done
18:38:59.186 > Initialize LEDs... done
18:38:59.186 > Check for default DTU serial... done
18:38:59.192 > Initialize Hoymiles interface... NRF: Connection successful
18:38:59.201 >   Setting radio PA level... 
18:38:59.201 >   Setting DTU serial... 
18:38:59.206 >   Setting poll interval... 
18:38:59.206 >   Adding inverter: 28410000C9DD - HERF-300 done
18:38:59.212 > done
18:38:59.212 > Switch to WiFi mode
18:38:59.215 > Setting Hostname... done
18:38:59.234 > Configuring WiFi STA using new credentials... done
18:38:59.244 > Configuring WiFi STA DHCP IP... done
18:39:02.161 > WiFi connected
18:39:02.220 > WiFi got ip: 192.168.1.126
18:39:02.220 > Network connected
18:39:02.220 > Connecting to MQTT...
18:39:02.271 > Connected to MQTT.
18:39:03.784 > Fetch inverter: 28410000C9DD
18:39:03.792 > TX RealTimeRunData Channel: 23 --> 15 00 00 C9 DD 80 15 43 20 80 0B 00 66 D0 A4 27 00 00 00 00 00 00 00 00 27 79 17 
18:39:04.333 > RX Period End
18:39:04.333 > All missing
18:39:04.333 > Nothing received, resend whole request
18:39:04.338 > TX RealTimeRunData Channel: 40 --> 15 00 00 C9 DD 80 15 43 20 80 0B 00 66 D0 A4 27 00 00 00 00 00 00 00 00 27 79 17 
18:39:04.885 > RX Period End
18:39:04.885 > All missing
18:39:04.885 > Nothing received, resend whole request
18:39:04.890 > TX RealTimeRunData Channel: 61 --> 15 00 00 C9 DD 80 15 43 20 80 0B 00 66 D0 A4 27 00 00 00 00 00 00 00 00 27 79 17 
18:39:05.437 > RX Period End
18:39:05.437 > All missing
18:39:05.437 > Nothing received, resend whole request
18:39:05.442 > TX RealTimeRunData Channel: 75 --> 15 00 00 C9 DD 80 15 43 20 80 0B 00 66 D0 A4 27 00 00 00 00 00 00 00 00 27 79 17 
18:39:05.989 > RX Period End
18:39:05.989 > All missing
18:39:05.989 > Nothing received, resend whole request
18:39:05.994 > TX RealTimeRunData Channel: 3 --> 15 00 00 C9 DD 80 15 43 20 80 0B 00 66 D0 A4 27 00 00 00 00 00 00 00 00 27 79 17 
18:39:06.541 > RX Period End
18:39:06.541 > All missing
18:39:06.541 > Nothing received, resend count exeeded
18:39:06.546 > TX AlarmData Channel: 23 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 27 00 00 00 00 00 00 00 00 FD 62 CC 
18:39:07.336 > RX Period End
18:39:07.336 > All missing
18:39:07.336 > Nothing received, resend whole request
18:39:07.342 > TX AlarmData Channel: 40 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 27 00 00 00 00 00 00 00 00 FD 62 CC 
18:39:08.138 > RX Period End
18:39:08.138 > All missing
18:39:08.138 > Nothing received, resend whole request
18:39:08.143 > TX AlarmData Channel: 61 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 27 00 00 00 00 00 00 00 00 FD 62 CC 
18:39:08.940 > RX Period End
18:39:08.940 > All missing
18:39:08.940 > Nothing received, resend whole request
18:39:08.946 > TX AlarmData Channel: 75 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 27 00 00 00 00 00 00 00 00 FD 62 CC 
18:39:09.741 > RX Period End
18:39:09.741 > All missing
[... 24 lines cut ...]
18:39:13.331 > RX Period End
18:39:13.331 > All missing
18:39:13.331 > Nothing received, resend count exeeded
18:39:13.337 > TX AlarmData Channel: 23 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 2E 00 00 00 00 00 00 00 00 AD 08 FF 
18:39:14.126 > RX Period End
18:39:14.126 > All missing
18:39:14.126 > Nothing received, resend whole request
18:39:14.132 > TX AlarmData Channel: 40 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 2E 00 00 00 00 00 00 00 00 AD 08 FF 
18:39:14.927 > RX Period End
18:39:14.927 > All missing
18:39:14.927 > Nothing received, resend whole request
18:39:14.933 > TX AlarmData Channel: 61 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 2E 00 00 00 00 00 00 00 00 AD 08 FF 
18:39:15.729 > RX Period End
18:39:15.729 > All missing
18:39:15.729 > Nothing received, resend whole request
18:39:15.735 > TX AlarmData Channel: 75 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 2E 00 00 00 00 00 00 00 00 AD 08 FF 
18:39:16.531 > RX Period End
18:39:16.531 > All missing
18:39:16.531 > Nothing received, resend whole request
18:39:16.537 > TX AlarmData Channel: 3 --> 15 00 00 C9 DD 80 15 43 20 80 11 00 66 D0 A4 2E 00 00 00 00 00 00 00 00 AD 08 FF 
18:39:17.333 > RX Period End
18:39:17.333 > All missing
18:39:17.333 > Nothing received, resend count exeeded
18:39:17.335 > Fetch inverter: 28410000C9DD
```

tbnobody, 2024-08-29:
What is the original serial number of the inverter before  any conversion? (So whats shown on the label on the inverter?) I assume the 2841xxxxxx was generated by the WebGUI. Maybe there is another logic how the serial number is created for this kind of model.

xenio, 2024-08-29:
Here it is...

SN: A21001JEW-1S2
Type: Micro
Model: HERF-300
Hardware Ver: H00.04.00
Software Ver: V01.00.02

xenio, 2024-08-30:
@tbnobody I got some data from the inverter, for a couples of seconds datas was available in the live view but after that now it is in "nothing received", "all missing", "last missing", "middle missing"

Maybe it is related to #2137 ?

I have two chinese NRF24L01+ I am trying with capacitor / cables etc with no luck.

tbnobody, 2024-08-30:
> Maybe it is related to https://github.com/tbnobody/OpenDTU/issues/2137 ?

Nope. NRF24 and CMT module uses completely different frequency handling.

xenio, 2024-08-30:
It is almost working now... with transmitting power to maximum 0dbm, but the data received does not match the dashboard information :(

Maybe 

[image]
[image]
[image]
[image]

```
13:20:27.091 > Last missing
13:20:27.091 > Request retransmit: 2
13:20:27.091 > TX RequestFrame Channel: 23 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:27.138 > RX Period End
13:20:27.138 > Last missing
13:20:27.138 > Request retransmit: 2
13:20:27.138 > TX RequestFrame Channel: 40 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:27.215 > RX Period End
13:20:27.215 > Last missing
13:20:27.215 > Request retransmit: 2
13:20:27.215 > TX RequestFrame Channel: 61 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:27.245 > Interrupt received
13:20:27.330 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 02 19 DE 00 00 00 00 01 71 00 00 09 21 13 88 06 07 92 | -80 dBm
13:20:27.439 > RX Period End
13:20:27.439 > Last missing
13:20:27.439 > Request retransmit: 3
13:20:27.439 > TX RequestFrame Channel: 75 --> 15 00 00 C9 DD 80 19 19 52 83 50 
13:20:27.695 > Interrupt received
13:20:27.721 > RX Channel: 23 --> 95 00 00 C9 DD 00 00 C9 DD 83 00 00 00 42 03 E7 01 80 01 A8 16 29 A7 | -80 dBm
13:20:27.758 > RX Period End
13:20:27.758 > Success
13:20:27.781 > TX AlarmData Channel: 3 --> 15 00 00 C9 DD 80 19 19 52 80 11 00 66 D1 AA F9 00 00 00 00 00 00 00 00 51 D6 21 
13:20:27.798 > Websocket: [/livedata][5] disconnect
13:20:28.283 > RX Period End
13:20:28.283 > All missing
13:20:28.283 > Nothing received, resend whole request
13:20:28.283 > TX AlarmData Channel: 23 --> 15 00 00 C9 DD 80 19 19 52 80 11 00 66 D1 AA F9 00 00 00 00 00 00 00 00 51 D6 21 
13:20:29.143 > RX Period End
13:20:29.143 > All missing
13:20:29.143 > Nothing received, resend whole request
13:20:29.143 > TX AlarmData Channel: 40 --> 15 00 00 C9 DD 80 19 19 52 80 11 00 66 D1 AA F9 00 00 00 00 00 00 00 00 51 D6 21 
13:20:29.214 > Interrupt received
13:20:29.234 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 01 00 01 80 01 00 01 46 23 46 23 00 00 00 00 80 02 97 | -80 dBm
13:20:29.261 > Interrupt received
13:20:29.318 > RX Channel: 40 --> 95 00 00 C9 DD 00 00 C9 DD 04 00 07 80 02 01 9D 98 35 98 35 FF FF FF FA 80 02 0F | -80 dBm
13:20:29.406 > Interrupt received
13:20:29.433 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 07 1C 2E 80 02 01 A1 9B DE 9B DE FF FF E3 DA 90 02 29 | -80 dBm
13:20:29.524 > Interrupt received
13:20:29.545 > RX Channel: 75 --> 95 00 00 C9 DD 00 00 C9 DD 09 9B E9 FF FF E3 D5 90 02 01 A4 9B E9 9B E9 00 00 EF | -80 dBm
13:20:29.578 > Interrupt received
13:20:29.601 > RX Channel: 40 --> 95 00 00 C9 DD 00 00 C9 DD 0B 01 A6 9C 02 9C 02 00 00 1C 26 80 02 01 A7 9C 06 BD | -80 dBm
13:20:29.871 > RX Period End
13:20:29.871 > Last missing
13:20:29.871 > Request retransmit: 12
13:20:29.871 > TX RequestFrame Channel: 61 --> 15 00 00 C9 DD 80 19 19 52 8C 5F 
13:20:29.908 > Interrupt received
13:20:29.943 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 8C 9C 06 FF FF E3 D5 B7 F3 F1 | -80 dBm
13:20:29.972 > RX Period End
13:20:29.972 > Middle missing
13:20:29.972 > Request retransmit: 2
13:20:29.972 > TX RequestFrame Channel: 75 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:30.106 > RX Period End
13:20:30.106 > Middle missing
13:20:30.106 > Request retransmit: 2
13:20:30.106 > TX RequestFrame Channel: 3 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:30.223 > RX Period End
13:20:30.223 > Middle missing
13:20:30.223 > Request retransmit: 2
13:20:30.223 > TX RequestFrame Channel: 23 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:30.397 > RX Period End
13:20:30.397 > Middle missing
13:20:30.397 > Request retransmit: 2
13:20:30.397 > TX RequestFrame Channel: 40 --> 15 00 00 C9 DD 80 19 19 52 82 51 
13:20:30.430 > Interrupt received
13:20:30.449 > RX Channel: 75 --> 95 00 00 C9 DD 00 00 C9 DD 02 01 9A 98 1B 98 1B FF FF E3 DB 80 02 01 9B 98 29 9D | -80 dBm
13:20:30.490 > RX Period End
13:20:30.490 > Middle missing
13:20:30.490 > Retransmit timeout
13:20:31.415 > Fetch inverter: 28410000C9DD
13:20:31.500 > TX RealTimeRunData Channel: 61 --> 15 00 00 C9 DD 80 19 19 52 80 0B 00 66 D1 AA FF 00 00 00 00 00 00 00 00 2B E6 77 
13:20:31.531 > Interrupt received
13:20:31.553 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 01 00 01 01 29 00 00 02 21 00 00 06 59 00 00 00 00 C1 | -80 dBm
13:20:31.594 > Interrupt received
13:20:31.626 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 02 19 DE 00 00 00 00 01 71 00 00 09 1E 13 88 06 06 AC | -80 dBm
13:20:31.681 > Interrupt received
13:20:31.722 > RX Channel: 75 --> 95 00 00 C9 DD 00 00 C9 DD 83 00 00 00 42 03 E7 01 80 01 A8 FB E2 81 | -80 dBm
13:20:31.752 > RX Period End
13:20:31.752 > Success
13:20:31.774 > TX AlarmData Channel: 75 --> 15 00 00 C9 DD 80 19 19 52 80 11 00 66 D1 AA FF 00 00 00 00 00 00 00 00 F1 FD AC 
13:20:31.906 > Interrupt received
13:20:31.958 > RX Channel: 23 --> 95 00 00 C9 DD 00 00 C9 DD 03 98 29 FF FF FF F9 80 02 01 9C 98 2F 98 2F 00 00 3E | -80 dBm
13:20:32.090 > Interrupt received
13:20:32.128 > RX Channel: 75 --> 95 00 00 C9 DD 00 00 C9 DD 07 1C 2E 80 02 01 A1 9B DE 9B DE FF FF E3 DA 90 02 29 | -80 dBm
13:20:32.379 > Interrupt received
13:20:32.409 > RX Channel: 40 --> 95 00 00 C9 DD 00 00 C9 DD 8C 9C 06 FF FF E3 D5 B7 F3 F1 | -80 dBm
13:20:32.594 > RX Period End
13:20:32.594 > Middle missing
13:20:32.594 > Request retransmit: 1
13:20:32.594 > TX RequestFrame Channel: 3 --> 15 00 00 C9 DD 80 19 19 52 81 52 
13:20:32.722 > RX Period End
13:20:32.722 > Middle missing
13:20:32.722 > Request retransmit: 1
13:20:32.722 > TX RequestFrame Channel: 23 --> 15 00 00 C9 DD 80 19 19 52 81 52 
13:20:32.813 > RX Period End
```

tbnobody, 2024-08-30:
> Maybe it is necessary to adapt the Herf-800 datas to the Herf-300 1ch.

Yes, I will to that using the information you provided. A new inverter class for the HERF-300 is required (like for the other HERF types) and the byte assignment has to be adjusted.

Will provide you a new patch soon. Does the event log show correct values? I see it shows 15 events in your screenshot

xenio, 2024-08-30:
I only took a part of the log where I saw a couple of ‘successes’, the log continues sending more or less always that kind of data. Do you need a longer log?

tbnobody, 2024-08-30:
> Do you need a longer log?

No, thats great

xenio, 2024-08-30:
> I guess you did turn off the HERF DCU already, as this would most likely interfere with OpenDTU connecting to the HERF-300 inverter.

Yes I turned it off, with the HERF DCU on I never got any data from the NRF24L01+
I also bought an EByte NRF24L01+ module, maybe it works better than the cheap ones I am using now.

tbnobody, 2024-08-30:
I analyzed the following snippet
```
13:20:31.500 > TX RealTimeRunData Channel: 61 --> 15 00 00 C9 DD 80 19 19 52 80 0B 00 66 D1 AA FF 00 00 00 00 00 00 00 00 2B E6 77 
13:20:31.531 > Interrupt received
13:20:31.553 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 01 00 01 01 29 00 00 02 21 00 00 06 59 00 00 00 00 C1 | -80 dBm
13:20:31.594 > Interrupt received
13:20:31.626 > RX Channel: 3 --> 95 00 00 C9 DD 00 00 C9 DD 02 19 DE 00 00 00 00 01 71 00 00 09 1E 13 88 06 06 AC | -80 dBm
13:20:31.681 > Interrupt received
13:20:31.722 > RX Channel: 75 --> 95 00 00 C9 DD 00 00 C9 DD 83 00 00 00 42 03 E7 01 80 01 A8 FB E2 81 | -80 dBm
13:20:31.752 > RX Period End
13:20:31.752 > Success
```

How realistic are the following values (based on the dump you pasted above) (I dont know how long you are already operating the inverter)

DC-Input Voltage: 29,7V
DC-Input Current: 5,45A
DC-Input Power: 162,5W

Yield Daily: 369Wh
Yield Total: 6,622kWh

AC Voltage: 233,4V
AC Current: 0,66A
AC Power: 154,2W
AC Reactive Power: ????°C
AC Power Factor: 0,999
AC Frequency: 50Hz

Temperature: ???? °C

[image]

The temperature was either 38,4°C or 42,4°C. Reactive Power also either 38,4var or 42,4var.

Therefor a additional data snippet e.g. at the morning when it's a little bit colder or at afternoon when it's really hot would be helpfull to distinct between the two values

xenio, 2024-08-30:
I have only been using the inverter for a few days, it is not yet installed on the roof.
Today I was switching between the HERF DCU and OpenDTU so I don't have precise data.

This is today screenshot from the Herf Monitor cloud.
Tomorrow I can get the json from the cloud to try to match some number.

[image]

tbnobody, 2024-08-30:
ah ok. but based on this screenshot my guess for yield total and yield daily is correct. the other values also make sense as e.g. Power = Voltage * Current.

Looking forward to see whether other values match up

xenio, 2024-08-31:
I've got the data from the HERF DCU and from OpenDTU to compare.

[image]

Response from api:
/platform/api/gateway/pvm-data/data_count_station_real_data
```
{
  "status": "0",
  "message": "success",
  "data":
    {
      "today_eq": "61.0",
      "month_eq": "6801",
      "year_eq": "6801",
      "total_eq": "6801",
      "real_power": "228.2",
      "co2_emission_reduction": "6780.597",
      "plant_tree": "0",
      "data_time": "2024-08-31 14:38:47",
      "last_data_time": "2024-08-31 14:38:47",
      "capacitor": "0.35",
      "is_balance": 0,
      "is_reflux": 0,
      "reflux_station_data": null,
    },
  "systemNotice": null,
}

```

**OpenDTU**

LOG:

[log_herf-300.txt](https://github.com/user-attachments/files/16824798/log_herf-300.txt)

[image]

tbnobody, 2024-08-31:
unfortunatly  the cloud does not show values like e.g. temperature. did you use the `generic_esp32` binary to flash the esp? Then I could provide you a version with partly corrected byte assignments. Then we just have to look at the temperature whether the value makes sense.

xenio, 2024-08-31:
I just installed the defaults from PlatformIO with vscode.
I have a AZ Delivery ESP32 NodeMCU Development Board.

tbnobody, 2024-08-31:
Make sure the PIO Environment in `Info` --> `System` shows generic_esp32
[image]

You can extract and flash this file via the web interface:
[generic_esp32.zip](https://github.com/user-attachments/files/16824899/generic_esp32.zip)

Then the first couple of values should make more sense.

tbnobody, 2024-08-31:
Yeah... just found the issue.... give me 3minutes..

tbnobody, 2024-08-31:
[generic_esp32.zip](https://github.com/user-attachments/files/16824926/generic_esp32.zip)

Should work better now :)

xenio, 2024-08-31:
WOOOOWWWW! I am so happy :))))

[image]

[image]

tbnobody, 2024-08-31:
looks good for the moment. but as expected the reactive power is wrong and I don't know how realistic the temperature is.

xenio, 2024-08-31:
I'll try to monitor the temperature, it is cloudy now.
Will you add the Herf-300 code to the repository in the future?
Thanks for your amazing work.

tbnobody, 2024-08-31:
> Will you add the Herf-300 code to the repository in the future?

Yes. The source will be available in the next release.

xenio, 2024-08-31:
Love it. Thanks!

tbnobody, 2024-08-31:
[generic_esp32.zip](https://github.com/user-attachments/files/16825141/generic_esp32.zip)

I've just updated the bit for PowerFactor to show my other guess... so either the shown temperature or power factor should be the right temperature :)

xenio, 2024-08-31:
With latest firmware.

[image]

tbnobody, 2024-09-20:
Implemented in a94977696645eab161bd3c8445c2ea7213c99ef5

closed 2024-09-20 by tbnobody