tbnobody/OpenDTU #2241: Buttons am Handy inaktiv

alekob, 2024-08-31:
### What happened?

Hallo,

am Smartphone (Nokia X20, Android 13) sind die beiden Buttons für die  Limit- und Energieeinstellungen ausgegraut und es können keine Einstellung darüber vorgenommen werden
[image]

Viele Grüße 
Alex

### To Reproduce Bug

Startseite von OpenDTU mit Chrome auf Android 13 öffnen

### Expected Behavior

Dialog zur Einstellung der Limits bzw. zum Ein- und Ausschalten des Inverterts sollte sich, wie am Tablet (iOS) oder PC (Linux, Windows) öffnen

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

mroenne2022, 2024-08-31:
Aber du bist auch sicher angemeldet an der OpenDTU?

schneeer, 2024-08-31:
Hi,

welche Version hat deine Chrome App?
Aktuell ist, glaub ich, 127.0.65533.106, die läuft ab Android 10.
Muss ja nicht an der Android Version (13) liegen.
Ich Habe ein Samsung A33 mit Android 14 und o.g. App Version.

Mal einen anderen Browser  versucht?
Wäre interessant, ob es an der App oder / und der Android Version liegt.

alekob, 2024-08-31:
@mroenne2022 Danke für den Hinweis! Ich war, ohne es zu wissen, abgemeldet, wahrscheinlich durch ein Chrome-Update. Mein Fehler, Sorry!

alekob, 2024-08-31:
Kein Bug, Problem sitzt vor dem Bildschirm

closed 2024-08-31 by alekob