tbnobody/OpenDTU #2244: No more communication between Opendtu and Hoymiles HM-800 inverters

Saroumanefr, 2024-09-01:
### What happened?

Since this morning there is no more communication between my opendtu and my Hoymiles HM-800 inverters.
I reset opendtu but it does'nt work.
Inverters : Hoymiles HM-800 - Vers. matérielH00.04.10 - Vers. logicielV01.00.16 - S/N : 11419********

### To Reproduce Bug

It's the first time that happens since 1 year.

### Expected Behavior

Recovering the connection between opendtu and my inverters

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

SteveeJackson, 2024-09-02:
Having the same issue with HM-600 since morning of August 23th while using Wroom 32 with NRF24 antenna. I double checked all cables and configurations but I had no succes. 
What I noticed so far is, that the NRF24 is well configured, but the chip status is marked as not connected. But actually, I never checked the status before, so it may has nothing to do with it.

AnthoineS, 2024-09-02:
I have same issue with my HMS-400-1T. Works perfectly from many months and after last update... every day, many disconnections.

schneeer, 2024-09-02:
I was interested to know whether the latest updates have led to increased DTU activity and thus increased power consumption.
It could have been that, for example, the power supply is too weak and therefore there are problems with the WLAN or problems with the connection to the inverter.

I couldn't find anything unusual.

When restarting, my DTU with display and NRF24 requires around 180 mA, and after a few seconds it only needs just under 100 mA. This is pretty much the same for several firmware versions.

So I can rule that out.

Saroumanefr, 2024-09-03:
> Salut @Saroumanefr can you provide more details about the issue from your OpenDTU, ie did you upgrade recently and if from which version are you coming to v24.8.5 ?

Hi, sorry i wasn't at home for few days and this morning everything worked fine........

AnthoineS, 2024-09-03:
> I have same issue with my Hoymiles HMS-400-1T. Works perfectly from many months and after last update, 24.8.5... every day, many disconnections. 
> I have also another OpenDTU with a Hoymiles HM-400BM, with the same version of FW... and no problem. 
> This issue seems only on a specific model. 

I solved issue by increasing level of power, +5dB...

OttiNC1, 2024-09-07:
I have the problem with disconnections (no data). Only a restart of the DTU helps. No matter which inverter I use.
Another problem is the displayed power of the openDTU. It is about 100 watts at 400 watts below the actual power that is produced.

OttiNC1, 2024-09-07:
When I switch the openDTU off and on again, the connection is established. If the inverter is off for longer, no connection is established, only when openDTU is off/on.

stefan123t, 2024-09-07:
@OttiNC1 thanks for posting your problem, too.

Please consider to open your own issue as we are waiting for @Saroumanefr to provide us more details about his issue (OP).

In order to analyse the issue we would need more details about your setup and also we may require a serial log.

Follow the link to the documentation to setup for USB / serial logging:
https://www.opendtu.solar/firmware/howto/serial_console/

Iserk86, 2024-09-09:
Hi, i have Same Problem. Des days it Works Fine, another day only few minutes of Connection to the DTU, then it goes offline and only Switch off and on is Working

asychev, 2024-09-10:
Same issue for last two days with HMS-800 (HMS-500 works just fine), problem comes out of nowhere, was working fine for 3+ months.

Saroumanefr, 2024-09-10:
Sorry, how can i find the logfiles ?
Since last week everything works fine for me.

> Salut @Saroumanefr can you provide more details about the issue from your OpenDTU, ie did you upgrade recently and if from which version are you coming to v24.8.5 ?

My problem happened before i upgrade to V24.8.5 and i think i was in V24.5.6, but i am not sure.

Saroumanefr, 2024-09-10:
> @Saroumanefr Okay so you can close your issue and others should open their own issue with proper Logfiles for their issue. Thanks!
> 
> Follow the link to the documentation to setup for USB / serial logging: https://www.opendtu.solar/firmware/howto/serial_console/

Ok, thanks

closed 2024-09-10 by Saroumanefr