tbnobody/OpenDTU #2259: Minor/optical Issue: VAr for reactive Power (not var)

harvey637, 2024-09-08:
### What happened?

In the webgui I see "var" as unit of reactive power
in german "Blindleistung x,x var"

### To Reproduce Bug

just take a look :-.)

### Expected Behavior

The unit of reactive power is "VAr" not "var"
"var" could be misunderstood an "variable", but it is V(olt) * A(mpere) r(eactive power) = "VAr"

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

harvey637, 2024-09-08:
api uses the correct "VAr" unit - ok

stefan123t, 2024-09-08:
@harvey637 so you would vote for VAr instead of var and % instead of no dimension for efficiency.
I recall that we have discussed the correct writing of var in the implementation of the Grid Profile and settled for one of the writings being international acknowledged, though maybe the german translation should be VAr ?

harvey637, 2024-09-08:
Hmmm, in my opinion voltage is "V" and ampere is "A", for spelling with capital "V" and "A" also international see
https://en.wikipedia.org/wiki/Volt and https://en.wikipedia.org/wiki/Ampere 
In the description of the value it IS "VAr" :-) "AC reactive power in VAr", so perfect.
With the "%" as unit of efficiency: yes, should be %. Even the description says "Ratio AC Power over DC Power in percent" so I guess it is just missing in the code to create this object, as the "unit" is created, but empty.
Hope this helps ...
Harvey

stefan123t, 2024-09-08:
Also Wikipedia:

https://en.wikipedia.org/wiki/Volt-ampere
https://de.wikipedia.org/wiki/Var_(Einheit)

The unit "var" is allowed by the [International System of Units](https://en.m.wikipedia.org/wiki/International_System_of_Units) (SI) even though the unit var is representative of a form of power.[[8]](https://en.m.wikipedia.org/wiki/Volt-ampere_reactive#cite_note-8) Per EU [directive 80/181/EEC](https://en.m.wikipedia.org/wiki/Directive_80/181/EEC) (the "metric directive"), the correct symbol is lower-case "var",[[1]](https://en.m.wikipedia.org/wiki/Volt-ampere_reactive#cite_note-dir-1) although the spellings "Var" and "VAr" are commonly seen, and "VAR" is widely used throughout the power industry.

stefan123t, 2024-12-06:
@harvey637 I think we have had this question already some time ago, though it was decided to stick with `var` as in the SI unit back then.

@tbnobody Would you want to keep this issue open or can we simply close it as not planned ?

harvey637, 2024-12-07:
Hi @ Stefan
- I still believe in VAr as abbreviation of [V]olt * [A]mpere * reactive, wikipedia allows all, so sticking with "var" is ok with me.
- as it is perfectly ok and only my optical feeling .... CLOSE this issue

closed 2024-12-07 by harvey637