tbnobody/OpenDTU #2260: minor/optical feature request: use kWh (not Wh) in YealdDay (adjust api/mqtt)

harvey637, 2024-09-08:
### What happened?

in the general tab the yealdDay is displayed in different units (Wh) than the YealdTotal (kWh).
As both values will jump over 1 kWh easily it is more easy to read both values in the same unit. The mixture of separators (omma or dot) might even confuse also.
The single module might stay as it is, as yieldday might no be so high in kWh. 
Or think about autoscale if value exceeds 1 kWh .... (but not in the api/mqtt)

### To Reproduce Bug

see in the gui

### Expected Behavior

use easy comparable units scales ....

and very minor: the textual explanation of the "yealdtotal" is wrongly "Energy converted to AC since reset watt hours" but it is in "kWh" :-) yealdday and yealdtotal both in "kWh" ????

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

creaimmo, 2024-12-21:
Hello,

I notice a display bug on the Total Yield value. Either a (.) or (,) is missing but the value expressed in kWh is wrong because this element is missing.

[image]

tbnobody, 2024-12-21:
@creaimmo this format is generated by your browser.

at least in chrome and vivaldi it looks like this:
[image]

closed 2025-04-18 by github-actions[bot]