tbnobody/OpenDTU #2261: Unknown Hardware shown in OpenDTU

tobiasburkart, 2024-09-08:
### What happened?

Error message:

Unbekanntes Modell! Bitte melden Sie die "Hardware Teilenummer" und das Modell (z.B. HM-350) [hier](https://github.com/tbnobody/OpenDTU/issues) als Problem.

my Modell is HM-800

### To Reproduce Bug

Nothing

### Expected Behavior

Knowing the HM-800
[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

24.8.5

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-09-08:
Please provide a screenshot of the web interface where you saw this message.

tobiasburkart, 2024-09-08:
[image]

tobiasburkart, 2024-09-08:
english UI

[image]

stefan123t, 2024-09-08:
@tobiasburkart 800HRS0.EU.HM sounds a bit strange to me. Can you please provide the first seven digits of your serial number signifying the model (digits 1-4) and the date of manufacture (digits 5-7). Thanks!

tobiasburkart, 2024-09-08:
> @tobiasburkart 800HRS0.EU.HM sounds a bit strange to me. Can you please provide the first seven digits of your serial number signifying the model (digits 1-4) and the date of manufacture (digits 5-7). Thanks!

1141831

mroenne2022, 2024-09-08:
Mein ehemaliger HM800  (läuft inzwischen bei meinem Bruder problemlos mit OpenDTU Version vom 5.8.24) sieht ein wenig anders aus auf dem Typenschild, SN beginnt aber genauso 1141831 !
Aber : 800HRS0**S**.EU.HM

[image]

stefan123t, 2024-09-08:
@mroenne2022 das S hat Hoymiles öfters verwendet, ua in Sub1G, HMS und DTU Lite/Pro S immer um die Verwendung des CMT Chips und die Sub 1GHz Funktechnik anzuzeigen.

mroenne2022, 2024-09-08:
@stefan123t : Okay, so tief steck ich in der Materie natürlich nicht drin,  mir ist lediglich der kleine Unterschied mit dem S aufgefallen und mit meinem HM800 und OpenDTU gabs bisher keine Probleme. Dann hat das Problem hier warscheinlich ganz andere Ursachen.

Iserk86, 2024-09-09:
Hallo, bei mir wird genau die gleiche Meldung gezeigt. Ist auch der selbe WR wie oben

tobiasburkart, 2024-09-10:
So what can be the solution? An update from OpenDTU?

Iserk86, 2024-09-10:
> So what can be the solution? An update from OpenDTU?

Hi, i tried an up and downgrade. Same Problem.
Also new flashing hasnt work , or another Plug and cable

stefan123t, 2024-09-10:
@tobiasburkart seltsam ist auch dass die Hardware Teilenummer 0 ist. Normalerweise ist das eine Zahl / Hex Darstellung der HW Version.
Die verwenden wir um unterschiedliche Modelle zu unterscheiden.

@tbnobody könnten wir da auch HW Part Num 0 als valide aufnehmen ?

stefan123t, 2024-09-18:
@tbnobody könnten wir die Fehlermeldung anpassen, viele Nutzer schwärzen leider auch die ersten vier/sieben Stellen der Seriennummer, obwohl die m.E. wichtig für die Fehleranalyse sind ?

Model: Unknown model! Please report the "Hardware Part Number" and model (e.g. HM-350) as an issue here.
->
Model: Unknown model `1XYZ`! Please report the "Hardware Part Number" and model (e.g. HM-350) as an issue here.

tobiasburkart, 2024-10-13:
Any news how we can fix it?

patti9000, 2025-01-13:
Having the same issue... Any news to this?
Bought the inverter in a set from "Media Markt Austria". Could it be that it is some kind of reselling version for other companies?

stefan123t, 2025-01-14:
@patti9000 & @tobiasburkart we need some more details from your inverters, i.e. the Model Name / Type, the Hardware Part Number and Version and the first seven digits of your Serial ID.
So you are the ones responsible for reporting the news :grinning:

patti9000, 2025-01-15:
Ok, well i'm not sure if it is the same problem. OpenDTU says now that it knows the device, but it doesn't receives any data. 

[image]

[image]
[image]
[image]

An this would be the configuration (used the preconfigured nodemcu file)
[image]
[image]

Wiring done according to this picture but wihout a display. Should match the "NRF24" configuration in the nodemcu file:
[image]

Thanks

closed 2025-04-18 by github-actions[bot]