tbnobody/OpenDTU #2272: Standby (Abschaltung WR) wird nicht mehr erkannt

super-mertl, 2024-09-11:
### What happened?

Bei mir läuft OpenDTU (v24.8.5) nun schon einige Wochen erfolgreich mit zwei HMS-2000-4T (FW1.0.18)
Es hat alles funktioniert. Abends wurde der Abschaltzeitpunkt sauber erkannt und beide WR gelb dargestellt.

Nun nach Ergänzen von zwei HMT-2250-6T (FW1.0.10) ist dies reproduzierbar nicht mehr der Fall.
Nun werden ALLE 4 WR irgendwann auf rot / nicht erreichbar gesetzt und munter die Sekunden bis zum nächsten morgen hochgezählt.

Außer dem Hinzufügen hat sich keine Einstellung geändert.
Ich habe an verschiedenen Einstellung (Dämmerung, Abfrageintervall, Sendeleistung) erfolglos herumgespielt.

Tagsüber werden die WR sauber abgefragt und blau dargestellt...

### To Reproduce Bug

täglich (abends) reproduzierbar

### Expected Behavior

saubere Darstellung / Abschaltverhalten

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

0cc55f3

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tugsi, 2024-09-12:
I can only agree with that.
I integrated 6x HMT-2250-6T inverters from the beginning and it was actually always red at night because it was unavailable. 
I thought that was normal because they switch off, but it was only now through this issue that I realized that it should look different. 
Therefore I would say that it must be due to the HMT-2250-6T settings in openDTU. 
Power indicator is 0W, so it should be yellow.

tbnobody, 2024-09-12:
> Nun werden ALLE 4 WR irgendwann auf rot / nicht erreichbar gesetzt und munter die Sekunden bis zum nächsten morgen hochgezählt.

So soll es sein. Wenn ein WR nicht erreichbar ist wird er rot. Wenn er erreichbar ist jedoch 0W produziert  ist er gelb.

tugsi, 2024-09-12:
> So soll es sein. Wenn ein WR nicht erreichbar ist wird er rot. Wenn er erreichbar ist jedoch 0W produziert ist er gelb.

Ah, ok, dann hatte ich das missverstanden, hatte irgendwo das mit dem gelb gelesen und es so verstanden, dass er dann gelb wird. Aber irgendwie auch logisch, woher soll er wissen, ob die 0W und nicht Erreichbarkeit durch Dunkelheit kommt oder durch eben nur der nicht Erreichbarkeit. Mein Fehler.
Dafür gibt´s ja den Punkt der Abschaltung bei Nacht. 
Danke .

super-mertl, 2024-09-13:
> So soll es sein. Wenn ein WR nicht erreichbar ist wird er rot. Wenn er erreichbar ist jedoch 0W produziert ist er gelb.

Ok, bin halt vom dem festgestellten Verhalten bei nur zwei angeschlossenen HMS-2000 ausgegangen.

Nur zum Verständnis:
Verhalten sich die WR hier unterschiedlich? Weil ist ist definitiv so, dass die zwei HMS (allein) noch 2h nach "Abschaltung" gelb 
 sind.
Sind die WR im Standby grundsätzlich gar nicht mehr erreichbar, obwohl sie AC seitig versorgt sind?
Abschaltung bei Nacht wird durch die Einstellung Dämmerung bestimmt oder für was ist die Auswahl Dämmerung gut?

tbnobody, 2024-09-13:
> Sind die WR im Standby grundsätzlich gar nicht mehr erreichbar, obwohl sie AC seitig versorgt sind?

Korrekt

> Abschaltung bei Nacht wird durch die Einstellung Dämmerung bestimmt oder für was ist die Auswahl Dämmerung gut?

Eine Abschaltung bei Nacht betrifft nur das Polling der DTU. Die WR sind Nachts so und so nicht erreichbar. Wenn die DTU natürlich schon mit dem Pollen aufhört (weil sie in den Nachtzeitraum fallen) obwohl die WR noch erreichbar sind, bleibt hier der letzte bekannte Zustand in der DTU erhalten.

super-mertl, 2024-09-13:
ok, danke, ich schließ das Ticket...

closed 2024-09-13 by super-mertl