tbnobody/OpenDTU #2277: openDTU bricht Verbindung zu HMS-1600-4T plötzlich ab

khschmidt, 2024-09-13:
### What happened?

DTU Version 24.4.24
Nulleinspeisung mit eigenem Pythonscript funktioniert seit 28.7.2024 problemlos.
Erstmals Abbruch am 3.9.2024 etwa Mittags, in der Konsole sieht man keine Antwort vom Wechselrichter. Daraufhin alles mögliche versucht, Leistung raufsetzen, Frequenz ändern, anderes Netzteil, Wechselrichter vom Netz genommen, nichts davon hat geholfen. Dann habe ich die DTU-Seriennummer geändert und sofort hat die Kommunikation wieder funktioniert.
Heute am 13.9.2024 erneut Verbindungsabbruch etwa um 8:00, davor wegen Bewölkung Produktion von etwa 60W. Diesmal hat die Änderung der DTU-Seriennummer nicht gehofen. Bislang noch keine Verbindung zum Wechselrichter.
Der Wechselrichter schaltet sich eigenartigerweise auf Limit von 0%, LED am Wechselrichter blinkt langsam grün.
Ich vermute keinen Bug der openDTU, sondern einen gewollten Trick der Hoymiles-Firmware um den Anwender zu zwingen, sich eine original DTUlite zuzulegen und damit die openDTU-Anwender zu ärgern. Meine Frage: gibt es diesbezüglich schon Erkenntnisse, wie man dieses Problem lösen kann?

### To Reproduce Bug

Kommt nach einiger Betriebszeit

### Expected Behavior

reproduzierbare Verbindungsaufnahme nach Abbruch

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.4.24

### Relevant log/trace output

```Shell
Websocket: [/livedata][11] disconnect
10:08:23.522 > RX Period End
10:08:23.522 > All missing
10:08:23.522 > Nothing received, resend whole request
10:08:23.522 > TX ActivePowerControl 865.00 MHz --> 51 A0 06 B5 53 80 17 26 34 81 0B 00 00 C8 00 01 5E 41 C8 
10:08:25.464 > RX Period End
10:08:25.464 > All missing
10:08:25.464 > Nothing received, resend whole request
10:08:25.464 > TX ActivePowerControl 865.00 MHz --> 51 A0 06 B5 53 80 17 26 34 81 0B 00 00 C8 00 01 5E 41 C8 
10:08:27.491 > RX Period End
10:08:27.491 > All missing
10:08:27.491 > Nothing received, resend count exeeded
10:08:27.887 > TX ActivePowerControl 865.00 MHz --> 51 A0 06 B5 53 80 17 26 34 81 0B 00 00 C8 00 01 5E 41 C8 
10:08:29.736 > RX Period End
10:08:29.736 > All missing
10:08:29.736 > Nothing received, resend whole request
10:08:29.736 > TX ActivePowerControl 865.00 MHz --> 51 A0 06 B5 53 80 17 26 34 81 0B 00 00 C8 00 01 5E 41 C8
```

### Anything else?

nichts

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

khschmidt, 2024-09-14:
Änderung der Frequenz schrittweise zwischen 864,0 bis 868,0MHz zeigt weiterhin keinen Verbindungsaufbau, erst Trennen von AC und allen PV-Panelen und einige Minuten warten führen nach dem Hochfahren wieder zu einem Verbindungsaufbau. Die Livedaten zeigen die Produktion von etwa 100W pro Panel (4 Panele, jedoch bewölkt) jedoch zeigt openDTU ein Limit von 0 an, was jedoch nicht stimmen kann. Das manuelle Setzen des Limits auf 50% funktioniert und wird dann auch angezeigt. Im Log findet sich 8 mal Time calibration nach dem Hochfahren des Wechselrichters. Heute morgen ist der Wechselrichter ganz normal um 6:23Uhr gestartet, kein Logeintrag von Time calibration

Gururom, 2024-09-15:
Ein typisches Problem, wahrscheinlich vom WR erzeugt.
Du kannst machen was Du willst, mal bringt es was, mal nicht.
Hab ich so alle 1,5 Monate. Das keine Verbindung besteht, löst sich allein, ich bin es leid zu suchen.
Bei mir jedoch fährt der WR hoch und produziert voll.
Nutze OpenDTU-OnBattery

dersch81, 2024-12-07:
Ich habe seit einigen Wochen mit 4 meiner 10 WR an der DTU das gleiche Problem. 2 der 4 schicken manchmal Daten aber sehr sporadisch. Das ganze lief über ein Jahr perfekt ohne Ausfälle. ein neustarten der WR bringt nix. Auch die DTU habe ich vollständig neu aufgesetzt mit tausch des ESP und dem NFR Modul und Netzteil. Ändert aber nix. Wie kann man das am besten Debuggen?

khschmidt, 2025-01-27:
Problem gelöst:
habe mein python script um folgende Funktion erweitert:
Nach einem Verbindungsabbruch am Tag programmiere ich über die web-api die Frequenz des CMT2300 um und zwar auf das nächsthöhere Band. Danach verbindet sich die DTU sofort, da der Wechselrichter offensichtlich auf diesem Nachbarkanal wartet. Die Bestätigung erfolgt bei der nächsten Anfrage des scripts nach 7 Sekunden (ist meine Abfrageintervall). dazu habe ich im script definierte Grenzen für die Frequenz von 865,0 bis 867,0 MHz. Gestartet wurde also vor etwa 7 Wochen mit 865MHz, mittlerweile ist die verwendete Frequenz nach insgesamt 4 Abbrüchen bei 866,25 MHz. Wird die Grenze von 867 MHz erreicht, so schalte ich in der Nacht zurück auf die untere Grenze.

dersch81, 2025-02-17:
Naja bei mir sind 3 WR von 10 gar nicht mehr am Kommunizieren. Ein anderer nur noch sporadisch. Ich glaube daher nicht, dass es was mit der Frequenz zu tun hat. Die betroffenen WR mal komplett Stromlos machen und ein paar Stunden warten bringt auch nix. Die Entfernung kann es auch nicht sein.

closed 2025-04-18 by github-actions[bot]