tbnobody/OpenDTU #2278: [Request]Button zum Auslösen  eines Durchsteppens der Frequenz des CMT2300A wenn keine Verbindung zum Wechselrichter zB HMS 1600-4T

khschmidt, 2024-09-14:
### Is your feature request related to a problem? Please describe.

Mehrfach wird im Forum über Verbindungsabbrüche beim HMS 1600-4T berichtet. In einem Beitrag wird erwähnt, dass der Wechselrichter offensichtlich den Frequenzkanal wechselt und nach dem Ändern der Frequenz des CMT2300A die Kommunikation wieder funktioniert. 

### Describe the solution you'd like

Da nicht jeder über einen Spectrumanalyser verfügt, wäre folgende Funktion hilfreich:  Ausgelöst durch einen Klick auf einen Button mit der niedrigsten erlaubten Frequenz eine Anfrage an den Inverter senden, die Antwort oder das Timeout abwarten. Bei Timeout auf den nächsthöheren Frequenzkanal schalten und so weiter. Wenn die Kommunikation erfolgreich ist, Frequenz anzeigen um sie zu speichern. Ist der Scan nicht erfolgreich, Fehlermeldung ausgeben

khschmidt, 2024-09-19:
Frequenzwechsel von HMS 1600-4T bestätigt:
Heute um 9:30 Uhr ging die openDTU wieder offline, nachdem sie seit Sonnenaufgang online auf 865MHz war. Zufällig sah ich gerade auf das Display als dies passierte. Ich wechselte dann die Frequenz auf 865,25MHZ und innerhalb weniger Sekunden war die Verbindung wieder hergestellt. Die Scanfunktion für den CMT2300A wäre für diesen Fall sehr hilfreich

Troubadix, 2024-09-19:
Hier das gleiche Problem mit HMS-1600-4T.
Nachdem die DTU seit gestern abend endlich mit CMT-Funkmodul läuft, war heute Vormittag auf einmal der WR nicht erreichbar.
Dank diesem Beitrag die Frequenz auf 865,25 geändert und schon ist die Verbindung wieder da.

Fragen dazu:
Was triggert die Änderung der Frequenz am WR? 
Kann man diese Frequenzänderung irgendwie verhindern?
Wie kann man die aktuell genutzte Frequenz ausfindig machen, ohne alle möglichen Kombinationen durchzutesten?

Danke.

stefan123t, 2024-09-20:
@khschmidt @Troubadix einen extra GPIO für einen Frequenzwechsel zu belegen ist mE nicht zielführend.
Auch ein Button in OpenDTU GUI um manuell einen Frequenzsweep von der Base Frequenz 865MHz in 0,25MHz Schritten aufwärts durchzuführen wäre nicht wirklich eine Lösung.
Es gibt ja einen definierten Frequenzbereich je nach Land/Region den die OpenDTU und WR verwenden dürfen. Auch den Verbindungsverlust stellt die DTU selbst fest. Evtl haben neuere WR eine andere Firmware Version die das bisherige feste Channel Setting von der DTU nicht mehr hinnehmen?

Könntet ihr die ersten sieben Stellen Eurer Seriennummer und die Seite mit der Firmware/Hardware Version aus OpenDTU hier dokumentieren?

khschmidt, 2024-09-20:
Seriennummer: 1164A006Bxxx

khschmidt, 2024-09-22:
habe gestern mein python script der Nulleinspeisung erweitert.
Es wird über die web-api livedata/status/reachable ausgewertet und in eine Logdatei geschrieben.
Heute morgen etwa um 7.00 war die DTU auf 865,25MHz noch verbunden, etwa um 7:30 war das Display der DTU dunkel.
Auswertung meiner Logdatei:
6:14:52 Wechselricher erstmals erreichbar
6:15:08 Wechselricher nicht erreichbar
6:16:54 Wechselricher erreichbar
das wiederholt sich bis 6:24:51 in Summe 9 mal (ist noch sehr dunkel) 
Sonnenaufgang war heute um 6:45
im Wechselrichterlog: 6:24:45 Wechselrichter gestartet
bis 7:21:00 erneut Wechselrichter nicht erreichbar
um 7:50 habe ich die Frequenz auf 865,00 geändert, etwa 30 Sekunden später war die DTU wieder verbunden

kopierschnitte, 2024-09-29:
Muss mich hier mal dranhängen. Als Frischling der heute seinen allerersten HMS-1800-4T (FW 1.0.27) mit einem OpenDTU Fusion v3 (generic_esp32s3_usb 24.9.27) in Betrieb genommen hat war ich na klar sofort in Panik als keine Daten mehr vom WR kamen.

Zum Glück hab ich dieses Issue gefunden bevor ich den WR herausgerissen habe ;-)

Müsste der HMS nicht einen Frequenzwechsel irgendwie vorher ankündigen? Ich meine, der Hoymiles-Stick muss das doch auch irgendwie mitbekommen.

ms49434, 2024-09-29:
Rein von der Logik macht es meines Erachtens keinen Sinn, wenn irgendein Wechselrichter der DTU sagt, auf welcher Frequenz sie zu kommunizieren hat. Eine DTU kann ja mehr als einen Wechselrichter verwalten und wird nicht jeden Wechselrichter auf dessen eigener Frequenz anfunken.

kopierschnitte, 2024-09-29:
Guter Punkt. Oder lässt das Funkmodul vielleicht ein größeres Empfangsspektrum zu?

Vom Signalduino-Projekt (FHEM) kenne ich das. Dort werden im 433MHz-Band ja auch diverse Geräte mit leicht unterschiedlichen Frequenzen empfangen. Ist aber ein anderer Chip.

stefan123t, 2024-09-29:
@khschmidt & @kopierschnitte könnt ihr von Eurem Problem ein Serial Log erstellen, dann weiß man vielleicht eher wann und wo genau das Problem auftritt. Das bei @kopierschnitte klingt seltsam, da er angeblich noch Kommandos schicken kann?

https://www.opendtu.solar/firmware/howto/serial_console/

kopierschnitte, 2024-09-29:
Seriallog dauert bei mir leider, da ich den ESP recht tief vergraben habe. 

Aber die Sache mit den Kommandos ist definitiv so. Habe ja direkt vor'm WR gekniet und gesehen, wie er direkt von grün zu rot (und wieder zurück) wechselte.

OpenDTU hat natürlich mangels ACK einen Fehler angezeigt.

khschmidt, 2024-09-30:
Serial Log ist bei mir auch aufwendig. Kann man die virtuelle Konsole in eine Datei umleiten, oder gibt es eine Debugschnittstelle über Telnet?

khschmidt, 2024-09-30:
Der Fall den Kopierschnitte beschreibt wäre auch so erklärbar: Wenn er sehr nahe am Wechselrichter mit seiner DTU war, dann ist die Empfangsfeldstärke sehr hoch. Wenn die Kanaltrennung des Chips im Wechselrichter nicht allzu hoch ist, dann kann das auch ein Kanalübersprechen sein, da bei mir der Wechselrichter von 865,00MHz auf den Nachbarkanal 865,25MHz gewechselt hat. Die Antwort vom Wechselrichter wird von der DTU nicht gesehen, wenn die Kanaltrennung des CMT2300 höher ist.

kopierschnitte, 2024-09-30:
In der Tat sind es bei mir max. 2m Entfernung. Könnte also sein!

stefan123t, 2024-09-30:
@khschmidt es gab erst kürzlich einen PR für die Umleitung auf einen Remote Syslog Server #2292 .

Ich habe leider die Vermutung dass Hoymiles seit ca. Ende 2023/Anfang 2024 das Verhalten / Verfahren der HMS/HMT Baureihe mit einem Update der WR Firmware geändert hat. Früher hatte das Change Channel Command von der DTU ausgereicht.

Vielleicht kann auch ein Nutzer (@nivadis ?) mit einer Hoymiles DTU-S Pro/Lite das aktuelle Verhalten nochmal mitschneiden ?

khschmidt, 2024-09-30:
@stefan123t danke für den Hinweis, dann werde ich meine Zweit-DTU mal auf die aktuelle Firmware updaten damit ich remote syslog einrichten kann, sende ich dann auf meinen raspi3

stefan123t, 2024-09-30:
@khschmidt den PR musst Du selber bauen, der ist m.W. noch nicht im aktuellen Trunk / Master enthalten.

khschmidt, 2024-10-02:
Hab den PR syslog in die aktuelle Version eingebaut, compiliert und geflasht. Leider taucht auf der Konfigurationsseite/Netzwerk der Syslogéintrag nicht auf. Irgendetwas habe ich wohl noch übersehen

khschmidt, 2024-10-03:
Heute ist wieder mal im Wechselrichterlog der folgende Eintrag:
12:06:22 Zeitabgleich
12:06:37 Zeitabgleich
im syslog-File finden sich in diesem Zeitfenster folgende Einträge:
mehrfach TX Powercontrol innerhalb weniger Sekunden
[neu 1.txt](https://github.com/user-attachments/files/17244197/neu.1.txt)

stefan123t, 2024-10-03:
Seltsam finde ich dass das Active Power Limit mE sehr niedrig ist 1W/1%, obwohl der WR ca. 20-30W mindestens braucht um zu Laufen. Man darf ihn nicht unter die Einschaltschwelle runterregeln, weil er sonst nicht mehr selber aufwacht oder antworten kann.

Hier ein altes Beispiel aus #35 
```
22:28:13.969 > TX 51 [72 61 55 82 78 56 34](tel:72 61 55 82 78 56 34) 12 81 0B 00 01 2C 01 00 C5 C0 3E
```

Hier aus Deinem Log
```
2024-10-03T12:06:32.503315+02:00 OpenDTU-75C360 OpenDTU[95e5087d] ﻿TX ActivePowerControl 865.00 MHz --> 51 A0 06 B5 53 80 17 26 36 81 0B 00 00 B4 00 01 86 80 AF 
```

Und hier im 1:1 Vergleich nochmal:
```
22:28:13.969 > TX 51 -72 61 55 82- -78 56 34 12- 81 -0B 00- 01 2C 01 00 C5 C0 3E
12:06:32.503 > ﻿TX 51 -A0 06 B5 53- -80 17 26 36- 81 -0B 00- 00 B4 00 01 86 80 AF 
```

@tbnobody sollen wir das untere Limit beschränken und stattdessen einen Fehler zurück liefern. Laut den bisherigen Erfahrungsberichten soll man den WR dann lieber mit Power Off bzw. on Aus- bzw. Einschalten, wenn man den Strom nicht nutzen möchte, zB OpenDTU-OnBattery.

khschmidt, 2024-10-03:
@stefan123t das ist sehr eigenartig, mein pythonscript hat als unteres limit 15% relativ, also 240W, es wird auch nie weniger in der dtu angezeigt und das script läuft mit 5s Raster und in diesem Zeitfenster ist dieses Kommando mehrfach innerhalb weniger Sekunden, danach ist das Logfile wieder komplett unauffällig. Ich habe noch ein Stück danach rauskopiert
[neu2danach.txt](https://github.com/user-attachments/files/17245970/neu2danach.txt)

khschmidt, 2024-10-04:
mein Script habe ich so geschrieben, das ich das Limit immer relativ, also in % an die DTU sende, also von minimal 15 bis maximal 100. Der Wechselrichter hat 1600W (HMS1600-4T). Auch sende ich nur ein neues Limit, wenn es sich vom alten bestätigten Limit unterscheidet. Wenn das vom Wechselrichter rückgesendete Limit sich von dem von mir zuvor gesendeten Limit unterscheidet, schreibe ich das ab heute in eine Logdatei. Wenn ich hier einen Eintrag bekomme, dann erhöhe ich den minimalen limitstep auf 2%. Das Limit regle ich nur, wenn es nötig ist, heute ist es beispielsweise sehr bewölkt, das Haus verbraucht jetzt ca. 300W, produziert werden aktuell lächerliche 78W. Da erfolgt natürlich keine Limitvorgabe, die Regelschleife läuft aktuell im Raster von 5 Sekunden, die Limitvorgabe an die DTU erfolgt aber nur wenn nötig. Das Raster von 5 Sekunden habe ich nur deshalb gewählt, weil die DTU ebenfalls im 5 Sekundenraster den Wechselrichter abfragt; ist also nicht unbedingt nötig

stefan123t, 2024-10-04:
Hallo @khschmidt 
das klingt erst mal plausibel und sinnig was die Herangehensweise angeht.
Ich denke wir müssen evtl unterscheiden:
* Event 2: Time calibration
* Kommunikationsprobleme mit dem WR

Das Event 2 kommt immer wieder mal vor, vermutlich wenn sich die Timestamps in den Commands von der DTU mit denen des WR beissen. Heute neu geflasht mit v24.10.02 und der WR hat sich beim Reset der OpenDTU jeweils ein paar Einträge im Eventlog gemerkt.
Beim letzten Mal habe ich gesehen dass die DTU noch keine aktuelle Zeit per NTP geholt hatte. Habe ich dann manuell veranlasst und dann auch keine weiteren Warnungen/Fehler gesehen.

Die Kommunikationsprobleme können mE von zwei prinzipiellen Ursachen kommen:
* die DTU / der WR wechselt die Frequenz bzw den Kanal und die DTU versucht auf einem anderen Kanal die Antwort zu empfangen.
* durch die Kanalwechsel können auch Folge-Fehler auftreten, so wird bei mir (NRF24) häufig das erste Paket verpasst und dieses erneut angefordert. Das treibt den TX Re-request Paket counter nach oben und führt zu zusätzlicher Kommunikation.

Die Radio statistics findest Du auf der Live-View Seite unten zum Aufklappen.

khschmidt, 2024-10-04:
Hallo @stefan123t, sehe ich auch so, das Thema Time calibration hat wahrscheinlich nichts mit den Kommunikationsabbrüchen/mögliche Frequenzwechsel zu tun, manchmal habe ich tagelang keine Einträge von Time calibration geshen. Ich lasse das Mitloggen der Konsole jetzt mal laufen, eventuell kommt es ja wieder zu Kommunikationsproblemen, dann sehe ich mir die Logs wieder an.

kopierschnitte, 2024-10-04:
Kleine Anmerkung von mir: Ich nutze keine Limitierung und das Frequenzproblem tauchte dennoch auf.
Seit zwei Tagen ist jedoch Ruhe...

Troubadix, 2024-10-12:
Ich hab kurzzeitig eine zweite OpenDTU an den gleichen WR gehängt - das soll man ja nicht machen, aber da kein anderer WR zum Test vorhanden war hab ich es halt doch gemacht - und das hat dann prompt den Frequenzwechsel getriggert.
Das hab ich dann noch ein paar Mal gemacht, nach wenigen Minuten immer der Frequenzwechsel.
Dann die zweite DTU ausgemacht und nun seit gut 2 Wochen kein Frequenzwechsel mehr.

Seriennummer | 1164a0092a30
Bootloader-Version | 0.1.1
Firmware-Version | 1.0.27
Firmware-Erstellungsdatum | 2023-06-05 10:24:00
Hardware-Teilenummer | 270680326
Hardware-Version | 01.10

khschmidt, 2024-10-14:
Ich logge seit einer Woche die Konsole mit. Am Morgen beim Start des Wechselrichters sieht das so aus:
6:55:27 Wechselrichter gestartet (aus Wechselrichterlog)
7:12:00 Sunrise
7:12:49 DTU beginnt mit Request SystemConfigPara auf 865MHz (wie konfiguriert)
erfolglos, da der WR wahrscheinlich auf einer anderen Frequenz empfangsbereit ist
30 mal resend, dann 5 mal ChannelChangeComand auf 868MHz der DTU erfolglos.
7:13:04 WR antwortet auf 865MHz und Verbindung bleibt dann den ganzen Tag stabil.
Ich nehme an, dass der Wechselrichter von sich aus solange durch das Frequenzband steppt, bis er erstmals erfolgreich eine Nachricht von der DTU empfängt. Auf dieser Frequenz bleibt er dann solange er erfolgreich Kontakt mit der DTU hält. Tritt eine längere Unterbrechung auf, beginnt er wieder das Frequenzband durchzusteppen. Warum er manchmal die zuvor verwendete Frequenz nicht mehr anwählt, ist bislang noch unklar. Bei mir läuft die Kommunikation allerdings seit Wochen stabil ohne Unterbrechungen.

SG-fabian, 2024-10-25:
Ich habe einen HMS-1600-4T  und die Verbindung bricht aktuell jeden Tag gegen 11 Uhr ab. Am späten Nachmittag habe ich dann plötzlich wieder eine Verbindung. Interessanterweise ist es jeden Tag das selbe Spiel, nur sehr selten kommt zwischendrin mal kurz eine neue Verbindung zustande.

kopierschnitte, 2024-10-26:
Vielleicht hat es doch etwas mit dem Funk an sich zu tun? Seitdem ich bei mir die Sendeleistung deutlich reduziert habe ist völlig Ruhe. Seit Wochen.
Ebenfalls 1600-4T

stefan123t, 2024-10-31:
Hier die ChannelChangeCommands der Original Hoymiles DTU Pro S als Referenz:
https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2427788809

stefan123t, 2024-11-02:
@khschmidt danke für die Logfiles in dem einen sendet die DTU einfach auf 865,00 MHz

Bei dem zweiten Logfile wird auch tatsächlich das ChannelChangeCommand 0x56 fünf mal gesendet und danach klappt die Kommunikation offenbar auch wieder. Zumindest habe ich ein 0x15 DevInfo Request RealTimeRunData_Debug 0x0b mit der entsprechenden 0x95 DevInfo Response Antwort gesehen.

@tbnobody wir sollten eigentlich den Inverter wieder ansprechen können wenn wir ihm das ChannelChangeCommand 0x56 auf allen Kanälen mal zukommen lassen und dann immer auf dem Zielkanal auf die Antwort warten. Wie ist das denn aktuell bei den CMT2300A Modellen programmiert. Ich habe selber kein HMS/HMT deshalb kann ich es nur anhand der Logfiles nachvollziehen.

<details><summary>Details</summary>
<p>

```
2024-11-01T07:53:02.570032+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿CMT TX power set to 0 dBm
2024-11-01T07:53:02.769983+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:02.769983+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿All missing
2024-11-01T07:53:02.802748+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Nothing received, resend whole request
2024-11-01T07:53:02.802748+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX SystemConfigPara 865.25 MHz --> 15 A0 06 B5 53 80 17 26 36 80 05 00 67 24 7A CA 00 00 00 00 00 00 00 00 11 69 DC 
2024-11-01T07:53:02.861422+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:02.861422+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 81 00 01 01 F4 00 00 03 E8 00 00 00 00 00 00 6B 6A CD | -54 dBm
2024-11-01T07:53:03.179192+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:03.179192+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Success
2024-11-01T07:53:03.867128+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Fetch inverter: 1164A006B553
2024-11-01T07:53:03.867128+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX ChannelChangeCommand 868.00 MHz --> 56 A0 06 B5 53 80 17 26 36 02 15 21 15 14 A6 
2024-11-01T07:53:03.867128+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:03.867128+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿All missing
2024-11-01T07:53:03.867128+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Nothing received, resend whole request
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX ChannelChangeCommand 868.00 MHz --> 56 A0 06 B5 53 80 17 26 36 02 15 21 15 14 A6 
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿All missing
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Nothing received, resend whole request
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX ChannelChangeCommand 868.00 MHz --> 56 A0 06 B5 53 80 17 26 36 02 15 21 15 14 A6 
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿All missing
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Nothing received, resend whole request
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX ChannelChangeCommand 868.00 MHz --> 56 A0 06 B5 53 80 17 26 36 02 15 21 15 14 A6 
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿All missing
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Nothing received, resend whole request
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX ChannelChangeCommand 868.00 MHz --> 56 A0 06 B5 53 80 17 26 36 02 15 21 15 14 A6 
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿All missing
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Nothing received, resend count exeeded
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿TX RealTimeRunData 865.25 MHz --> 15 A0 06 B5 53 80 17 26 36 80 0B 00 67 24 7A CF 00 00 00 00 00 00 00 00 8F 59 79 
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 01 00 01 01 6B 01 73 00 A0 00 AF 02 48 02 8C 00 02 83 | -55 dBm
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:04.213871+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 02 53 01 00 02 1C C9 00 1B 00 1E 01 68 01 6A 00 8A 58 | -55 dBm
2024-11-01T07:53:04.304541+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:04.304541+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 03 00 9B 01 F4 02 35 00 02 32 3B 00 02 3A 18 00 15 36 | -55 dBm
2024-11-01T07:53:04.304541+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:04.304541+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 04 00 18 09 4B 13 89 08 8B 00 00 00 5B 03 E8 00 7D D8 | -55 dBm
2024-11-01T07:53:04.304541+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:04.304541+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 85 00 15 62 83 23 | -55 dBm
[... cut ...]
2024-11-01T07:53:09.067181+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 04 00 18 09 4B 13 89 08 9C 00 00 00 5C 03 E7 00 7D C7 | -54 dBm
2024-11-01T07:53:09.094741+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Interrupt received
2024-11-01T07:53:09.117496+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX 865.25 MHz --> 95 A0 06 B5 53 80 17 26 36 85 00 15 94 C5 93 | -54 dBm
2024-11-01T07:53:09.272703+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿RX Period End
2024-11-01T07:53:09.272703+01:00 OpenDTU-75C360 OpenDTU[a864ee4d] ï»¿Success
``` 

</p>
</details>

tbnobody, 2024-11-02:
> wir sollten eigentlich den Inverter wieder ansprechen können wenn wir ihm das ChannelChangeCommand 0x56 auf allen Kanälen mal zukommen lassen und dann immer auf dem Zielkanal auf die Antwort warten.

Definiere "Alle Kanäle"? Es sind keine Kanäle definiert. Es gibt alle möglichen Frequenzen in 0,25mhz Schritten von 860,25 - 923,50. Was also ~250 mögliche Frequenzen sind. Ich sehe es gerade nicht ein hier das ganze 868MHz Band zuzumüllen. Das kann so nicht die Lösung sein.

khschmidt, 2024-11-02:
@stefan123t Beide Dateien kommen aus einem Tageslogfile. die beiden beiden Files habe ich aus diesem Logfile ausgeschnitten, da ich nur die relevanten Teile zeigen wollte. Das zweite File habe ich ausgeschnitten **ab dem Umschaltbefehl** von 865,0 auf 865,25 MHz. davor sind nur die Versuche den Inverter auf 865MHz zu erreichen und die jeweils 5 Versuche des ChannelChanceCommands.

Bitte nicht missinterpretieren: Im File ManuellerFrequenzwechsel sieht man, dass die ChannelChangeCommands hier nichts bewirken, die Kommunikation auf 865,25 MHz ist bereits ab Zeile 7 vom Inverter bestätigt, die ChannelChangeCommands sind hier nicht hilfreich, da die Kommunikation zu diesem Zeitpunkt schon wieder funktioniert.

@tbnobody Mit alle Kanäle meine ich, dass man 2 einstellbare Grenzen (zB.: von: 865,0 MHz bis: 868 MHz). Aktuell wird bei Verbindungsunterbrechung laufend RealTimeRunData gesendet (ich meine so 30 mal) und dann jeweils 5 mal das ChannelChangeCommand. Die neue Funktion würde ebenfalls zB. 30 mal das RealTimeRunData senden und bei nicht erfolgreicher Verbindung den CMT um 0,25MHz hochschalten, dann wieder 30 mal RealTimeRunCommand usw. bis zur oberen Grenze, dann w mit der unteren Grenze fortsetzen, bis Verbindung erfolgreich bestätigt wird.

stefan123t, 2024-11-02:
@tbnobody ich will nicht das ganze Frequenzband zu spammen sondern würde mich auf die üblichen Frequenzen / Kanäle beschränken, ich muss nochmal nachlesen; es gab in der DTU Pro S FDI Anmeldung oder anderswo eine Liste der wirklich genutzten Frequenzen / Kanäle.

Hier nochmal Deine Doku aus dem OpenDTU Code:

https://github.com/tbnobody/OpenDTU/blob/dc5eb96f5035ba9e058e38d83df2d6c691b0764b/lib/Hoymiles/src/commands/ChannelChangeCommand.cpp#L9-L18

Das ChannelChangeCommand nimmt mW nur einen Kanal und sieht keine Frequenz für die Kommunikation vor.

`56 <A0 06 B5 53> <80 17 26 36> 02 [15 21] [>15< 14] A6`

Die Requests werden mit 01 und 02 als Paketcounter von der DTU an den WR geschickt.

`56 80164633 80164633 01 [1521] [>21<14] 56`
`56 81313070 81313070 02 [1521] [>18<14] 6c`

Davon ist das zweite [>15< 14] mW und auch nach Deiner Doku der Kanal, in den Logfiles von Bernhard steht an dieser Stelle >18< 14, >21< 14, >15< 14 also verwendet die DTU Pro nur jeden dritten Kanal.

Schau Dir dazu doch bitte mal die Logfiles von Bernhard an, die ich hier in Auszügen angehängt habe: https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2427788809

Die Antwort kommt nach der Trainingsphase wie üblich mit 0x80 Komplement also 0x56 | 0x80 = 0xd6 vom WR.

zB:
`d6 80164633 80164633 01 00 1521 e3`

khschmidt, 2024-11-03:
Noch eine Ergänzung zu meinen Beobachtungen:
Wenn der HMS-1600-4T aus welchem Grund auch immer meint, den Kanal ändern zu müssen, dann sperrt er offensichtlich diesen Kanal eventuell bis zum nächsten Reset. Bei mir ist das bisher 4 oder 5 mal passiert, beim ersten mal dachte ich, dass der Inverter defekt ist, ich habe dann aber den Inverter vom Netz getrennt und die Panele abgesteckt, damit der Inverter sicher resetiert, danach hat alles wie davor auf der zuvor verwendeten Frequenz von 865,0 MHz tagelang funktioniert. Danach habe ich nach einem Hinweis aus dem Forum nach dem Verbindungsverlust die Frequenz geändert auf 865,25 MHz. Wenige Sekunden später war die Verbindung dann wieder da. Ich habe die Frequenz dann auf 865,25 MHz belassen bis zum nächsten Verbindungsabbruch einige Tage später. Ich habe dann die Frequenz wieder auf 865 MHz zurückgestellt und die Verbindung war kurz darauf wieder hergestellt. Seit kurzem logge ich die Systemkonsole mit und da sieht man bei dem geloggten Daten, dass der Inverter offensichtlich am Nachbarkanal wartet (vielleicht ein Zufall), da die Verbindung nach dem Umschalten auf 865,25 MHz sofort bestätigt wird.
Ich würde daher bei meiner DTU als untere Grenze 865,0 MHz und als obere Grenze 866,0 MHz einstellen, bis jetzt hätte das immer zu einer Verbindungswiederaufnahme geführt. Ich logge aber weiter, vielleicht kann ich ja noch einen Fall aufzeichnen

stefan123t, 2024-11-04:
@khschmidt ich denke es ist auch wichtig die Zeit zwischen Verbindungsabbruch und dem von Dir initiierten erneuten Aufbau (mit ChannelChangeCmmand) zu berücksichtigen. Der Inverter sperrt sich bei verschiedenen Fehlern ca 10-15 Minuten lang gegen Verbindungen, erst danach reagiert er wieder auf das ChannelChangeCommand und kommuniziert dann auf dem im CCC vorgegebenen Kanal bzw dessen zugehöriger Frequenz.
In dem Logfile von Bernhard aus der Original DTU Pro S sieht man dass die DTU es fast zehn Minuten lang mit dem ChannelChangeCommand an alle drei Inverter versucht und erst dann (wieder?) eine Verbindung aufgebaut werden kann.

khschmidt, 2024-12-11:
Das Verhalten des HMS-1600-4T beim Verbindungsabbruch ist für mich jetzt klar. Ich erhalte bei einem Verbindungsabbruch eine Pushnachricht auf mein Handy durch mein Python-Script. Ein Verbindungsabbruch erfolgt bei mir so etwa alle 1 bis 3 Wochen, meist zwischen 9:00 bis 11:00 Uhr.
Start vor etwa 2 Monaten mit einer Frequenz von 865,00 MHz. Verbindungsabbruch nach ein paar Wochen; Pushnachricht aufs Handy, danach innerhalb von 2 Minuten manuelle Umstellung der Frequenz auf 865,25 MHz. Die Verbindung zum Wechselrichter ist sofort bestätigt. Der Wechselrichter hat eindeutig auf dieser Frequenz gewartet.
Einige Wochen später wieder das selbe wie zuvor geschildert, nun manuelle Umstellung der Frequenz auf 865,50 MHz. Wieder ist die Verbindung sofort wieder hergestellt.
Wieder einige Wochen später das selbe, nun manuelle Umstellung der Frequenz auf 865,75 MHz. Wieder ist die Verbindung sofort wieder hergestellt.
Resümee: Nach einem Verbindungsabbruch wartet der Wechselrichter auf der nächsthöheren Frequenz. ChannelChangeCommands der DTU gehen ins leere, da der Wechselrichter ja am Nachbarband auf Empfang wartet.
Lösungsansatz: Nach einem Verbindungsabbruch auf die nächsthöhere Frequenz schalten, eventuell in der Nacht die Frequenz wieder auf die ursprüngliche Frequenz zurückstellen; damit würden nur 2 Frequenzen im Band benützt. Ich werde diese Funktion, sobald ich Zeit habe, in mein Pythonscript erst mal einbauen.

stefan123t, 2024-12-11:
@khschmidt bitte mal die Vermutung / den Workaround in folgenden Issue überprüfen https://github.com/tbnobody/OpenDTU/issues/2438#issuecomment-2519051995

Hierbei wird nicht die Frequenz sondern nur die Sende-/Empfangsstärke des CMT2300A bzw. NRF24L01+ angepasst. Ich vermute dass hierdurch der komplette RF/Radio Stack der OpenDTU neu initialisiert wird.
Wenn es also das Problem ebenso - wie der von Dir beobachtete Wechsel auf die nächste Frequenz - behebt, dann liegt es evtl. nur an dieser Neuinitialisierung?

fr-fhe, 2025-03-01:
Hello, 
lost comm pb here also, with 2x HMS-1600-4T.

Today, non stop loss of connection.
Other days, all can be allright all day long.

**>  Wenn es also das Problem ebenso - wie der von Dir beobachtete Wechsel auf die nächste Frequenz - behebt, dann liegt es evtl. nur an dieser Neuinitialisierung?

When doing frequency changing from WebAPI, Radio Power Level (setPALevel) is issued also.
So it's not frequency change that restart communication, but really resetting PA Level.
Looking inside code, this is just a call to radio::setPALevel(), no radio stack reinitialisation.**

Hypothesis : 
- SPI communication failed, and setPALevel(), calling spi_device_polling_transmit() of IDF Library, restart SPI Communication (then radio comm, then inverter comm. etc....)
- Does CMT2300A automatically go in sleep mode ? So, setting PA Level wake up radio comm ?

BUT i have 2 inverters, and when i lost comm. with one, the second stay online (not for long).
So perhaps another hypothesis to find...

My 2 cents.

fr-fhe, 2025-03-05:
Please note : i never had a problem when using one inverter only (4 weeks continuously).
Problems appeared when installing second one.

tappas197112, 2026-09-12:
Ich habe diese Idee inzwischen extern über ioBroker und die bestehende OpenDTU-Web-API umgesetzt und erfolgreich im realen Betrieb getestet.

Der Scanner durchsucht nicht das komplette Frequenzband, sondern sucht symmetrisch um die beim Verbindungsverlust aktuell konfigurierte Frequenz `F`:

`F -250 / F +250 / F -500 / F +500 / F -750 / F +750 kHz`

Nach jedem Frequenzwechsel wird zunächst über die Web-API geprüft, ob OpenDTU die neue Frequenz tatsächlich übernommen hat. Anschließend wird geprüft, ob der Wechselrichter wieder erreichbar ist. Ein Treffer gilt erst dann als erfolgreich, wenn die Verbindung stabil bleibt und außerdem neue Telemetriedaten empfangen werden.

Bei einem kontrollierten End-to-End-Test startete OpenDTU auf 866,25 MHz:

- 866,00 MHz (-250 kHz): kein Empfang
- 866,50 MHz (+250 kHz): kein Empfang
- 865,75 MHz (-500 kHz): kein Empfang
- 866,75 MHz (+500 kHz): Kommunikation sofort wiederhergestellt

Damit hat sich zumindest in meinem Fall gezeigt, dass nur ±250 kHz als Suchbereich nicht ausgereicht hätten.

Vom erkannten Verbindungsverlust bis zur vollständig verifizierten Wiederherstellung dauerte der komplette Vorgang etwa 25 Sekunden.

Die ausführliche Beschreibung mit Recovery-Log, Algorithmus und einer neutralisierten ioBroker/JavaScript-Referenzimplementierung habe ich in Discussion #3216 veröffentlicht:

Der vollständige Beispielcode liegt zusätzlich als Gist vor:

[Field-tested ioBroker recovery scanner for sporadic HMS CMT frequency shifts using the OpenDTU Web API](https://gist.github.com/tappas197112/55b57f6912ac5b8501b76e1cba849aef)

Vielleicht sind diese Praxisergebnisse auch für die native Umsetzung in PR #2573 hilfreich.

khschmidt, 2026-09-14:
@tappas197112 ich habe damals die von dir beschriebene Lösung mit python auf einem Raspberry umgesetzt. Nach nun 2 Jahren Praxisbetrieb benötigt es in ganz seltenen Fällen sogar +-4 250khz Sprünge