tbnobody/OpenDTU #2288: [Request] Show result of WIFI scan in web ui or at least in serial console

ceggers01, 2024-09-18:
### Is your feature request related to a problem? Please describe.

There are repeated issues with the 2.4 GHz WIFI of the AVM Fritz!Box 7590 [1]. As a result, the OpenDTU cannot connect to the WIFI anymore while other devices (mobiles phones, laptops) are working fine. For the user it can be quite difficult to make a proper diagnosis (#2281).

[1] https://www.borncity.com/blog/2024/05/30/nachlese-ausfall-des-2-4-ghz-wlan-an-der-fritzbox-7590-nach-ca-5-jahren/

### Describe the solution you'd like

I suggest showing the result of the WIFI scan (I assume there is such in scan in OpenDTU) in the web ui (SSID, RSSI) or at least in the serial console. Additionally the serial console log should show some more useful information.

**Current console output:**
```
WiFi disconnected
Try reconnecting
Network lost connection
```
**Suggested output:**
```
WiFi disconnected (<reason>)
...
```
Where "reason" could be something like "AP not found", "invalid PSK", "DHCP failed", ...

### Describe alternatives you've considered

Any kind of extra information in case of a WIFI connection failure would be welcome.

tbnobody, 2024-09-18:
> I assume there is such in scan in OpenDTU

Nope, there is no active scan as every scan interrupts the active wifi connection as all channels have to be cycled to scan for SSIDs

tbnobody, 2024-09-18:
> So there is also no WIFI scan before attempting to connect to the network?

Correct. No need to scan for network SSIDs if the target SSID is already known.

> Is there any information available in case the WIFI connection cannot be established?

Yes. As you already wrote above, it shows the errors in the console.

stefan123t, 2024-09-18:
@ceggers01 thanks for the hint on the failing DCDC converter chips on 5 year old Fritz!Box 7590 models.
I will look closely at mine in an instance :eyes: :fearful:

Regarding your question on a more detailled WiFi disconnect reason, we have been discussing this in conjunction with WLAN Roaming and Mesh Steering as defined in WiFi 802.11k/r/v Standards somewhere in this thread https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2324352636 on WiFi issues.

tbnobody, 2024-09-23:
Closing this issue as a "wifi scan" is not done when connecting to a wifi network and the disconnect reason is already implemented.

closed 2024-09-23 by tbnobody