tbnobody/OpenDTU #2290: OpenDTU nicht mehr erreichbar | WLAN Abbruch mit Versionen v24.8.5 & v24.8.1 --> nur Downgrade auf v24.6.29 ist stabil

HomeAutoUser, 2024-09-20:
### What happened?

OpenDTU ist nicht mehr erreichbar, vermutlich wenn der WLAN Repeater das WLAN kurz unterbindet bei einem Kanalwechsel.
Betrieben wird das Gerät in der Konstellation WLAN ROUTER ---> Repeater --> OpenDTU betrieben.

Eine Unterscheidung gibt es bei der Firmware.
Mit der Version [v24.6.29](https://github.com/tbnobody/OpenDTU/releases/tag/v24.6.29) funktioniert es tadellos. (über Tage getestet) Bei einem Abbruch der Verbindung, ist das Gerät nicht mehr erreichbar im WLAN. Man ist gezwungen das Device vom Strom zu trennen und danach verbindet es sich mit dem WLAN neu. Der Fall ist reproduzierbar. Leider habe ich keine Terminalausgaben an dieser Stelle.

Die Schrittweise Fehlereingrenzung führte mich zum Release [v24.6.29](https://github.com/tbnobody/OpenDTU/releases/tag/v24.6.29) zurück. Nur dieses läuft stabil. Alle Neueren brechen ab.

https://github.com/tbnobody/OpenDTU/issues/2185 -> leider geschlossen
https://github.com/tbnobody/OpenDTU/issues/2202 -> ähnliches Problem

### To Reproduce Bug

Update zu v24.8.5 || v24.8.1 und warten ...

### Expected Behavior

Korrektur des WLAN Problemes (WiFi Stabilität für Version v24.8.5 & v24.8.1)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.8.5 && v24.8.1

### Relevant log/trace output

```Shell
"leider in dem Falle nicht auslesbar"
```

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-09-20:
Was ist der unterschied zu #2202?

stefan123t, 2024-09-21:
@HomeAutoUser bitte die Logs dazu sonst können wir das Problem nicht analysieren. Danke!

Follow the link to the documentation to setup for USB / serial logging:
https://www.opendtu.solar/firmware/howto/serial_console/

tbnobody, 2024-09-23:
Closing this issue as duplicate of #2202

closed 2024-09-23 by tbnobody