tbnobody/OpenDTU #2294: Kein Neustart der GUI nach Firmware upload

jblond4711, 2024-09-22:
### What happened?

Mit der 2024.9.21 war das flashen noch schlimmer. Das Fenster für de Firmware Aktualisierung bleibt permanent im Vordergrund. Mit dort 2024.9.20 wurde nach dem flashen wenigstens noch versucht die GUI neu zu starten.
Jetzt ist aber gewaltig was kaputt. 
Der letzte Firmware flash der funktioniert ist mit der 2024.8.5.

### To Reproduce Bug

Ja

### Expected Behavior

keine Ahnung

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

habe keinen Zugriff 

### What firmware variant (PIO Environment) are you using?

generic_esp32.bin

### Relevant log/trace output

```Shell
-
```

### Anything else?

-

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-09-22:
Das Problem wurde mit 2024.9.21 behoben.... Beim Upgrade auf 2024.9.21 läuft aber natürlich 2024.9.20 welches das Problem noch beinhaltet.... Es wird dadurch natürlich erst mit einem weiteren Upgrade auf die nächste kommende Version nicht mehr auftauchen.

closed 2024-09-22 by tbnobody

jblond4711, 2024-09-22:
Ok, wurde am Wifi was geändert? Ich hatte nämlich schon befürchtet neu flashen zu müssen. Zum Glück hat es gereicht das Wifi einmal neu zu starten am iPad.