tbnobody/OpenDTU #2295: DTU "friert" beim Upload der SW bei 100% ein !

mroenne2022, 2024-09-22:
### What happened?

Beim Updateprozeß auf die Verseion vom 21.9.24 friert die DTU nach Upload bei 100% ein, ohne jegliche weitere Meldung. Kommt in dem Fenster auch nicht wieder. Im neuen Fenster ist sie dann erreichbar, läuft auch alles und hat auch laut Info/System die neue Version. Das gleiche schon bei 2 DTUn gehabt. Browser Firefox ! Chip ESP32 WROOM : ESP32-D0WD-V3 !

### To Reproduce Bug

Reproduzierbar auch bei der 2.ten DTU

### Expected Behavior

Gut wäre wenn die positive Rückmeldung vom Update wieder kommt.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

98f4aed

### What firmware variant (PIO Environment) are you using?

generic

### Anything else?

[image]

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-09-22:
Duplicate of #2294 and fixed with the latest version (3b3e6995c2a178fd6b4f667b7d44726e8c9a9773). (But at the moment of updating, you are not running the latest version for obvious reasons...)

closed 2024-09-22 by tbnobody

mroenne2022, 2024-09-22:
@tbnobody : OK sorry - bei den Closed Issues hatte ich heute nicht morgen geguckt.... 
Klingt plausibel, hatte ja gelesen das genau das Problem eigentlich gefixt ist. 
Dann verhält es sich beim nächsten Update wieder wie gewohnt. Danke !