tbnobody/OpenDTU #2298: v24.9.22 - Speichern der Konfiguration funktioniert nicht mehr

jupzup, 2024-09-23:
### What happened?

v24.9.22 - Speichern der Konfiguration funktioniert nicht mehr

### To Reproduce Bug

v24.9.22 - Speichern der Konfiguration funktioniert nicht mehr

### Expected Behavior

v24.9.22 - Speichern der Konfiguration funktioniert nicht mehr

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v24.9.22

### What firmware variant (PIO Environment) are you using?

opendtu-generic.bin

### Please confirm the following

- [X] I believe this issue is a bug that affects all users of OpenDTU, not something specific to my installation.
- [X] I have already searched for relevant existing issues and discussions before opening this report.
- [X] I have updated the title field above with a concise description.
- [X] I have double checked that my inverter does not contain a W in the model name (like HMS-xxxW) as they are not supported

tbnobody, 2024-09-23:
Schreibe doch bitte mal genau was nicht geht... das ist etwas zu allgemein. Es gibt viel Konfiguration.

jupzup, 2024-09-23:
Sicherung: Sicherung der Konfigurationsdatei config.json und pin.mapping.json gehen nicht bzw, wird keine Datei zum speichern erzeugt.

Kroki0815, 2024-09-23:
Kann ich nicht bestätigen. Funktioniert bei mir einwandfrei.
Hast du mal den Browser-Cache gelöscht ?

tbnobody, 2024-09-23:
Habe es hier auch mit Chrome/Vivaldi und Safari getestet. Keinerlei Probleme.

DumpName, 2024-09-23:
Ich füge mich mal mit einem +1 und einer Fehlerbeschreibung meinerseits hinzu:
Wenn ich versuche die Netzwerk Konfiguration zu speichern erhalte ich im Browser (Firefox)den Fehler NS_ERROR_NET_RESET. Chrome wirft ERR_CONNECTION_RESET.
Es spielt hierbei keine Rolle ob ich Werte ändere oder nicht. Betrifft V24.9.21 & v24.9.22

tbnobody, 2024-09-23:
@DumpName dein Fehler hat nichts mit dem Fehler von @jupzup zu tun. @jupzup beschreibt einen Fehler beim Download der Config JSON. Das ist absolut nicht das was du beschreibst.

DumpName, 2024-09-23:
Verdammt überlesen. ICh mache einen neuen Issue auf teste aber auch einmal die config.json bei mir

DumpName, 2024-09-23:
Ich erhalte beim Upload einer neuen (oder auch der selben config Datei) auch ein NS_ERROR_NET_RESET

DumpName, 2024-09-23:
Kann es sein, dass sich das System bei allen genannten Veränderungen (und bei Änderungen auf Einstellungen -> Hardware) selbst neu startet? Dann wäre zumindest das der Zusammenhang. In dem Fall würde der Browser davon ausgehen, dass die Verbindung nicht mehr möglich ist (da der Host nicht erreichbar ist) und den oben genannten Fehler werfen.
Da jupzup oben beschreibt dass der Fehler beim Speichern der Konfigurationsdatei auftritt bin ich davon ausgegangen dass das speichern im Sinne von Upload gemeint ist (Speichern auf der DTU und nicht dem eigenen Host)

tbnobody, 2024-09-23:
> Kann es sein, dass sich das System bei allen genannten Veränderungen (und bei Änderungen auf Einstellungen -> Hardware) selbst neu startet?

Nein. Es wird nur beim Upload einer neuen JSON (und bei pin_mapping Änderungen) neu gestartet. Das Upload Thema und bei pin_mapping Änderungen gibt es aber wirklich noch ein Problem mit den Restart Handling. Wird mit der nächsten Version behoben (entstand durch das Webserver Upgrade von 3.2.1 auf 3.3.0)

Das ändert aber nicht an der Tatsache, dass @jupzup keine Configs downloaden kann (Zitat: Sicherung der Konfigurationsdatei config.json und pin.mapping.json gehen nicht bzw, wird keine Datei zum speichern erzeugt.)

jupzup, 2024-09-23:
Besten Dank für die Antworten, habe den Fehler gefunden ich greife auf openDTU über eine andere html Seite zu mittels iFrame und da klappt der Download nicht. Wenn ich direkt darauf zugreife geht es also kein Bug !!

sorry für die Verwirrung und nochmals vielen Dank für das tolle Projekt hier 🤗👍

tbnobody, 2024-09-23:
Danke für die Rückmeldung. Würde das Issue dann schließen

closed 2024-09-23 by tbnobody